ratgdo/esphome-ratgdo #1: ESP32 hardware UART issue

mariusmuja, 2023-06-15:
Using an esphome builtin `uart` component does not work on ESP32 because it uses a hardware uart, for which transmitting stops working after the TX pin is (re-)initialized in `RATGDOComponent::setup()`. 

Currently, on ESP32 only the 1305us high / 1260us low sequence appears on the wire and nothing after that. If the TX pin `setup()` and `pin_mode()` calls are commented out, the UART transmission appears on the wire, but not the 1305us high / 1260us low sequence.

The ESP8266 uses software UART and doesn't have this issue.

bdraco, 2023-06-15:
I don't see a solution here since esphome doesn't support software uart on esp32

I removed the esp32 option so nobody else tries it

If we come up with a solution later, we can always add it back

closed 2023-06-15 by bdraco

mariusmuja, 2023-06-15:
Yes, probably the solution would be to use a software serial implementation, similar to what  the`ratgdo` project does.

bdraco, 2023-06-15:
It will probably work with https://github.com/ESPHome-RATGDO/esphome-ratgdo/commit/2f9882102db6aaad8ba9ad76a121e237a69caa8a reverted but I'm traveling for the next few weeks so I won't be able to test

bdraco, 2023-06-15:
Happy to take a PR to re-add it if someone wants to fix in the mean time.  Otherwise I'll look when I get back home.

bdraco, 2023-06-17:
@mariusmuja latest push should fix it

The firmwares are building now

Unfortunately I can't validate the fix since I'm traveling. All I can test right now is that it doesn't regress the 8266

bdraco, 2023-06-17:
I had someone test it for me. Report is it works as expected

mariusmuja, 2023-06-17:
I switched to using an esp8266 and can confirm that the latest update works fine. I'll probably switch back to an esp32 now. Thanks for the quick fix @bdraco !

PR #2 (2023-06-15): remove esp32
closes #1
commit 5b633ba6 (2023-06-17): remove esp32 since #1 is unlikely to be solved

closes #1
commit 89d88c16 (2023-06-17): remove esp32 since #1 is unlikely to be solved

closes #1
PR #3 (2023-06-17): Restore esp32 support
fixes #1