ratgdo/esphome-ratgdo #9: Platform not found: 'number.ratgdo'

kantlivelong, 2023-06-24:
Hiya!

Trying to set this up in a slightly tweaked way since I use ESPHome w/ MQTT instead of the built in Home Assistant integration. 

Getting `Platform not found: 'number.ratgdo'.` and I'm not sure why since other platforms work.

Version `a7e7ff82e9ec9f417e0132817c5800e1b28acf65`

```
# cat garagedoor1.yaml 
substitutions:
  device_name: "garagedoor1"
  device_ip: "10.99.13.151"
  id_prefix: ${device_name}
  uart_tx_pin: D4
  uart_rx_pin: D2
  input_obst_pin: D7
  status_door_pin: D0
  status_obstruction_pin: D8
  dry_contact_open_pin: D5
  dry_contact_close_pin: D6
  dry_contact_light_pin: D3

esphome:
  name: ${device_name}

esp8266:
  board: d1_mini_lite
  restore_from_flash: true

external_components:
  - source:
      type: local
      path: components/esphome-ratgdo/components

<<: !include common/wifi.yaml
<<: !include common/api.yaml
<<: !include common/web_server.yaml
<<: !include common/mqtt.yaml

ota:
  password: !secret ota_password

logger:
  baud_rate: 0
  level: DEBUG

ratgdo:
  id: ${id_prefix}
  input_gdo_pin: ${uart_rx_pin}
  output_gdo_pin: ${uart_tx_pin}
  input_obst_pin: ${input_obst_pin}

sensor:
  - platform: ratgdo
    id: ${id_prefix}_openings
    type: openings
    entity_category: diagnostic
    ratgdo_id: ${id_prefix}
    name: "Openings"
    unit_of_measurement: "openings"
    icon: mdi:open-in-app

switch:
  - platform: ratgdo
    id: ${id_prefix}_lock_remotes
    type: lock
    entity_category: config
    ratgdo_id: ${id_prefix}
    name: "Lock remotes"
  - platform: gpio
    id: "${id_prefix}_status_door"
    internal: true
    pin: 
      number: ${status_door_pin}  # D0 output door status, HIGH for open, LOW for closed
      mode:
        output: true
    name: "Status door"
    entity_category: diagnostic
  - platform: gpio
    id: "${id_prefix}_status_obstruction"
    internal: true
    pin: 
      number: ${status_obstruction_pin}  # D8 output for obstruction status, HIGH for obstructed, LOW for clear
      mode:
        output: true
    name: "Status obstruction"
    entity_category: diagnostic

binary_sensor:
  - platform: ratgdo
    type: motion
    id: ${id_prefix}_motion
    ratgdo_id: ${id_prefix}
    name: "Motion"
    device_class: motion
  - platform: ratgdo
    type: obstruction
    id: ${id_prefix}_obstruction
    ratgdo_id: ${id_prefix}
    name: "Obstruction"
    device_class: problem
    on_press:
      - switch.turn_on: ${id_prefix}_status_obstruction
    on_release:
      - switch.turn_off: ${id_prefix}_status_obstruction
  - platform: ratgdo
    type: button
    id: ${id_prefix}_button
    ratgdo_id: ${id_prefix}
    name: "Button"
    entity_category: diagnostic      
  - platform: ratgdo
    type: motor
    id: ${id_prefix}_motor
    ratgdo_id: ${id_prefix}
    name: "Motor"
    device_class: running
    entity_category: diagnostic
  - platform: gpio
    id: "${id_prefix}_dry_contact_open"
    pin: 
      number: ${dry_contact_open_pin}  # D5 dry contact for opening door
      inverted: true
      mode:
        input: true
        pullup: true
    name: "Dry contact open"
    entity_category: diagnostic
    on_press:
      - if:
          condition:
            binary_sensor.is_off: ${id_prefix}_dry_contact_close
          then:
            - cover.open: ${id_prefix}_garage_door
  - platform: gpio
    id: "${id_prefix}_dry_contact_close"
    pin: 
      number: ${dry_contact_close_pin}  # D6 dry contact for closing door
      inverted: true
      mode:
        input: true
        pullup: true    
    name: "Dry contact close"
    entity_category: diagnostic
    on_press:
      - if:
          condition:
            binary_sensor.is_off: ${id_prefix}_dry_contact_open
          then:
            - cover.close: ${id_prefix}_garage_door
  - platform: gpio
    id: "${id_prefix}_dry_contact_light"
    pin: 
      number: ${dry_contact_light_pin}  # D3 dry contact for triggering light (no discrete light commands, so toggle only)
      inverted: true
      mode:
        input: true
        pullup: true    
    name: "Dry contact light"
    entity_category: diagnostic
    on_press:
      - light.toggle: ${id_prefix}_light

number:
  - platform: ratgdo
    id: ${id_prefix}_rolling_code_counter
    type: rolling_code_counter
    entity_category: config
    ratgdo_id: ${id_prefix}
    name: "Rolling code counter"
    mode: box
    unit_of_measurement: "codes"

cover:
  - platform: ratgdo
    id: ${id_prefix}_garage_door
    device_class: garage
    name: "Door"
    ratgdo_id: ${id_prefix}
    on_closed:
      - switch.turn_off: ${id_prefix}_status_door
    on_open:
      - switch.turn_on: ${id_prefix}_status_door

light:
  - platform: ratgdo
    id: ${id_prefix}_light
    name: "Light"
    ratgdo_id: ${id_prefix}

button:
  - platform: restart
    name: "Restart"
  - platform: safe_mode
    name: "Safe mode boot"
    entity_category: diagnostic
  - platform: ratgdo
    id: ${id_prefix}_sync
    type: sync
    entity_category: diagnostic
    ratgdo_id: ${id_prefix}
    name: "Sync"
  - platform: ratgdo
    id: ${id_prefix}_query
    type: query
    entity_category: diagnostic
    ratgdo_id: ${id_prefix}
    name: "Query"
```

Full log:
```
INFO Reading configuration /config/garagedoor1.yaml...
ERROR Unable to load component ratgdo.number:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/components/esphome-ratgdo/components/ratgdo/number/__init__.py", line 20, in <module>
    number.number_schema(RATGDONumber)
AttributeError: module 'esphome.components.number' has no attribute 'number_schema'
Failed config

number.ratgdo: [source /config/garagedoor1.yaml:157]
  
  Platform not found: 'number.ratgdo'.
  platform: ratgdo
  id: garagedoor1_rolling_code_counter
  type: rolling_code_counter
  entity_category: config
  ratgdo_id: garagedoor1
  name: Rolling code counter
  mode: box
  unit_of_measurement: codes
button.ratgdo: [source /config/garagedoor1.yaml:189]
  
  Must be string, got <class 'esphome.cpp_generator.MockObj'>. did you forget putting quotes around the value?.
  platform: ratgdo
  
  [id] is an invalid option for [button.ratgdo]. Please check the indentation.
  id: garagedoor1_sync
  type: sync
  entity_category: diagnostic
  ratgdo_id: garagedoor1
  name: Sync
button.ratgdo: [source /config/garagedoor1.yaml:195]
  
  Must be string, got <class 'esphome.cpp_generator.MockObj'>. did you forget putting quotes around the value?.
  platform: ratgdo
  
  [id] is an invalid option for [button.ratgdo]. Please check the indentation.
  id: garagedoor1_query
  type: query
  entity_category: diagnostic
  ratgdo_id: garagedoor1
  name: Query
```

Thanks :)

mariusmuja, 2023-06-24:
Looks like you need a fairly recent version of esphome (2023.3.0 or newer), what version are you using?

kantlivelong, 2023-06-24:
Ah! Bumping to `2023.3.0` got me a different error.

```
Compiling .pioenvs/garagedoor1/src/esphome/components/ratgdo/switch/ratgdo_switch.cpp.o
src/esphome/components/ratgdo/cover/ratgdo_cover.cpp: In member function 'virtual esphome::cover::CoverTraits esphome::ratgdo::RATGDOCover::get_traits()':
src/esphome/components/ratgdo/cover/ratgdo_cover.cpp:47:16: error: 'class esphome::cover::CoverTraits' has no member named 'set_supports_stop'; did you mean 'set_supports_tilt'?
   47 |         traits.set_supports_stop(true);
      |                ^~~~~~~~~~~~~~~~~
      |                set_supports_tilt
*** [.pioenvs/garagedoor1/src/esphome/components/ratgdo/cover/ratgdo_cover.cpp.o] Error 1
========================= [FAILED] Took 24.91 seconds =========================
```

mariusmuja, 2023-06-24:
Indeed, looks like `set_supports_stop()` was added in version 2023.5.0, any reason why you cannot use the latest esphome version?

kantlivelong, 2023-06-24:
Alrighty it built with `2023.6.2` 🎉.

I pin versions and was just using the version you suggested :)

Thanks!

PS: Would it be worth PRing a change that breaks out the parts of base I copied into it's own so users can simply include that yaml without inheriting everything else?

mariusmuja, 2023-06-24:
Great! 

Yes, of course, you're very much welcome to send a PR and we'll have a look at it.

closed 2023-06-24 by mariusmuja

AaronAutomation, 2023-11-09:
Same issue. Updating to ESPHome docker image to latest (2023.10.6) allowed me to build. Thank-you!