ratgdo/esphome-ratgdo #25: Garage Status Always On

rogerquake, 2023-06-29:
I'm trying to understand the expected behavior when using just the 3 wires attached to my garage door motor. Right now, the garage opens and closes, however the status page and inside HA says the garage door is always open.

Should the door status be accurate without any sensors attached on the opposite side of the ratgdo board? If I use the physical button or remote, should I see a status change reflected on the device and in the logs?

My end goal is for this integration to provide the correct status so I can use it along with with a deadbolt lock. I have the deadlock lock attached to a zigbee relay board and it works great, however I need a open signal to accurately trigger the unlock.

mariusmuja, 2023-06-29:
The garage door status should be accurate both when you open/close the door with ratgdo or with the button/remote. If it's not, it's either because the opener doesn't sent the status message or ratgdo cannot read the messages from the opener for whatever reason. Do other status sensors update, for example motion, openings? Does the light in HA update when you turn it on/off from the button?

Are you using the latest version of esphome-ratgdo? Can you send a log captured while you are opening & closing the door?

bdraco, 2023-06-29:
https://esphome.io/guides/cli.html#logs-command

rogerquake, 2023-06-29:
Thanks for confirming. The sensors for "Obstruction" works perfectly however the "Motor" & "Openings" sensors are Unknown. No, the animation inside HA doesn't occur, it just always says the door is open.

Yes, I should be using the latest, I just flashed it last evening.

[image]

Is looking in the logs that appear on the local web service sufficient? When I trigger events outside of the device (button, clicker) I don't see any items appear in the log. Only when I trigger the Obstruction sensors or manually open/close the door through the website/HA so I see items appear in the logs.

[esp32-ratgdo-logs1.txt](https://github.com/ESPHome-RATGDO/esphome-ratgdo/files/11910018/esp32-ratgdo-logs1.txt)

mariusmuja, 2023-06-29:
It's strange, I don't see any opener messages being received in your log. The Motor and Openings sensors are Unknown, which is consistent with no messages received, however the "Opening duration" and "Closing duration" have values, did you set those manually or were they automatically populated?

How did you flash the firmware, using the web installer? Could you flash the latest on `main` branch with esphome:
`esphome run v2board_esp8266_d1_mini_lite.yaml`, but change the logging level to VERBOSE, replace 

```
logger:
``` 
in `v2board_esp8266_d1_mini_lite.yaml` with:
```
logger:
  level: VERBOSE
```

rogerquake, 2023-06-30:
Yeah, I set those manually. I flashed it through the website. I just flashed the main branch through esphome on Windows and enabled verbose logging. Not really anything different except from json API messages:

20:56:42 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
-- | -- | -- | --
20:56:42 | [V] | [json:056] | Size after shrink 88 bytes
20:56:44 | [D] | [esp8266.preferences:238] | Saving preferences to flash...
20:57:14 | [D] | [api:102] | Accepted 192.168.4.18
20:57:14 | [W] | [api.connection:097] | 192.168.4.18: Reading failed: BAD_INDICATOR errno=11
20:57:14 | [V] | [api:114] | Removing connection to 192.168.4.18

mariusmuja, 2023-06-30:
Do you have an ESP32 board or an ESP8266? The log indicates you have an 8266..

Is the ratgdo a v2 board or a v1 board?

rogerquake, 2023-06-30:
You are correct, the board is a ESP8266, mistyped earlier. It's a v2 board.

mariusmuja, 2023-06-30:
> You are correct, the board is a ESP8266, mistyped earlier. It's a v2 board.

That all sounds good.

It seems like you are able to transmit, but not read the signal line for some reason. Yeah, might be worth trying with another esp8266 board if you have extras with the same layout/pinout.

rogerquake, 2023-06-30:
I have some extras that aren't soldered yet, so I'm working on that. I order an extra one from ratgdo to be safe incase I messed up the soldering, so we'll see. Appreciate all the help!

rogerquake, 2023-07-10:
The issue was resolved by using a new ratgdo shield.

closed 2023-07-10 by rogerquake