ratgdo/esphome-ratgdo #32: Able to Learn Rolling Code?

joe-cole1, 2023-07-16:
I apologize for what is likely a very basic question.

I am unable to flash ratgdo using their online flash and always intended to use this project as I use ESPHome for everything else anyway. So the thing I'm missing is the rolling code, which from the documentation it seems like you have to use ratgdo to get, then change over to ESPHome. Is this project capable of gathering that info, or do you absolutely HAVE to go through ratgdo first then flash the ESPHome version?

Thanks!

mariusmuja, 2023-07-17:
No, you don't have to use ratgdo first. The part of the documentation that you're referencing assumes you're coming from ratgdo and provides the steps you need to migrate to esphome-ratgdo (because ratgdo and esphome-ratgdo use the same remote id, the rolling code needs to be updated if ratgdo was used first).

If you never used ratgdo you can just go ahead and use esphome-ratgdo and the rolling code can start from zero because the garage opener has never seen the remote id before.

@bdraco Maybe we should use a different default remote_id than ratgdo, so people migrating from ratgdo don't need to worry about the rolling code (and update that part of the documentation).

bdraco, 2023-07-17:
The only downside to that is I think the opener will only accept a limited number of remotes and that might break openers at the limit.

bdraco, 2023-07-17:
I guess we could split the docs to have a section if they are starting fresh vs migrating

joe-cole1, 2023-07-17:
Ah! That makes sense. So I'll set up in the same way as ratgdo. Perfect. 

Thank you for the answer! Ya, that's pretty unclear how to start from scratch. The main documentation, to me, seems to indicate you can only migrate from ratgdo.

closed 2023-07-17 by joe-cole1