ratgdo/esphome-ratgdo #33: Lock opening and closing duration?

DennisGarvey, 2023-07-20:
Sometimes the lock and closing duration change off of the values they are supposed to be. I cannot catch when they do it, but every time I have to set them to zero, fully open and close to door to recablibrate it. Is there a way to force those values to a certain number?

mariusmuja, 2023-07-20:
> Sometimes the lock and closing duration change off of the values they are supposed to be

Currently the open and close duration are continuously calibrating by averaging the exiting value with the latest opening/closing duration.

The only reason I can see for them to change (significantly) is if the opener doesn't always send a status message when the door has finished opening/closing and esphome-ratgdo mistakenly records an incorrect duration at some later point when a status message is eventually received. 

It might make sense to remove the continuous calibration in light of this issue and only calibrate them when they are zero.

bdraco, 2023-08-07:
Closing as this was implemented in https://github.com/ESPHome-RATGDO/esphome-ratgdo/pull/34

closed 2023-08-07 by bdraco

agelwarg, 2023-12-11:
> Is there any consideration of making these values non-editable by the user to avoid this confusion (unless I'm the only person who's tried this)?

I agree with @nwithan8 and was confused by the same thing.  I thought somehow it was maybe related to the TTC (Time To Close) topic that's been discussed, causing flashing/beeping when closing to warn anyone nearby.