ratgdo/esphome-ratgdo #35: mqtt support

jonathanjacob, 2023-07-27:
Any chance for direct MQTT support similar to the original ratgdo?

I tried via the yaml file but had an issue.

jonathanjacob, 2023-07-27:
also thanks for all of the hard work!

jonathanjacob, 2023-08-28:
@mariusmuja any thoughts?

mariusmuja, 2023-08-28:
Esphome requires the use of the yaml file to configure mqtt. You need to to replace:
```
api:
  id: api_server
```
in file https://github.com/ESPHome-RATGDO/esphome-ratgdo/blob/main/static/v2board_esp8266_d1_mini_lite.yaml
with something like:
```
mqtt:
  broker: ...
  username: ...
  password: ...
```
Then upload the firmware to the device by running: `esphome run v2board_esp8266_d1_mini_lite.yaml`

PaulWieland, 2023-09-14:
This isn't really possible to do because of the way ESPHome works. We've been discussing this and there doesn't seem to be a viable way to have one firmware built on ESPHome that also supports MQTT.  You have to custom compile your own ESPHome firmware which has your MQTT credentials baked in.

closed 2024-02-08 by jonathanjacob