ratgdo/esphome-ratgdo #40: Obstruction sensor triggers when turning on light

yieldhog, 2023-08-25:
Hi --

First of all, fantastic job by all. I replaced my MyQ setup w/ this as it was so flaky and happy I've done it. The setup works stunningly, however, I did notice that when I turn _on_ my garage light, it briefly triggers a problem on the obstruction sensor (which then reverts back to OK). I didn't see this as an existing issue, not sure how to troubleshoot, but, my obstruction sensors work correctly when I manually trigger them, for what it's worth. Screengrab movie attached to demonstrate.

https://github.com/ESPHome-RATGDO/esphome-ratgdo/assets/3329021/ed2e6bfa-b891-4c8f-bbf4-ca52371d1ba6

mulcmu, 2023-08-26:
Here were a few troubleshooting questions to try and define the extent where the glitch occurs:

* Does the obstruction sensor trigger every time you turn the light on from HA user interface or is it intermittent?

* Does the obstruction sensor trigger when light is turned on using control pad, when light turns on by raising or lowering door, and/or motion sensor triggers light to turn on.

* If the light bulbs are removed from the opener, does the brief obstruction trigger still occur?

mariusmuja, 2023-08-26:
Normally the obstruction sensors receive a ~6V signal which they clamp to zero every 7ms when there is no obstruction and leave as 6V when there is an obstruction. However when the light is off (and door closed) the obstruction sensors are in "sleep mode", they don't receive a voltage. When the light is turned on, the voltage to the obstruction sensors raises from zero to the 6V they normally receive, and during that transition the obstruction detection code is fooled in thinking there is a obstruction. 

I'll push a PR soon with a more robust handing of the obstruction detection.

yieldhog, 2023-08-26:
I was able to confirm that this temporary obstruction does not occur when the light is switched on and garage door is up. Solid debugging

PR #41 (2023-08-26): Make obstruction detection ignore spurrious detections 
Make obstruction detection ignore spurrious detections when sensors awake from sleep mode, for example when light is turned on.

Fixes #40.
closed 2023-08-26 by bdraco