ratgdo/esphome-ratgdo #42: Request: ability to control beeping

yieldhog, 2023-08-26:
Request .. if it’s possible, any chance you can add the ability to control the gdo “beeping”? 

I have an automation to close the garage door after being open 10 minutes and was hoping to alert with noise and lights before doing so for safety purposes w kids. Thanks

mariusmuja, 2023-08-27:
I don't know what command to send to the opener that would make it beep (or close the door with beeps). The myQ app has this behaviour, but is triggered by the opener itself and it doesn't send any command on the wire when it performs it.

@bdraco If I remember correctly, you have a myQ bridge, is there a way to make the myQ bridge trigger beeps/close with beeps for which you can capture the cmd/data?

mariusmuja, 2023-08-27:
> I have an automation to close the garage door after being open 10 minutes and was hoping to alert with noise and lights before doing so for safety purposes w kids. Thanks

There's the TTC built-in behaviour that does exactly that. We could expose a service call to enable/disable it. However for my gdo it seems it only works if the door is fully opened, it it's partially open it doesn't do anything.

bdraco, 2023-08-27:
> I don't know what command to send to the opener that would make it beep (or close the door with beeps). The myQ app has this behaviour, but is triggered by the opener itself and it doesn't send any [...]

I've never been able to capture traffic from the Wi-Fi bridge. I'm not sure if it's exposed to the channel we are talking on.  I have an extra on I'm no longer using and happy to send it to you if you'd think it would be useful.

mariusmuja, 2023-08-27:
@bdraco I doubt it would be useful since you were unable to capture traffic from it.

@yieldhog For this issue the best we can do at this point is to expose the TTC (time-to-close) functionality built into the gdo, but which (at least on my gdo) only works when the door is fully open.

yieldhog, 2023-08-28:
@mariusmuja if we have the ability to exposure that function without a ton of effort, I personally think it would be worthwhile to be able to automate a door closing and provide the OEM safety functionality. Thanks, the project is awesome.

cgbale, 2024-02-15:
It would be great if ratgdo could add this safety feature for UL325 compliance. MyQ achieves this by sounding the keypad beep alarm for approximately 10 seconds before closing the garage door. It would be great if ratgdo could send the same command as MyQ to initiate the "remote close warning and delay" feature instead of immediately closing the garage door without warning. Based on the posts above, it seems like this command is send over wireless and the equivalent wired command has not been discovered, if it exists at all.

I almost experienced a related problem when a remote close command was sent just before I was going to back out of the garage. I almost backed into the garage door as it lowered because there was no warning or delay, and the descending door was not visible from the car until was half closed.

FYI, the UL325 safety standard specifies that a garage door must use a siren and/or strobe to prevent and/or warn of "pending entrapment." This is required for garage doors when they are closing, and for gates when both opening and closing. LiftMaster/Chamberlain uses the beeping keypad to comply with this safety standard.

jerrm, 2024-02-15:
If this is ever implemented it must be optional.  I despised the MyQ delay and was happy to see it gone.

Before ratgdo, I had implemented the relay soldered to remote hack just to bypass the delay with MyQ.

cgbale, 2024-02-15:
The beep is definitely annoying, but not as annoying as your spouse backing into the garage door because it closed without warning.

bdraco, 2024-02-15:
Currently this isn't possible because nobody has reverse engineered the commands to close with an alert yet

mullender, 2024-02-15:
I suspect the beeper is through the proprietary myQ web api only, and can never be triggered through ratgdo. 

There are a few things you can do to increase safety:
- use the presence sensor 
- Use the light status
  - Both can hint a person is present and automation should not run
- blink the lights before closing 
- don’t close the garage if it has recently been opened 
-  you could use an external beeper or speaker to alert people 
- 

I think that last option is one that ratgdo could possibly help:
- add an option to connect a recommended beeper on one of the free terminals. But I have no idea if that is even electrically possible

jerrm, 2024-02-15:
> * add an option to connect a recommended beeper on one of the free terminals. But I have no idea if that is even electrically possible
Looks like that's what konnected.io is doing with their ratgdo knock off.  No reason ratgdo couldn't do the same, but it would probably take soldering skills with existing boards, unless possibly some of the dry contact leads could be repurposed.

cgbale, 2024-02-15:
I think adding support for an external beeper and/or strobe is the best alternative. A 12v car back-up beeper could be controlled controlled via a relay if the door position status / door obstruction status leads could be repurposed.