ratgdo/esphome-ratgdo #44: Button entity

tfcarlin, 2023-09-08:
Hi,

In the Diagnostic section, there is an entity named Button. What does the state of this Button indicate?

Thanks,

Tim

mariusmuja, 2023-09-08:
It indicates the state of the wall opener button (pressed/unpressed).

tfcarlin, 2023-09-08:
Thanks!

I'm pretty sure the button is NOT being touched. Any idea why the state is "ON"?

mariusmuja, 2023-09-08:
The state is shown as received from the wall button in a message sent on the bus. Possible reason for the incorrect state is that a button release message was missed/corrupted or not sent at all.

Does the button state change at all when you press the wall button? Do the other sensors (Motor, Motion)/functions (open door, light toggle) work as expected?

tfcarlin, 2023-09-08:
1. Yes, the Button state does change when pressed. Unfortunately, it doesn't appear to "reset".
2. The Motor, Motion, and Light functions all perform (and update) as expected.

Might be a local install issue with the button (not sure what), a local bus issue, or a bug. Happily, this is the least important feature of a most AWESOME solution.

mariusmuja, 2023-09-08:
It's possible that your button doesn't send a released message, only a pressed message. For example, here's how the log looks like for my gdo when I press the wall button:

```
[13:44:27][D][ratgdo:209]: Open: button=PRESSED
[13:44:27][D][binary_sensor:036]: 'Button': Sending state ON
[13:44:27][D][ratgdo:209]: Open: button=RELEASED
[13:44:27][D][binary_sensor:036]: 'Button': Sending state OFF
[13:44:28][D][ratgdo:206]: Motor: state=ON
[13:44:28][D][binary_sensor:036]: 'Motor': Sending state ON
[13:44:28][D][ratgdo:188]: Status: door=OPENING light=ON lock=UNLOCKED
```

You could check the logs with `esphome logs v2board_esp8266_d1_mini_lite.yaml` if you see both `Open: button=PRESSED` and `Open: button=RELEASED` messages.

However, in the end the button sensor is not particularly useful, one can make a case we don't really need to show it.

tfcarlin, 2023-09-08:
Thanks for the esphome cli command to view the log. I hadn't used that before.

In the following output, I pressed and released the wall opener button twice not three times as appears to be the case in the log. I only see 1 RELEASED. 

```
[16:30:41][D][ratgdo:209]: Open: button=PRESSED
[16:30:42][D][ratgdo:188]: Status: door=OPENING light=ON lock=UNLOCKED
[16:30:42][D][cover:170]: 'Door' - Publishing:
[16:30:42][D][cover:173]:   Position: 3%
[16:30:42][D][cover:186]:   Current Operation: OPENING
[16:30:42][D][esp8266.preferences:238]: Saving preferences to flash...
[16:30:42][D][ratgdo:206]: Motor: state=ON
[16:30:42][D][binary_sensor:036]: 'Motor': Sending state ON
[16:30:42][D][cover:170]: 'Door' - Publishing:
[16:30:42][D][cover:173]:   Position: 6%
[16:30:42][D][cover:186]:   Current Operation: OPENING
[16:30:43][D][ratgdo:209]: Open: button=PRESSED
[16:30:43][D][ratgdo:209]: Open: button=RELEASED
[16:30:43][D][binary_sensor:036]: 'Button': Sending state OFF
[16:30:43][D][cover:170]: 'Door' - Publishing:
[16:30:43][D][cover:173]:   Position: 9%
[16:30:43][D][cover:186]:   Current Operation: OPENING
[16:30:44][D][cover:170]: 'Door' - Publishing:
[16:30:44][D][cover:173]:   Position: 12%
[16:30:44][D][cover:186]:   Current Operation: OPENING
[16:30:44][D][ratgdo:595]: Cancelling position callbacks
[16:30:44][D][ratgdo:188]: Status: door=STOPPED light=ON lock=UNLOCKED
[16:30:44][D][cover:170]: 'Door' - Publishing:
[16:30:44][D][cover:173]:   Position: 12%
[16:30:44][D][cover:186]:   Current Operation: IDLE
[16:30:44][D][binary_sensor:036]: 'Motor': Sending state OFF
[16:30:44][D][ratgdo:209]: Open: button=PRESSED
[16:30:44][D][binary_sensor:036]: 'Button': Sending state ON
[16:30:45][D][ratgdo:188]: Status: door=CLOSING light=ON lock=UNLOCKED
[16:30:45][D][cover:170]: 'Door' - Publishing:
[16:30:45][D][cover:173]:   Position: 9%
[16:30:45][D][cover:186]:   Current Operation: CLOSING
[16:30:45][D][ratgdo:206]: Motor: state=ON
[16:30:45][D][binary_sensor:036]: 'Motor': Sending state ON
[16:30:46][D][cover:170]: 'Door' - Publishing:
[16:30:46][D][cover:173]:   Position: 5%
[16:30:46][D][cover:186]:   Current Operation: CLOSING
[16:30:46][D][cover:170]: 'Door' - Publishing:
[16:30:46][D][cover:173]:   Position: 1%
[16:30:46][D][cover:186]:   Current Operation: CLOSING
[16:30:47][D][ratgdo:595]: Cancelling position callbacks
[16:30:47][D][ratgdo:188]: Status: door=CLOSED light=ON lock=UNLOCKED
[16:30:47][D][cover:170]: 'Door' - Publishing:
[16:30:47][D][cover:173]:   Position: 0%
[16:30:47][D][cover:186]:   Current Operation: IDLE
[16:30:47][D][switch:016]: 'Status door' Turning OFF.
[16:30:47][D][binary_sensor:036]: 'Motor': Sending state OFF
[16:30:47][D][number:012]: 'Rolling code counter': Sending state xxx
[16:30:47][D][ratgdo:215]: Openings: 6055
[16:30:47][D][sensor:093]: 'Openings': Sending state 6055.00000 openings with 0 decimals of accuracy
[16:30:47][D][esp8266.preferences:238]: Saving preferences to flash...
[16:30:50][D][ratgdo:224]: Motion: DETECTED
[16:30:50][D][binary_sensor:036]: 'Motion': Sending state ON

```

closed 2023-09-08 by tfcarlin