ratgdo/esphome-ratgdo #46: Question: Can ESP8266 NodeMCU be used?

zd3sf, 2023-09-15:
I flashed this to ESP8266 NodeMCU. Does anyone know how the pinout would translate between D1 to the NodeMCU?

zd3sf, 2023-09-15:
I found this in init.py

`CONF_OUTPUT_GDO = "output_gdo_pin"
DEFAULT_OUTPUT_GDO = (
    "D4"  # D4 red control terminal / GarageDoorOpener (UART1 TX) pin is D4 on D1 Mini
)
CONF_INPUT_GDO = "input_gdo_pin"
DEFAULT_INPUT_GDO = (
    "D2"  # D2 red control terminal / GarageDoorOpener (UART1 RX) pin is D2 on D1 Mini
)
CONF_INPUT_OBST = "input_obst_pin"
DEFAULT_INPUT_OBST = "D7"  # D7 black obstruction sensor terminal
`

I wonder if I can hook up the NodeMCU directly to opener; D2/D4: red, D7 black, Ground to white. ?

bdraco, 2023-09-15:
> I found this in init.py
>
> `CONF_OUTPUT_GDO = "output_gdo_pin" DEFAULT_OUTPUT_GDO = ( "D4" # D4 red control terminal / GarageDoorOpener (UART1 TX) pin is D4 on D1 Mini ) CONF_INPUT_GDO = [...]

You need the ratgdo board unless want to make your own unless someone makes an ESP board with the needed mosfets already built-in

mariusmuja, 2023-09-15:
> I wonder if I can hook up the NodeMCU directly to opener; D2/D4: red, D7 black, Ground to white. ?

No, not do that, you would burn the NodeMCU.

If you have some MOSFETs, you can use the schematic here: https://github.com/mariusmuja/esphome-ratgdo

For the pinout, check the table at the bottom of the page here: https://www.wemos.cc/en/latest/d1/d1_mini_lite.html. For example, D1 would be GPIO5.

zd3sf, 2023-09-15:
Thank you guys for the schematic. I might end up purchasing the board from Paul when theyre not backordered. Based on the pinouts, I think I can jump wire the shield tp the NodeMCU.