ratgdo/esphome-ratgdo #47: ESP Home logging error "Collision detected, waiting to send packet"

bigmak40, 2023-09-16:
I just set up a brand new ratgdo, using the web installer and then adopted into esphome in Home Assistant. I'm seeing the logs in esphome spamming with "Collision detected, waiting to send packet", resulting in the esphome page locking up. 

Steps followed:

* Purchased brand new ratgdo v2 with a d1 mini
* Installed esphome-ratgdo using the web installer
* Followed the link to add it into home assistant
* In home assistant, set a name, adopted it, and set the encryption key
* Examined logs for the device and saw the errors
* Verified no IP address conflicts
* Verified entities in Home Assistant and they all seem to be reporting OK
** Note: Have not connected to the garage door yet
* Tried powering up the ESP both from a PC and a wall wart, no change in error being reported

Any thoughts?

**Backup data**

ESP home yaml (auto generated)

```
substitutions:
  name: ratgdov2-d59472
  friendly_name: Large Garage Door ESP
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v2board_esp8266_d1_mini_lite.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: [redacted]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Logs collected wirelessly in esphome

```
INFO ESPHome 2023.8.3
INFO Reading configuration /config/esphome/ratgdov2-d59472.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Detected timezone 'America/New_York'
INFO Starting log output from ratgdov2-d59472.local using esphome API
INFO Successfully connected to ratgdov2-d59472.local
[14:30:42][I][app:102]: ESPHome version 2023.8.3 compiled on Sep 16 2023, 14:14:17
[14:30:42][I][app:104]: Project ratgdo.esphome version 2.0
[14:30:42][C][wifi:543]: WiFi:
[14:30:42][C][wifi:379]:   Local MAC: EC:FA:BC:D5:[redacted]
[14:30:42][C][wifi:380]:   SSID: [redacted]
[14:30:42][C][wifi:381]:   IP Address: 192.168.1.168
[14:30:42][C][wifi:382]:   BSSID: [redacted]
[14:30:42][C][wifi:384]:   Hostname: 'ratgdov2-d59472'
[14:30:42][C][wifi:386]:   Signal strength: -58 dB ▂▄▆█
[14:30:42][C][wifi:390]:   Channel: 1
[14:30:42][C][wifi:391]:   Subnet: 255.255.255.0
[14:30:42][C][wifi:392]:   Gateway: 192.168.1.1
[14:30:42][C][wifi:393]:   DNS1: 1.1.1.1
[14:30:42][C][wifi:394]:   DNS2: 1.0.0.1
[14:30:42][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:42][C][logger:301]: Logger:
[14:30:42][C][logger:302]:   Level: DEBUG
[14:30:42][C][logger:303]:   Log Baud Rate: 115200
[14:30:42][C][logger:305]:   Hardware UART: UART0
[14:30:42][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:42][D][ratgdo:419]: Collision detected, waiting to send packet
[... 149 lines cut ...]
[14:30:44][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:44][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:44][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:44][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:44][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:44][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:44][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
[14:30:45][D][ratgdo:419]: Collision detected, waiting to send packet
```

mariusmuja, 2023-09-17:
> ** Note: Have not connected to the garage door yet

You need to connect to to the garage door. It's trying to send a query message, but because it's not connected to the garage door it sees the TX line in the wrong state.

bigmak40, 2023-09-20:
Installed and confirmed that error message was operator error.

closed 2023-09-20 by bigmak40