ratgdo/esphome-ratgdo #49: Schematic, BOM and home made board

carlos-jenkins, 2023-09-18:
Hi, thanks a lot for this project and migrating ratgdo to ESPHome.

I've been trying to test this project but for me buying boards (still in backorder, and payment is a PITA) and ship them to my country is very difficult. 

Thanks @mariusmuja a lot for publishing the base schematic: https://github.com/mariusmuja/esphome-ratgdo

I would like to build it myself as it seems simple enough. I've come with the following BOM:

2 x 2n7000 MOSFET transistor
1 x 150 Ohm resistor
2 x 3.3 KOhm resistor
1 x 3.3V 1N4728A zener diode

Are these correct? Any other element that I'm missing?

I would be great to have it here and maybe instructions on how to build your own board. I can help with that as soon as I can get my hands on the components.

As this is a open source project, it would be great to make it open hardware too. I can submit a KiCAD design if you are interested in having it here.

Regards

mariusmuja, 2023-09-18:
The BOM looks good. Myself I'm using 30N06 MOSFETs because it's what I had around (talk about overkill...  )

Note that I don't know that the schematic is the same as the ratgdo v2 board, I don't have a ratgdo v2 board (tried to order one and couldn't get the website to accept a Canadian shipping address). It's just a schematic I came up with that works for me.

A KiCAD design would be great.

PaulWieland, 2023-09-18:
Shipping to Canada is $14. Shipping to most eu countries is about $15, though I haven’t enabled it because I’m not even certain what protocol EMEA chamberlain GDOs use. You might be able to get cheaper shipping by using https://www.myus.com/

That being said, I am not open sourcing the board. I don’t think $15 is an exorbitant amount to ask give the functionality it provides and I would greatly appreciate it if those who want to support the project would do so by buying boards. 

The stock shortage situation was caused by the myQ outage generated an unexpected jump in interest in the project. Everything is ramping up and I don’t think there will be another shortage after mid October.

closed 2023-09-18 by PaulWieland

mariusmuja, 2023-09-19:
> I don’t think $15 is an exorbitant amount to ask give the functionality it provides and I would greatly appreciate it if those who want to support the project would do so by buying boards.

Absolutely, $15 is an excellent price! I just tried again and was able to place an order, but I am sure it didn't work a couple months ago when I tried.

I do think the / a schematic should be available, most people that would buy the board would buy it anyway, I know I would have happily spent $15 instead of trying to make my own.