ratgdo/esphome-ratgdo #51: retrieve rolling code?

skooj, 2023-09-26:
It's possible that this is not the issue....

When I went to zip tie the ratgdo in its final mounting place after testing it for a week, I was not careful enough and ended up breaking the USB port off of my ESP8266.

As a result, I ordered some [esp32 d1 minis](https://www.amazon.com/gp/product/B08FWXFSVN/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) to try instead, hoping to use them also as bluetooth proxies (it looks like I can not do that, as of yet, but that's a separate issue and not a problem).

Long and short, I went more than a week without the ratgdo being able to work, soldered the headers on the new board, flashed it with the esp32 variant and didn't think I needed to copy the rolling code.

Now, I'm not able to open my garage door, although it is reporting it's status. Home assistant is throwing an error 

`"Failed to communicate with garage opener on startup; Check the Garage Door Opener Rolling code counter number entity history and set the entity to one number larger than the largest value in history."`
 but setting it to what it was this morning before I changed the board, or one or two after that, is not working. Am I stuck and no longer able to use the ratgdo, or am I able to somehow retrieve a working rolling code?

bdraco, 2023-09-26:
You can dig though the database to find the code from the old number entity.

Look up the entity in `states_meta` and than link it to `states`.

Or factory reset the GDO by pressing and holding the pair button 3 times for 15 seconds.  Than repair the wall control by pressing it, and repair all your remotes.

The ratgdo esp device should than be able to pair

skooj, 2023-09-26:
Thanks. I was just digging through that table and found some older codes, but they still don't seem to be working. Would the code continue changing with failed attempts to open, so, if I was on code 377 when the old board broke, I could be at 500+ now because I have been messing around trying to fix it?

closed 2023-09-26 by skooj