ratgdo/esphome-ratgdo #52: Question/Request: Support for WEMOS LOLIN S2 Mini

rako77, 2023-09-28:
Basically I have some WEMOS LOLIN S2 Mini boards and I would like to use them with the ratgdo shield. I'm willing to try and get it working myself but I don't really understand how to compile this project?

Any pointers or tips?

mariusmuja, 2023-09-28:
Here's some some quick steps:

* Install `esphome`: https://esphome.io/guides/getting_started_command_line
* Checkout the project and modify `v2board_esp32_d1_mini.yaml` by changing `board: wemos_d1_mini32` to `board: lolin_s2_mini`.
* To compile the project run: `esphome compile v2board_esp32_d1_mini.yaml`
* To compile and upload to the board run: `esphome run v2board_esp32_d1_mini.yaml`

rako77, 2023-09-28:
Thank you!

TL;DR that helped. would you be interested in a pull request that adds support for the s2?

so far I have compiled and flashed the board. (the web flasher didn't work, I had to use: https://community.home-assistant.io/t/esp32-s2-and-esphome-upload/382885/18) I was able to connect the board to home assistant and set up the wifi using the web flasher.

todo: my ratgdo just shipped and hasn't arrived yet. So testing still hasn't happened.

Presuming it all functions as expected and given the caveat that the first flash might not work with the web flasher. would you like a PR?

bdraco, 2023-09-28:
> TL;DR that helped. would you be interested in a pull request that adds support for the s2?

Yes

mattjacobson6, 2023-10-01:
I'm still waiting on my board (October backorder), but I was also thinking of using the S2 Mini.  I assume there wouldn't be any issues with using the inner row of pinouts to connect other sensors (e.g., a PIR sensor)?  

Let us know how the S2 works with the Ratgdo!

closed 2023-10-02 by rako77