ratgdo/esphome-ratgdo #58: Set Position Issue

efaden, 2023-10-03:
I'm having an issue where if I set the position be less than it's currently open it doesn't seem to go to the position, but rather just closes all of the way.   I can't seem to figure out why.  Any thoughts?

For example,

While closed, cover set Position 15 works... And it goes to 15 percent.

While open, cover set Position 15 just closes the door.

mariusmuja, 2023-10-03:
Can you have a look at the "Opening duration" and "Closing duration" and see if the "Closing duration" value looks reasonable? If the "Closing duration" is incorrect (a too large value), that would result in the behaviour that you're describing.

If the "Closing duration" looks incorrect, just reset it to zero and fully open & close the door to re-calibrate.

efaden, 2023-10-03:
They seem reasonable.  14 open, 12 closed.

efaden, 2023-10-03:
Interestingly looking at the logs out even seems to say a reasonable time.  Eg moving to position x in 2 seconds... It just doesn't actually seem to stop in 2 seconds.

efaden, 2023-10-03:
Here is a log where I closed the door.   From 15 percent, then open to 20 percent, then open to 100 percent, then close to 20 percent.  Everything is fine, but the last close to 20%... the door closes all the way.

```
[15:17:20][D][cover:076]: 'Door' - Setting
[15:17:20][D][cover:084]:   Position: 0%
[15:17:20][D][number:012]: 'Rolling code counter': Sending state 1432.000000
[15:17:21][D][ratgdo:188]: Status: door=CLOSING light=ON lock=UNLOCKED
[15:17:21][D][cover:170]: 'Door' - Publishing:
[15:17:21][D][cover:173]:   Position: 11%
[15:17:21][D][cover:186]:   Current Operation: CLOSING
[15:17:21][D][esp8266.preferences:238]: Saving preferences to flash...
[15:17:21][W][component:204]: Component preferences took a long time for an operation (0.06 s).
[15:17:21][W][component:205]: Components should block for at most 20-30ms.
[15:17:22][D][cover:170]: 'Door' - Publishing:
[15:17:22][D][cover:173]:   Position: 7%
[15:17:22][D][cover:186]:   Current Operation: CLOSING
[15:17:22][D][cover:170]: 'Door' - Publishing:
[15:17:22][D][cover:173]:   Position: 3%
[15:17:22][D][cover:186]:   Current Operation: CLOSING
[15:17:24][D][ratgdo:599]: Cancelling position callbacks
[15:17:24][D][ratgdo:188]: Status: door=CLOSED light=ON lock=UNLOCKED
[15:17:24][D][cover:170]: 'Door' - Publishing:
[15:17:24][D][cover:173]:   Position: 0%
[15:17:24][D][cover:186]:   Current Operation: IDLE
[15:17:24][D][switch:016]: 'Status door' Turning OFF.
[15:17:24][D][number:012]: 'Rolling code counter': Sending state 1433.000000
[15:17:24][D][ratgdo:215]: Openings: 759
[15:17:26][D][esp8266.preferences:238]: Saving preferences to flash...
[15:17:26][W][component:204]: Component preferences took a long time for an operation (0.06 s).
[15:17:26][W][component:205]: Components should block for at most 20-30ms.
[15:17:35][D][cover:076]: 'Door' - Setting
[15:17:35][D][cover:084]:   Position: 20%
[15:17:35][D][ratgdo:587]: Moving to position 0.20 in 2.8s
[15:17:35][D][cover:170]: 'Door' - Publishing:
[15:17:35][D][cover:173]:   Position: 0%
[15:17:35][D][cover:186]:   Current Operation: IDLE
[15:17:35][D][switch:016]: 'Status door' Turning OFF.
[15:17:35][D][number:012]: 'Rolling code counter': Sending state 1434.000000
[15:17:36][D][cover:170]: 'Door' - Publishing:
[15:17:36][D][cover:173]:   Position: 0%
[15:17:36][D][cover:186]:   Current Operation: IDLE
[15:17:36][D][switch:016]: 'Status door' Turning OFF.
[15:17:36][D][esp8266.preferences:238]: Saving preferences to flash...
[15:17:36][W][component:204]: Component preferences took a long time for an operation (0.06 s).
[15:17:36][W][component:205]: Components should block for at most 20-30ms.
[15:17:36][D][cover:170]: 'Door' - Publishing:
[15:17:36][D][cover:173]:   Position: 0%
[15:17:36][D][cover:186]:   Current Operation: IDLE
[15:17:36][D][switch:016]: 'Status door' Turning OFF.
[15:17:37][D][cover:170]: 'Door' - Publishing:
[15:17:37][D][cover:173]:   Position: 0%
[15:17:37][D][cover:186]:   Current Operation: IDLE
[15:17:37][D][switch:016]: 'Status door' Turning OFF.
[15:17:37][D][cover:170]: 'Door' - Publishing:
[15:17:37][D][cover:173]:   Position: 0%
[15:17:37][D][cover:186]:   Current Operation: IDLE
[15:17:37][D][switch:016]: 'Status door' Turning OFF.
[15:17:38][D][cover:170]: 'Door' - Publishing:
[15:17:38][D][cover:173]:   Position: 0%
[15:17:38][D][cover:186]:   Current Operation: IDLE
[15:17:38][D][switch:016]: 'Status door' Turning OFF.
[15:17:38][D][number:012]: 'Rolling code counter': Sending state 1435.000000
[15:17:39][D][ratgdo:188]: Status: door=STOPPED light=ON lock=UNLOCKED
[15:17:39][D][cover:170]: 'Door' - Publishing:
[15:17:39][D][cover:173]:   Position: 20%
[15:17:39][D][cover:186]:   Current Operation: IDLE
[15:17:41][D][esp8266.preferences:238]: Saving preferences to flash...
[15:17:41][W][component:204]: Component preferences took a long time for an operation (0.06 s).
[15:17:41][W][component:205]: Components should block for at most 20-30ms.
[15:17:43][D][cover:076]: 'Door' - Setting
[... 167 lines cut ...]
[15:18:08][D][cover:186]:   Current Operation: CLOSING
[15:18:09][D][ratgdo:206]: Motor: state=ON
[15:18:09][D][cover:170]: 'Door' - Publishing:
[15:18:09][D][cover:173]:   Position: 20%
[15:18:09][D][cover:186]:   Current Operation: CLOSING
[15:18:09][D][number:012]: 'Rolling code counter': Sending state 1438.000000
[15:18:11][D][esp8266.preferences:238]: Saving preferences to flash...
[15:18:11][W][component:204]: Component preferences took a long time for an operation (0.07 s).
[15:18:11][W][component:205]: Components should block for at most 20-30ms.
[15:18:13][D][ratgdo:188]: Status: door=CLOSED light=ON lock=UNLOCKED
[15:18:13][D][cover:170]: 'Door' - Publishing:
[15:18:13][D][cover:173]:   Position: 0%
[15:18:13][D][cover:186]:   Current Operation: IDLE
[15:18:13][D][switch:016]: 'Status door' Turning OFF.
[15:18:13][D][switch:055]: 'Status door': Sending state OFF
[15:18:13][D][binary_sensor:036]: 'Motor': Sending state OFF
[15:18:13][D][number:012]: 'Rolling code counter': Sending state 1439.000000
[15:18:13][D][ratgdo:215]: Openings: 760
[15:18:13][D][sensor:093]: 'Openings': Sending state 760.00000 openings with 0 decimals of accuracy
[15:18:13][D][number:012]: 'Rolling code counter': Sending state 1440.000000
[15:18:13][D][ratgdo:188]: Status: door=CLOSED light=ON lock=UNLOCKED
[15:18:16][D][esp8266.preferences:238]: Saving preferences to flash...
[15:18:16][W][component:204]: Component preferences took a long time for an operation (0.05 s).
[15:18:16][W][component:205]: Components should block for at most 20-30ms.
```

efaden, 2023-10-03:
You hit it on the head.... not sure why but it's completely ignoring the stop button while closing.  At least when messing with the cover through home assistant.  When I use the physical button it doesn't actually stop either, but rather reverses direction.  So if I open to 100, click the physical button to close, let it close half way then click the button again it actually opens.  

When clicking stop on the cover interface it just ignores it and goes all the way closed..... 

```
[15:36:26][D][cover:076]: 'Door' - Setting
[15:36:26][D][cover:084]:   Position: 0%
[15:36:26][D][number:012]: 'Rolling code counter': Sending state 1445.000000
[15:36:26][D][esp8266.preferences:238]: Saving preferences to flash...
[15:36:26][W][component:204]: Component preferences took a long time for an operation (0.06 s).
[15:36:26][W][component:205]: Components should block for at most 20-30ms.
[15:36:27][D][ratgdo:188]: Status: door=CLOSING light=ON lock=UNLOCKED
[15:36:27][D][cover:170]: 'Door' - Publishing:
[15:36:27][D][cover:173]:   Position: 96%
[15:36:27][D][cover:186]:   Current Operation: CLOSING
[15:36:27][D][ratgdo:206]: Motor: state=ON
[15:36:27][D][binary_sensor:036]: 'Motor': Sending state ON
[15:36:27][D][cover:170]: 'Door' - Publishing:
[15:36:27][D][cover:173]:   Position: 92%
[15:36:27][D][cover:186]:   Current Operation: CLOSING
[15:36:28][D][cover:170]: 'Door' - Publishing:
[15:36:28][D][cover:173]:   Position: 88%
[15:36:28][D][cover:186]:   Current Operation: CLOSING
[15:36:28][D][cover:170]: 'Door' - Publishing:
[15:36:28][D][cover:173]:   Position: 85%
[15:36:28][D][cover:186]:   Current Operation: CLOSING
[15:36:29][D][cover:076]: 'Door' - Setting
[15:36:29][D][cover:080]:   Command: STOP
[15:36:29][D][cover:170]: 'Door' - Publishing:
[15:36:29][D][cover:173]:   Position: 81%
[15:36:29][D][cover:186]:   Current Operation: CLOSING
[15:36:29][D][number:012]: 'Rolling code counter': Sending state 1446.000000
[15:36:29][D][cover:170]: 'Door' - Publishing:
[15:36:29][D][cover:173]:   Position: 77%
[15:36:29][D][cover:186]:   Current Operation: CLOSING
[15:36:30][D][cover:170]: 'Door' - Publishing:
[15:36:30][D][cover:173]:   Position: 73%
[15:36:30][D][cover:186]:   Current Operation: CLOSING
[15:36:30][D][cover:170]: 'Door' - Publishing:
[15:36:30][D][cover:173]:   Position: 69%
[15:36:30][D][cover:186]:   Current Operation: CLOSING
[15:36:31][D][cover:170]: 'Door' - Publishing:
[15:36:31][D][cover:173]:   Position: 65%
[15:36:31][D][cover:186]:   Current Operation: CLOSING
[15:36:31][D][esp8266.preferences:238]: Saving preferences to flash...
[15:36:31][W][component:204]: Component preferences took a long time for an operation (0.06 s).
[15:36:31][W][component:205]: Components should block for at most 20-30ms.
[15:36:31][D][cover:170]: 'Door' - Publishing:
[15:36:31][D][cover:173]:   Position: 62%
[15:36:31][D][cover:186]:   Current Operation: CLOSING
[15:36:32][D][cover:076]: 'Door' - Setting
[15:36:32][D][cover:080]:   Command: STOP
[15:36:32][D][number:012]: 'Rolling code counter': Sending state 1447.000000
[15:36:32][D][cover:170]: 'Door' - Publishing:
[15:36:32][D][cover:173]:   Position: 58%
[15:36:32][D][cover:186]:   Current Operation: CLOSING
[15:36:32][D][cover:170]: 'Door' - Publishing:
[15:36:32][D][cover:173]:   Position: 54%
[15:36:32][D][cover:186]:   Current Operation: CLOSING
[15:36:33][D][cover:170]: 'Door' - Publishing:
[15:36:33][D][cover:173]:   Position: 50%
[15:36:33][D][cover:186]:   Current Operation: CLOSING
[15:36:33][D][cover:076]: 'Door' - Setting
[15:36:33][D][cover:080]:   Command: STOP
[15:36:33][D][number:012]: 'Rolling code counter': Sending state 1448.000000
[15:36:33][D][ratgdo:206]: Motor: state=ON
[15:36:33][D][cover:170]: 'Door' - Publishing:
[15:36:33][D][cover:173]:   Position: 46%
[15:36:33][D][cover:186]:   Current Operation: CLOSING
[15:36:34][D][cover:170]: 'Door' - Publishing:
[... 30 lines cut ...]
[15:36:38][D][cover:170]: 'Door' - Publishing:
[15:36:38][D][cover:173]:   Position: 8%
[15:36:38][D][cover:186]:   Current Operation: CLOSING
[15:36:39][D][cover:170]: 'Door' - Publishing:
[15:36:39][D][cover:173]:   Position: 4%
[15:36:39][D][cover:186]:   Current Operation: CLOSING
[15:36:39][D][cover:170]: 'Door' - Publishing:
[15:36:40][D][cover:173]:   Position: 0%
[15:36:40][D][cover:186]:   Current Operation: CLOSING
[15:36:40][D][switch:016]: 'Status door' Turning OFF.
[15:36:40][D][switch:055]: 'Status door': Sending state OFF
[15:36:40][D][ratgdo:599]: Cancelling position callbacks
[15:36:40][D][ratgdo:188]: Status: door=CLOSED light=ON lock=UNLOCKED
[15:36:40][D][cover:170]: 'Door' - Publishing:
[15:36:40][D][cover:173]:   Position: 0%
[15:36:40][D][cover:186]:   Current Operation: IDLE
[15:36:40][D][switch:016]: 'Status door' Turning OFF.
[15:36:40][D][number:012]: 'Rolling code counter': Sending state 1449.000000
[15:36:40][D][binary_sensor:036]: 'Motor': Sending state OFF
[15:36:40][D][ratgdo:215]: Openings: 761
[15:36:40][D][sensor:093]: 'Openings': Sending state 761.00000 openings with 0 decimals of accuracy
[15:36:41][D][esp8266.preferences:238]: Saving preferences to flash...
[15:36:41][W][component:204]: Component preferences took a long time for an operation (0.06 s).
[15:36:41][W][component:205]: Components should block for at most 20-30ms.
```

mariusmuja, 2023-10-03:
> You hit it on the head.... not sure why but it's completely ignoring the stop button while closing. At least when messing with the cover through home assistant. When I use the physical button it [...]

Yes, that is the expected behaviour, all door openers do that as far as I know.

When using the physical button, it sends a "door toggle" command to the opener, different than the "door stop" command that is sent when clicking the stop button in home assistant. The "door toggle" switches direction while the door is closing and stops it while the door is opening.

For some reason your opener ignores the "door stop" command when closing (but strangely not when opening). Unfortunately, I don't think there's much that can be done about this. A hack-ish workaround could be to send two "door toggle" commands in quick succession to make the door stop when closing, but that would require some experimentation (the opener could ignore the second command if it's send too quickly after the first).

efaden, 2023-10-03:
That's an interesting idea.

Alternatively if the door position is changed less than the current
position, close all the way, then open to the position.  That's actually
what I did with the myq integration before switched to the ratgdo setup.

efaden, 2023-10-03:
The key would be to send two quick enough to catch the correct position,
but not so fast that the opener ignores them.

For reference my other opener does exactly the same thing.  Both
Chamberlain belt drive....

mariusmuja, 2023-10-03:
> For reference my other opener does exactly the same thing.  Both Chamberlain belt drive....

Interesting, I haven't seen/heard of this issue before. It's likely a fairly common issue if it happens for 2 out of 2 openers in your case.

When I have some time, I'll experiment with the double toggle hack to stop the door.

mariusmuja, 2023-10-03:
What is the mfg. date of your opener? I have Chamberlain belt drive with mfg. date of 11/13/15 that doesn't have the issue.

efaden, 2023-10-03:
Sounds good.

Mine are both the 980s.

Let me know if you need me test it.

mariusmuja, 2023-10-03:
Can you test branch https://github.com/ratgdo/esphome-ratgdo/tree/stop_while_closing_workaround, I added a configuration option switch "Stop while closing workaround" that when enabled sends "door toggle" twice to stop the door instead of "door stop":

[image]

The door seems to ignore the second toggle command until the status changes from CLOSING to OPENING (door starts moving up).

efaden, 2023-10-03:
It actually did work..... mostly.  But when I set it for the 15 percent it wound up at 26%.... so I'm not sure the timing part worked correctly.  here is the log

```
[19:53:18][D][cover:076]: 'Door' - Setting
[19:53:18][D][cover:084]:   Position: 15%
[19:53:18][D][ratgdo:633]: Door command: 0
[19:53:18][D][ratgdo:238]: [99592] Encode for transmit rolling=00005b7 fixed=0200000539 data=01010080
[19:53:18][D][ratgdo:587]: Moving to position 0.15 in 11.1s
[19:53:18][D][ratgdo:105]: [99640] received mine: rolling=00005b7 fixed=0200000539 data=01010080
[19:53:18][D][ratgdo:106]: cmd=280 (DOOR_ACTION) byte2=01 byte1=01 nibble=0
[19:53:18][D][cover:170]: 'Door' - Publishing:
[19:53:18][D][cover:173]:   Position: 100%
[19:53:18][D][cover:186]:   Current Operation: IDLE
[19:53:18][D][switch:012]: 'Status door' Turning ON.
[19:53:18][D][ratgdo:238]: [99728] Encode for transmit rolling=00005b7 fixed=0200000539 data=01000080
[19:53:18][D][ratgdo:105]: [99759] received mine: rolling=00005b7 fixed=0200000539 data=01000080
[19:53:18][D][ratgdo:106]: cmd=280 (DOOR_ACTION) byte2=01 byte1=00 nibble=0
[19:53:18][D][number:012]: 'Rolling code counter': Sending state 1464.000000
[19:53:19][D][cover:170]: 'Door' - Publishing:
[19:53:19][D][cover:173]:   Position: 100%
[19:53:19][D][cover:186]:   Current Operation: IDLE
[19:53:19][D][switch:012]: 'Status door' Turning ON.
[19:53:19][D][cover:170]: 'Door' - Publishing:
[19:53:19][D][cover:173]:   Position: 100%
[19:53:19][D][cover:186]:   Current Operation: IDLE
[19:53:19][D][switch:012]: 'Status door' Turning ON.
[19:53:19][D][ratgdo:109]: [100661] received rolling=000aa1e fixed=40b3d5d0d2 data=02600581
[19:53:19][D][ratgdo:110]: cmd=081 (STATUS) byte2=02 byte1=60 nibble=5
[19:53:19][D][ratgdo:187]: Status: door=CLOSING light=ON lock=UNLOCKED
[19:53:19][D][cover:170]: 'Door' - Publishing:
[19:53:19][D][cover:173]:   Position: 88%
[19:53:19][D][cover:186]:   Current Operation: CLOSING
[19:53:19][D][esp8266.preferences:238]: Saving preferences to flash...
[19:53:19][W][component:204]: Component preferences took a long time for an operation (0.06 s).
[19:53:19][W][component:205]: Components should block for at most 20-30ms.
[19:53:19][D][ratgdo:109]: [100974] received rolling=000aa1e fixed=42b3d5d0d2 data=00000084
[19:53:19][D][ratgdo:110]: cmd=284 (MOTOR_ON) byte2=00 byte1=00 nibble=0
[19:53:19][D][ratgdo:205]: Motor: state=ON
[19:53:19][D][binary_sensor:036]: 'Motor': Sending state ON
[19:53:20][D][cover:170]: 'Door' - Publishing:
[19:53:20][D][cover:173]:   Position: 85%
[19:53:20][D][cover:186]:   Current Operation: CLOSING
[19:53:20][D][cover:170]: 'Door' - Publishing:
[19:53:20][D][cover:173]:   Position: 81%
[19:53:20][D][cover:186]:   Current Operation: CLOSING
[19:53:21][D][cover:170]: 'Door' - Publishing:
[19:53:21][D][cover:173]:   Position: 77%
[19:53:21][D][cover:186]:   Current Operation: CLOSING
[19:53:21][D][cover:170]: 'Door' - Publishing:
[19:53:21][D][cover:173]:   Position: 73%
[19:53:21][D][cover:186]:   Current Operation: CLOSING
[19:53:22][D][cover:170]: 'Door' - Publishing:
[19:53:22][D][cover:173]:   Position: 69%
[19:53:22][D][cover:186]:   Current Operation: CLOSING
[19:53:22][D][cover:170]: 'Door' - Publishing:
[19:53:22][D][cover:173]:   Position: 65%
[19:53:22][D][cover:186]:   Current Operation: CLOSING
[19:53:23][D][cover:170]: 'Door' - Publishing:
[19:53:23][D][cover:173]:   Position: 62%
[19:53:23][D][cover:186]:   Current Operation: CLOSING
[19:53:23][D][cover:170]: 'Door' - Publishing:
[19:53:23][D][cover:173]:   Position: 58%
[19:53:23][D][cover:186]:   Current Operation: CLOSING
[19:53:24][D][cover:170]: 'Door' - Publishing:
[19:53:24][D][cover:173]:   Position: 54%
[19:53:24][D][cover:186]:   Current Operation: CLOSING
[19:53:24][D][cover:170]: 'Door' - Publishing:
[19:53:24][D][cover:173]:   Position: 50%
[19:53:24][D][cover:186]:   Current Operation: CLOSING
[19:53:24][D][esp8266.preferences:238]: Saving preferences to flash...
[... 52 lines cut ...]
[19:53:31][D][binary_sensor:036]: 'Motor': Sending state OFF
[19:53:31][D][cover:170]: 'Door' - Publishing:
[19:53:31][D][cover:173]:   Position: 19%
[19:53:31][D][cover:186]:   Current Operation: OPENING
[19:53:31][D][ratgdo:238]: [112336] Encode for transmit rolling=00005b9 fixed=0200000539 data=01010280
[19:53:31][D][ratgdo:105]: [112367] received mine: rolling=00005b9 fixed=0200000539 data=01010280
[19:53:31][D][ratgdo:106]: cmd=280 (DOOR_ACTION) byte2=01 byte1=01 nibble=2
[19:53:31][D][ratgdo:238]: [112466] Encode for transmit rolling=00005b9 fixed=0200000539 data=01000280
[19:53:31][D][ratgdo:105]: [112498] received mine: rolling=00005b9 fixed=0200000539 data=01000280
[19:53:31][D][ratgdo:106]: cmd=280 (DOOR_ACTION) byte2=01 byte1=00 nibble=2
[19:53:31][D][number:012]: 'Rolling code counter': Sending state 1466.000000
[19:53:31][D][cover:170]: 'Door' - Publishing:
[19:53:31][D][cover:173]:   Position: 22%
[19:53:31][D][cover:186]:   Current Operation: OPENING
[19:53:32][D][cover:170]: 'Door' - Publishing:
[19:53:32][D][cover:173]:   Position: 26%
[19:53:32][D][cover:186]:   Current Operation: OPENING
[19:53:32][D][ratgdo:109]: [113399] received rolling=000aa20 fixed=40b3d5d0d2 data=02600381
[19:53:32][D][ratgdo:110]: cmd=081 (STATUS) byte2=02 byte1=60 nibble=3
[19:53:32][D][ratgdo:599]: Cancelling position callbacks
[19:53:32][D][ratgdo:187]: Status: door=STOPPED light=ON lock=UNLOCKED
[19:53:32][D][cover:170]: 'Door' - Publishing:
[19:53:32][D][cover:173]:   Position: 26%
[19:53:32][D][cover:186]:   Current Operation: IDLE
```

mariusmuja, 2023-10-04:
Yeah, the opening position is an approximation based on the closing/opening times, there is no message from the opener that tells what the position is. 

Because the opener reports the status message with a slight delay, that introduces an error in the approximation, and it's worse for short door moves. This change from closing -> opening -> stopped involves a tiny move of the door which is close to the worst case. We can probably improve things a bit for this workaround case, by ignoring the position update for short door move before stopping.

efaden, 2023-10-04:
Yeah I follow you.  May be worth testing.

Worst case I'll just set my automation to close first.  Maybe have options
in there for how you want it handled...

Eg can choose between regular, toggle toggle or whatever this mode is, and
a third mode that will always close the door first if the Delta is less
than the current one

mariusmuja, 2023-10-10:
The door position code has been improved, so if you retest the latest on branch https://github.com/ratgdo/esphome-ratgdo/tree/stop_while_closing_workaround, it should have less of an error after stopping with the double toggle workaround.

froman753, 2023-12-11:
I also am running into this same issue, but I have not yet tried the work around branch. Is that current and effective for this problem?

efaden, 2023-12-11:
Never really got it to work correctly.  I just wound up closing then reopening it in my script instead. More reliable

ericvogel84, 2023-12-12:
Same issue. My model is B4505T and was manufactured in 2022. 

I am unable to test the branch as I have a 2.5 board. @mariusmuja, any chance you could update the test branch or bring the option into the main branch?

mariusmuja, 2023-12-13:
I updated the test branch, give it a try.

ericvogel84, 2023-12-13:
Getting an error trying to update to the branch.

```
INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/ratgdov25i-f61506.yaml...
Failed config

packages: [source /config/esphome/ratgdov25i-f61506.yaml:5]
  
  Failed to load packages. v25iboard.yaml does not exist in repository.
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@stop_while_closing_workaround

```

Is it not as simple as changing the _main_ to _stop_while_closing_workaround_ in the packages line of my .yaml?

```
substitutions:
  name: ratgdov25i-f61506
  friendly_name: ratgdov2.5i f61506
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@stop_while_closing_workaround
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 5fBoOm67S+yTA+LrrBFZMhg2Enf+WpWseDlP8N20/0s=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

```

jake-b, 2024-01-14:
Just want to add another data point.  My GDO is relatively now, with a manufacture data of 8/2023.  It exhibits this behavior as well.   Seems like a "safety thing" on new GDOs that a press while closing is a "panic" and it reverses rather than stops.

I built and installed the branch, and removed/reinstalled the device in home assistant, but I don't see the "stop while closing workaround" entity to try it for myself.

My hope is that this will be eventually merged, as I think we'll see this behavior on many late model garage door openers.

mariusmuja, 2024-01-14:
> Seems like a "safety thing" on new GDOs that a press while closing is a "panic" and it reverses rather than stops.

It's not a press while closing, a press is a "toggle" command and all GDOs that I know of reverse direction when button is pressed while closing. In addition to "toggle" we discovered other commands that can be sent to the GDO in sec+2, such as "stop", "open", "close" that do exactly that. This issue is that the "stop" command is ignored while the door is closing, which is less safe since the door keeps on moving and the workaround is to use the "toggle" command twice to stop the door.

> but I don't see the "stop while closing workaround" entity to try it for myself.

It's only on this branch currently: https://github.com/ratgdo/esphome-ratgdo/tree/stop_while_closing_workaround

jake-b, 2024-01-14:
> It's not a press while closing, a press is a "toggle" command and all GDOs that I know of reverse direction when button is pressed while closing. In addition to "toggle" we discovered other commands [...]

Ahh, I understand the behavior now. The stop button does seem to work while the garage is opening, but it does nothing while the door is closing.  A double toggle would work, but I think an "open then stop" two step would work as well.  (Just based on my playing with the HomeAssistant UI.)

> It's only on this branch currently: https://github.com/ratgdo/esphome-ratgdo/tree/stop_while_closing_workaround

Yeah, I thought I installed this branch, but I admit I find the ESPHome build system a little inscrutable.  I may have built the branch wrong.  I'll give it another try.

efaden, 2024-08-23:
Any updates to this?  Would be nice to utilize, even if it has to close first then re-open?