ratgdo/esphome-ratgdo #59: Home Assistant Card Code?

keithnet, 2023-10-03:
Sorry, I've looked around a bunch and cant find the code to do a card in home assistant as shown in the Readme (with opening percentages). Can someone point me in the right direction?

mariusmuja, 2023-10-03:
You could add the garage door cover entity to a number of cards, for example you could add a Button card with the following content:
```yaml
show_name: true
show_icon: true
type: button
entity: cover.garage_door
icon: mdi:garage
name: Garage door
```

[image]

and clicking on it would open the cover dialog as shown in the Readme. The entity name would likely be different for you.

keithnet, 2023-10-03:
I'm currently using a custom button-card and it's similar to what you show, but it doesn't show the percentage the garage door has opened:

type: custom:button-card
name: Keith's Garage (Double Click to Open)
card:
  - height: 10px
  - width: 60px
  - title: test
double_tap_action:
  action: call-service
  service: cover.open_cover
  service_data:
    entity_id: cover.keiths_garage_opener_door
entity: cover.keiths_garage_opener_door
show_state: true
size: 40%
state:
  - color: green
    icon: mdi:garage
    value: closed
  - color: orange
    icon: mdi:garage-open
    value: open
styles: null
tap_action:
  action: call-service
  service: cover.close_cover
  service_data:
    entity_id: cover.keiths_garage_opener_door

mariusmuja, 2023-10-03:
If you open the entity dialog (clicking on the door cover entity in a number of places) will show you the door position like so:

[image]

If you want to see the position in a card you could use a tile card:

[image]

keithnet, 2023-10-03:
OK, i understand now. Thank you!

closed 2023-10-03 by keithnet