ratgdo/esphome-ratgdo #62: Support for Non-Security+ 2.0 Series Door Hardware

northrup, 2023-10-11:
In the original project an [issue was opened](https://github.com/PaulWieland/ratgdo/issues/54) to support the strange hybrid of lift master that is the older Liftmaster 8500. This model uses Security+ 2.0 for the wireless remotes, but the hard wired openers do not speak the protocol. This creates issues. 

It looks like Paul has solved the problem and has the work in a branch: https://github.com/PaulWieland/ratgdo/tree/2.0-secplus1.0

Just creating an issue here to see if it would be possible to get this same support in the ESPHome version?

PaulWieland, 2023-10-12:
Porting to ESPHome can begin once I have a stable working solution. Right now I'm still making a lot of changes.

dubhunter, 2023-11-12:
Now that https://github.com/PaulWieland/ratgdo/issues/101 is resolved, will we get this in ESPHome soon? I am debating setting up MQTT w/ my new Rats, or wait for ESPHome.

Thanks!

northrup, 2023-11-12:
🙏 that would be awesome

niemyjski, 2023-11-22:
@PaulWieland Are we now at a stable solution? Is there anything I can do to help out?

dubhunter, 2023-11-30:
Hi @PaulWieland,

Any update? When I tried the MQTT version, my GDO would randomly open (caught it via MyQ), so I unplugged it. Might need a reflash, but I’m really hoping for this version.

PaulWieland, 2023-12-17:
I think the implementation I have in the mqtt firmware is working well but it requires that you have a myq wall panel (e.g. 889lm). I have to add lock/unlock but that’s trivial. 

These wall panels constantly query the gdo for status. 

If you don’t have a myq wall panel then the door status isn’t available. I wrote an 889lm emulator, and that does work but it has the side effect of stomping out the light and lock signals from wall panels such as the 78lw. 

Lastly there are some wall panels which send a constant stream of serial commands to the gdo (like the 889lm does) but it doesn’t include the door status for some reason. The command to request door status is `0x38`, and these panels request `0x37`, which causes the gdo to return a seemingly constant value. I don’t see a solution other than dry contact control with limit switches for these.  

I’m going to continue experimenting with the emulator to see if pausing the polling as soon as the 78lm transmits can help the problem.

dubhunter, 2023-12-17:
Paul,

I _do_ have the 889LM, but the GDO would randomly open (sometimes in the middle of the night). Still really hoping for the ease of the ESPHome firmware which is working great on my other GDO.

agelwarg, 2023-12-17:
@PaulWieland I have the 78lm, and as you mention the wall lock/light buttons do not work.  That's not a big deal to me.  Are you saying that the emulator you have is part of the firmware or that you use the emulator for your testing/development?

Additionally, the light itself seems to turn off after a few seconds whenever I turn it on (from HA), and other times it's the reverse where I'm trying to turn it off, but it keeps automatically turning back on.  I can't figure out what changes the "direction" of the problem, but it's a problem regardless.  I'm not sure if this is a known issue as of now, and we should just leave the bulbs disconnected which is what I have right now.

FrankJaco, 2023-12-18:
@PaulWieland   I have the Chamberlain 41A6318 wall control and when it is connected with the ratgdo the symptoms are bizarre. Often in Home Assistant the arrows are reversed offering up the down arrow when up should be available. About 50% of the time if I hit an arrow the door won't at first move, the ratdgo goes unavailable in HA, then after several **minutes** the ratgdo again comes available and the command I issues minutes ago actually executes.  (BTW: my WiFi connection is perfect and throughout the event where it goes unavailable I am able to ping -t the ratgdo without packet loss.)

The light when I turn it on via HA, turns off in a few seconds or blinks once. 

If I disconnect the wall panel, things work as expected, so it does seem I have one of these strange interfering panels.

Although I would like to have my wall panel function again, I can emulate all of the missing functions via HA devices and automations so I am content with my situation.  I  have a spare 41A6318 in my spare parts.  If if would be helpful I could give/send it to you.  Especially if it eventually will let me re-enable my wall panel.  

https://www.movingupgaragedoors.com/chamberlain-41a6318-motion-detecting-control-console-with-lcd-monitor

fj

jeeftor, 2023-12-18:
Are you running the MQTT firmware or the ESPHome? 

I'm just finding that my mqtt dropped off the network and I'm not sure what is causing it... I just rebooted it but I'm thinking I'll have to plug a computer into unit to figure out whats up.

gullygossner, 2023-12-18:
> Are you running the MQTT firmware or the ESPHome?
> 
> I'm just finding that my mqtt dropped off the network and I'm not sure what is causing it... I just rebooted it but I'm thinking I'll have to plug a computer into unit to figure out whats up.

Running mqtt firmware

kpfleming, 2023-12-23:
I just installed a 2.5i board with a LiftMaster 8500W (not using the manufacturer WiFi/light unit), using the latest code from this ESPHome repo. It seems to be working, I get the correct status and the up/down controls work. I don't have an 889LM, I have an 800LMW, so no motion sensing for me either. I also have one keypad remote and it still works fine, although that's to be expected :-)

maddios, 2023-12-24:
Any updates or ETA for implementing Security + 1.0 in ESPHome version of ratgdo? Unfortunately I'm unfortunately stuck with one of the older liftmaster 8500 units and I'd rather use ESPHome version with HA unless ETA to get this implemented is super long.

Thanks

PaulWieland, 2023-12-24:
> 8500W

8500W is sec plus 2.0

All other 8500 models are sec plus 1

kpfleming, 2023-12-24:
Thanks, I'm glad to hear that :-)

allendustin, 2024-01-06:
It would be nice to know whether the ESPHome version or ratgdo is going to support Security + 1.0 openers.  Is there a good place to go to follow any developments on that front?  Nice work on this project @PaulWieland !!!  I absolutely love what you've done with this.

jeeftor, 2024-01-06:
I've run into my second issue with the mainline security 1.0 release - where my device basically drops off the network w/out a reboot. 

Any updates for us :)

bdraco, 2024-01-06:
Its probably a long way out honestly as I don't have any 1.0 devices to test in person with.

restauffer, 2024-01-06:
> Its probably a long way out honestly as I don't have any 1.0 devices to test in person with.

It might be there are people waiting for the Sec +1.0 ESPHome version that would be willing to contribute toward you getting the needed h/w to do testing. I, for one, would contribute.

rbflurry, 2024-01-06:
> Its probably a long way out honestly as I don't have any 1.0 devices to test in person with.

I could offer a remote session, a ratgdo and 889lm if it would be beneficial. I'm sure having the physical hardware would be easier.

outgeeked, 2024-01-06:
> > Its probably a long way out honestly as I don't have any 1.0 devices to test in person with.
> 
> I could offer a remote session, a ratgdo and 889lm if it would be beneficial. I'm sure having the physical hardware would be easier. 

I'm likewise happy to offer a remote session, ratgdo, and "dumb" remote (representing the 889LM-ineligible crowd). Let me know!

NukeThemTillTheyGlow, 2024-01-06:
> > Its probably a long way out honestly as I don't have any 1.0 devices to test in person with.
> 
> It might be there are people waiting for the Sec +1.0 ESPHome version that would be willing to contribute toward you getting the needed h/w to do testing. I, for one, would contribute.

Count me in also.  I'm laterally a few minutes away from mounting my 3rd/last ratgdo on my only Security 1.0 GDO and would much prefer to use the ESPHome code after the frustrating efforts to get the MQTT code to work across different subnets.

jeeftor, 2024-01-07:
Is there certain data/data streams we can capture and share back with project?

mariusmuja, 2024-01-11:
There is now a branch (https://github.com/ratgdo/esphome-ratgdo/tree/sec_plus_v1) where Sec+1.0 support is implemented. It has received very little testing so far due to not having sec+1.0 hardware, so expect bugs to still be present. 

Testing this branch and reporting issues here (together with log output) would help fix any bugs and speed up this getting merged to main.

tfcarlin, 2024-01-11:
I have an older Sec+ 1.0 controller and am happy to test. How would I update the yaml file to use this branch?

maddios, 2024-01-11:
> I have an older Sec+ 1.0 controller and am happy to test. How would I update the yaml file to use this branch?

Same, I'll gladly test this for you on my 8500 + 889LM

mariusmuja, 2024-01-11:
> I have an older Sec+ 1.0 controller and am happy to test. How would I update the yaml file to use this branch?

You should be able to compile&flash from this branch using the command: 

`esphome run v25iboard_secplusv1.yaml --device <ip_address_of_ratgdo>`

maddios, 2024-01-12:
tried to run the command and got these errors, first time messing with the esphome cli, not sure what i'm screwing up.

`esphome version
Version: 2022.9.4`

<details><summary>Details</summary>
<p>

esphome run v25iboard_secplusv1.yaml --device 20.1.2.110
INFO Reading configuration v25iboard_secplusv1.yaml...
INFO Cloning https://github.com/ratgdo/esphome-ratgdo@sec_plus_v1
INFO Cloning https://github.com/ratgdo/esphome-ratgdo@sec_plus_v1
ERROR Unable to load component ratgdo.number:
Traceback (most recent call last):
  File "/Users/maddios/Library/Python/3.8/lib/python/site-packages/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File ".esphome/external_components/cc6eb0ad/components/ratgdo/number/__init__.py", line 23, in <module>
    number.number_schema(RATGDONumber)
AttributeError: module 'esphome.components.number' has no attribute 'number_schema'
ERROR Unable to load component ratgdo.number:
Traceback (most recent call last):
  File "/Users/maddios/Library/Python/3.8/lib/python/site-packages/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File ".esphome/external_components/cc6eb0ad/components/ratgdo/number/__init__.py", line 23, in <module>
    number.number_schema(RATGDONumber)
AttributeError: module 'esphome.components.number' has no attribute 'number_schema'
ERROR Unable to load component ratgdo.number:
Traceback (most recent call last):
  File "/Users/maddios/Library/Python/3.8/lib/python/site-packages/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File ".esphome/external_components/cc6eb0ad/components/ratgdo/number/__init__.py", line 23, in <module>
    number.number_schema(RATGDONumber)
AttributeError: module 'esphome.components.number' has no attribute 'number_schema'
ERROR Unable to load component ratgdo.number:
Traceback (most recent call last):
  File "/Users/maddios/Library/Python/3.8/lib/python/site-packages/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File ".esphome/external_components/cc6eb0ad/components/ratgdo/number/__init__.py", line 23, in <module>
[... 34 lines cut ...]
  name: Closing duration
  unit_of_measurement: s
number.ratgdo: [source .esphome/packages/cc6eb0ad/base_secplusv1.yaml:171]

  Platform not found: 'number.ratgdo'.
  platform: ratgdo
  id: ratgdov25i_client_id
  type: client_id
  entity_category: config
  ratgdo_id: ratgdov25i
  name: Client ID
  mode: box
esphome: None
  name: ratgdov25i

  [friendly_name] is an invalid option for [esphome]. Please check the indentation.
  friendly_name: ratgdov2.5i
  name_add_mac_suffix: True
  project:
    name: ratgdo.esphome
    version: 2.5i
  build_path: .esphome/build/ratgdov25i

</p>
</details>

maddios, 2024-01-12:
> @maddios You need a newer version of esphome cli, try updating to the latest version.

Ahh, makes sense. Ok, I'll go the docker route then. I tried just doing pip3 install esphome on my mac.

maddios, 2024-01-12:
Made some progress, this time it gets to 36% uploading and times out, I tried it a few times, it fails at the same % every time.

<details><summary>Details</summary>
<p>

esphome run v25iboard_secplusv1.yaml --device 20.1.2.110
INFO ESPHome 2023.12.5
INFO Reading configuration v25iboard_secplusv1.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@sec_plus_v1
INFO Updating https://github.com/ratgdo/esphome-ratgdo@sec_plus_v1
INFO Detected timezone 'America/New_York'
INFO Generating C++ source...
INFO Compiling app...
Processing ratgdov25i (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
|-- Secplus @ 1.0.0+sha.f98c322
|-- EspSoftwareSerial @ 8.1.0+sha.57f1aa7
|-- Improv @ 1.2.3
RAM:   [====      ]  43.5% (used 35660 bytes from 81920 bytes)
Flash: [=====     ]  50.4% (used 526337 bytes from 1044464 bytes)
=================================================================================================================== [SUCCESS] Took 3.06 seconds ===================================================================================================================
INFO Successfully compiled program.
INFO Connecting to 20.1.2.110
INFO Uploading .esphome/build/ratgdov25i/.pioenvs/ratgdov25i/firmware.bin (530496 bytes)
INFO Compressed to 355112 bytes
Uploading: [======================                                      ] 36% 
ERROR Error sending data: timed out

</p>
</details> 

I get this output on the ratgdo itself:

`
21:54:36 | [D] | [ota:151] | Starting OTA Update from 20.1.1.36...
-- | -- | -- | --
21:54:36 | [D] | [ota:320] | OTA in progress: 0.3%`

And nothing further. At this point if I just hit refresh the gdo loads a blank-ish page with the same version # at the top and left table is empty, only showing "Name" and "State".

Rebooting the gdo brings it back to the same version as before it started "ratgdov2.5i 86b402"

jeeftor, 2024-01-12:
Just for clarification -> you would need to flash the board with esp home 1st before uploading this firmware? or can you OTA against the MQTT version?

taylorcoffelt, 2024-01-12:
I managed to get mine flashed, flashing via esphome via serial or OTA kept crashing out like @maddios 
Eventually I used the web ui to build the OTA update bin file and uploaded it via the http interface OTA upload field, and that worked.

Looks like the following state for me:

- Door Toggle Button - Works
- Light Switch - Works
- Door State - Reported as 'OPEN' constantly, don't see anything change here
- Motion State - Reported as unavailable
- Obstruction State - Reported as unavailable

Getting the following repeating logs from the device:

```
[20:40:46][D][ratgdo_secplus1:227]: Received packet: [39 00]
[20:40:47][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[20:40:47][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[20:40:47][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[20:40:47][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[20:40:47][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[20:40:47][D][ratgdo_secplus1:227]: Received packet: [3A 58]
[20:40:47][D][ratgdo:185]: Light state=OFF
[20:40:47][D][ratgdo:191]: Lock state=UNLOCKED
```

maddios, 2024-01-12:
> I managed to get mine flashed, flashing via esphome via serial or OTA kept crashing out like @maddios Eventually I used the web ui to build the OTA update bin file and uploaded it via the http [...]

How did you build the bin file? I tried to flash the firmware.bin file esphome compile generated but my OTA update just hangs at 11% and eventually the update request times out

taylorcoffelt, 2024-01-12:
I used the esphome web interface to build it with the following config:

```
---
substitutions:
  id_prefix: ratgdov25i
  friendly_name: "ratgdov2.5i"
  uart_tx_pin: D1
  uart_rx_pin: D2
  input_obst_pin: D7
  status_door_pin: D0
  status_obstruction_pin: D8
  dry_contact_open_pin: D5
  dry_contact_close_pin: D6
  dry_contact_light_pin: D3

web_server:

esphome:
  name: ratgdov25i
  friendly_name: ${friendly_name}
  project:
    name: ratgdo.esphome
    version: "2.5i"

esp8266:
  board: d1_mini
  restore_from_flash: true

dashboard_import:
  package_import_url: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main

packages:
  remote_package:
    url: https://github.com/ratgdo/esphome-ratgdo
    files: [base_secplusv1.yaml]
    refresh: 1s
    ref: sec_plus_v1

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

api:
  id: api_server

ota:

improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

logger:
````

niemyjski, 2024-01-12:
Anything we need to do for 2.0 devices? Not sure what's different between this and 2.5i

mariusmuja, 2024-01-12:
> Just for clarification -> you would need to flash the board with esp home 1st before uploading this firmware? or can you OTA against the MQTT version?

Yeah, when I suggested the esphome cli OTA command above I didn't consider flashing from mqtt (which most people testing this would probably need to do). I'm not sure if/how OTA can be done from mqtt to esphome directly, I need to look into that.

taylorcoffelt, 2024-01-12:
> Anything we need to do for 2.0 devices? Not sure what's different between this and 2.5i

It looks like most of it is making sure the refs to `sec_plus_v1` branch are set up correctly in the config. Probably just update the `board` and pin `substitutions` as needed for your board version as needed, using the standard config as a reference.

mariusmuja, 2024-01-12:
> * Door State - Reported as 'OPEN' constantly, don't see anything change here

Can you post a longer log? The door state packet should look like this:
```
[XX:XX:XX][D][ratgdo_secplus1:227]: Received packet: [38 XX]
```
> 
> * Motion State - Reported as unavailable

This is not implemented, does your wall panel have a motion sensor?
> 
> * Obstruction State - Reported as unavailable

The obstruction handling is the same as for sec+2.0, using the obstruction wire. Is the obstruction wire connected ?

> 
> * Lock Button - Can enable the locked state, but can't go to an unlocked state

That's strange, the same command is used for both enabling and disabling the lock (lock toggle). Can you post a log where you enable/disable lock from the wall panel and mention where in the log you do each action?

mariusmuja, 2024-01-12:
> Anything we need to do for 2.0 devices? Not sure what's different between this and 2.5i

The 2.0 board uses a TX pin that's being pulled HIGH at boot, which means that under certain conditions (power failure, when power comes back for example) it could open the garage door. If you're handy with a soldering iron, it's possible to convert a 2.0 board to 2.5 one.

mariusmuja, 2024-01-12:
> How did you build the bin file? I tried to flash the firmware.bin file esphome compile generated but my OTA update just hangs at 11% and eventually the update request times out

Not sure what the issue is, maybe try flashing by uploading the .bin file on the ratgdo web page...

taylorcoffelt, 2024-01-12:
>This is not implemented, does your wall panel have a motion sensor?
Sure does, looks like this: [https://www.northshorecommercialdoor.com/limo39smcopa.html/](https://www.northshorecommercialdoor.com/limo39smcopa.html/)

I'll get some longer logs for you shortly

mariusmuja, 2024-01-12:
> I'll get some longer logs for you shortly

Great! Try to have a section in the log where there's only motion to figure out what messages get sent when motion is detected.

taylorcoffelt, 2024-01-12:
Here's some logs for the following actions:
- Enter Garage
- Press 'Toggle' button in HA
- Wait for Garage door to open fully and come to a complete stop
- Press the 'Close' Button in HA (No action taken by the GDO)
- Press the 'Toggle' button, wait for the door to close

I'll try to get some specific to motion next.

>The obstruction handling is the same as for sec+2.0, using the obstruction wire. Is the obstruction wire connected ?
I do not, I wasn't sure if that was a state that could be communicated by the controller connection

<details><summary>RATGDO ESPHome Logs</summary>
<p>

```
[21:25:57][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:25:57][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:25:57][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:25:57][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:25:57][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:25:57][D][ratgdo_secplus1:227]: Received packet: [3A 58]
[21:25:57][D][ratgdo:185]: Light state=OFF
[21:25:57][D][ratgdo:191]: Lock state=UNLOCKED
[21:25:57][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:25:58][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:25:58][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:25:58][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:25:58][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:25:58][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:25:58][D][ratgdo_secplus1:227]: Received packet: [3A 58]
[21:25:58][D][ratgdo:185]: Light state=OFF
[21:25:58][D][ratgdo:191]: Lock state=UNLOCKED
[21:25:59][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:25:59][D][button:010]: 'Toggle door' Pressed.
[21:25:59][D][ratgdo_secplus1:123]: Door action: TOGGLE, door state: UNKNOWN
[21:25:59][W][ratgdo_secplus1:125]: Unknown current door state, ignoring command: TOGGLE
[21:25:59][D][ratgdo_secplus1:232]: Sending packet: [30 31]
[21:25:59][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:25:59][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:25:59][D][ratgdo_secplus1:189]: Ignoring byte (0): 80, baud: 1200
[21:25:59][D][ratgdo_secplus1:189]: Ignoring byte (0): 82, baud: 1200
[21:25:59][D][ratgdo_secplus1:189]: Ignoring byte (0): 3F, baud: 1200
[21:26:00][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:26:00][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:26:00][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:26:00][D][ratgdo_secplus1:227]: Received packet: [3A 5C]
[21:26:00][D][ratgdo:191]: Lock state=UNLOCKED
[21:26:01][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:26:01][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:26:01][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:26:01][D][ratgdo_secplus1:189]: Ignoring byte (0): 80, baud: 1200
[21:26:01][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:26:01][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:26:02][D][ratgdo_secplus1:227]: Received packet: [3A 5C]
[21:26:02][D][ratgdo:185]: Light state=ON
[21:26:02][D][ratgdo:191]: Lock state=UNLOCKED
[21:26:02][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:26:02][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:26:02][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:26:02][D][ratgdo_secplus1:189]: Ignoring byte (0): 80, baud: 1200
[21:26:03][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:26:03][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:26:03][D][ratgdo_secplus1:227]: Received packet: [3A 5C]
[21:26:03][D][ratgdo:185]: Light state=ON
[21:26:03][D][ratgdo:191]: Lock state=UNLOCKED
[21:26:03][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:26:04][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:26:04][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:26:04][D][ratgdo_secplus1:189]: Ignoring byte (0): 80, baud: 1200
[... 174 lines cut ...]
[21:26:32][D][ratgdo_secplus1:227]: Received packet: [3A 5C]
[21:26:32][D][ratgdo:185]: Light state=ON
[21:26:32][D][ratgdo:191]: Lock state=UNLOCKED
[21:26:32][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:26:32][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:26:32][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:26:33][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:26:33][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:26:33][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:26:33][D][ratgdo_secplus1:227]: Received packet: [3A 5C]
[21:26:33][D][ratgdo:185]: Light state=ON
[21:26:33][D][ratgdo:191]: Lock state=UNLOCKED
[21:26:33][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:26:34][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:26:34][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:26:34][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:26:34][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:26:34][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:26:34][D][ratgdo_secplus1:227]: Received packet: [3A 5C]
[21:26:34][D][ratgdo:185]: Light state=ON
[21:26:34][D][ratgdo:191]: Lock state=UNLOCKED
[21:26:35][D][ratgdo_secplus1:227]: Received packet: [39 00]
```
</p>
</details>

maddios, 2024-01-12:
Ok, so compiling via esphome in home assistant worked. OTA update wiped out my wifi settings, but I was able to fix them with the installer on this github page.

<html>
<body>

23:29:21 | [D] | [ratgdo_secplus1:060] | Looking for security+ 1.0 wall panel...
-- | -- | -- | --
23:29:21 | [D] | [ratgdo_secplus1:067] | No wall panel detected. Switching to emulation mode.

</body>
</html>

I'll install this ratgdo on my garage door tomorrow and run some tests

taylorcoffelt, 2024-01-12:
@mariusmuja Here's logs from the following actions:

- GDO in Light Off state
- Motion in front of PIR sensor
- ~2m of logs after

<details><summary>Details</summary>
<p>

```
[21:34:01][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:34:02][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:34:02][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:34:02][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:34:02][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:34:02][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:34:02][D][ratgdo_secplus1:227]: Received packet: [3A 58]
[21:34:02][D][ratgdo:185]: Light state=OFF
[21:34:02][D][ratgdo:191]: Lock state=UNLOCKED
[21:34:03][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:34:03][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:34:03][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:34:03][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:34:03][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:34:03][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:34:04][D][ratgdo_secplus1:227]: Received packet: [3A 58]
[21:34:04][D][ratgdo:185]: Light state=OFF
[21:34:04][D][ratgdo:191]: Lock state=UNLOCKED
[21:34:04][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:34:04][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:34:04][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:34:04][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:34:05][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:34:05][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:34:05][D][ratgdo_secplus1:227]: Received packet: [3A 58]
[21:34:05][D][ratgdo:185]: Light state=OFF
[21:34:05][D][ratgdo:191]: Lock state=UNLOCKED
[21:34:05][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:34:06][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:34:06][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:34:06][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:34:06][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:34:06][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:34:06][D][ratgdo_secplus1:227]: Received packet: [3A 58]
[21:34:06][D][ratgdo:185]: Light state=OFF
[21:34:06][D][ratgdo:191]: Lock state=UNLOCKED
[21:34:07][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:34:07][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:34:07][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:34:07][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:34:07][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:34:07][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:34:08][D][ratgdo_secplus1:227]: Received packet: [3A 58]
[21:34:08][D][ratgdo:185]: Light state=OFF
[21:34:08][D][ratgdo:191]: Lock state=UNLOCKED
[21:34:08][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:34:08][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:34:08][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:34:08][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:34:09][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:34:09][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:34:09][D][ratgdo_secplus1:227]: Received packet: [3A 58]
[21:34:09][D][ratgdo:185]: Light state=OFF
[21:34:09][D][ratgdo:191]: Lock state=UNLOCKED
[21:34:09][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:34:10][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:34:10][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:34:10][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:34:10][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:34:10][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[... 1119 lines cut ...]
[21:36:53][D][ratgdo:191]: Lock state=UNLOCKED
[21:36:53][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:36:54][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:36:54][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:36:54][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:36:54][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:36:54][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:36:55][D][ratgdo_secplus1:227]: Received packet: [3A 5C]
[21:36:55][D][ratgdo:185]: Light state=ON
[21:36:55][D][ratgdo:191]: Lock state=UNLOCKED
[21:36:55][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:36:55][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
[21:36:55][D][ratgdo_secplus1:189]: Ignoring byte (0): 40, baud: 1200
[21:36:55][D][ratgdo_secplus1:189]: Ignoring byte (0): 00, baud: 1200
[21:36:56][D][ratgdo_secplus1:189]: Ignoring byte (0): 53, baud: 1200
[21:36:56][D][ratgdo_secplus1:189]: Ignoring byte (0): 01, baud: 1200
[21:36:56][D][ratgdo_secplus1:227]: Received packet: [3A 5C]
[21:36:56][D][ratgdo:185]: Light state=ON
[21:36:56][D][ratgdo:191]: Lock state=UNLOCKED
[21:36:56][D][ratgdo_secplus1:227]: Received packet: [39 00]
[21:36:56][W][ratgdo_secplus1:216]: Discard incomplete packet, length: 1
```

</p>
</details>

outgeeked, 2024-01-12:
I just tested this on my own garage hardware, and unfortunately no dice. Logs attached.

[ratgdov25i-c94232_logs.txt](https://github.com/ratgdo/esphome-ratgdo/files/13914882/ratgdov25i-c94232_logs.txt)

mariusmuja, 2024-01-12:
@taylorcoffelt 

I added obstruction from status support, you need to set the ` input_obst_pin: "0"` if no obstruction sensor is connected for obstruction from status to be used. Let me know if it works as expected.

I couldn't figure out what the motion command is, it might be in the incomplete packets that are discarded and not printed. I updated the code to print those too and also added millisecond timestamps to the logs to better make sense of things. When you get a chance, can you get more logs of motion, especially of transition from no motion/light off to motion and then back to no motion.

Regarding the door status: you wall panel doesn't seem to query for door status, or it's using a different query message that we don't know how to interpret. That's why the door status is unknown and only door toggle works (we can't issue a "toggle door" command to close the door unless we know the door status is open, and we don't know how to issue a "close door" command in sec+1.0). More logs of door opening/closing with the new debug info would also help.

mariusmuja, 2024-01-12:
> I just tested this on my own garage hardware, and unfortunately no dice. Logs attached.

It seems like your garage opener is not switching from analog mode to digital mode (or we don't know how to make it do so). I recently got my hands on a free GDO with a red learn button thinking I could use it for testing and it has the same issue. We'll add dry contact support soon as well.

PaulWieland, 2024-01-12:
There are non myQ digital wall panels that use serial communication but for whatever reason, they request `0x37` instead of `0x38`. When the GDO receives a `0x37` query, it does not report its status back. I wanted to test what would happen if ratgdo injected a `0x38` immediately after the response from `0x37` was received, but it requires having one of these wall panels, or writing an emulator for one.

So in short there are three scenarios for sec + 1 GDOs:

* You have a digital "myQ" panel model# 889LM or equivalent which requests door status from the GDO
  * ratgdo listens to the sec + 1 serial protocol
* You have a digital panel which requests `0x37` from the door status
  * no door status available, either ditch the wall panel or use dry contact control mode (add reed switches for door status)
* You have an analog panel which doesn't use serial communication (78LM or door bell momentary button)
  *  ratgdo goes into sec + 1 emulation mode

PaulWieland, 2024-01-12:
Instructions for anyone wishing to test the ESP Home firmware who is currently running MQTT firmware can do an OTA update as follows (using the esphome cli on your computer, there might be other ways to compile).

With esphome cli installed:

* Save the [yaml file](https://raw.githubusercontent.com/ratgdo/esphome-ratgdo/sec_plus_v1/static/v25iboard_secplusv1.yaml) to your computer 
* Edit the file and set your wifi credentials under the "wifi" section as follows:
  *  delete `ap`
  * add `ssid: YOUR_SSID`
  * add `password: YOUR_PASSWORD`
  * save
 * run `esphome run v25iboard_secplusv1.yaml` to compile
 
Once compilation finishes, you will see something like this:
[image]

Press `ctrl+c` to break (exit) esphome. 
Now we need to find the resulting bin file which was compiled. In my case it was built in the current working directory under `.esphome/build/ratgdov25i/.pioenvs/ratgdov25i/firmware.bin`. In windows this might be different, but the the log output from esphome should give you some clue as to where it was built.

Navigate to the ratgdo mqtt web UI and upload the `firmware.bin` file you just compiled:
[image]

After uploading you will see: `Update Success! Rebooting...`
Wait a few minutes for the reboot, and then you will be presented with the ratgdo ESP Home web UI.

To revert, you can download the mqtt firmware .bin file from the [MQTT releases page](https://github.com/ratgdo/mqtt-ratgdo/releases) and upload it from ESP Home's OTA form: 
[image]

PaulWieland, 2024-01-12:
> I couldn't figure out what the motion command is, it might be in the incomplete packets that are discarded and not printed.

Sec + 1 protocol does not support motion detection. The wall panel just turns the light on.

niemyjski, 2024-01-12:
I stand corrected 2.5vs2.5i does this change anything with this config?Sent from my iPhone

agelwarg, 2024-01-12:
@mariusmuja and @PaulWieland -- since this work is still taking place on the `sec_plus_v1` branch, don't you need to update the package_import_url from `github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main` to `github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@sec_plus_v1` ?

mariusmuja, 2024-01-12:
> @mariusmuja and @PaulWieland -- since this work is still taking place on the `sec_plus_v1` branch, don't you need to update the package_import_url from `github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main` to `github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@sec_plus_v1` for the time being?

Updated.

mariusmuja, 2024-01-12:
@taylorcoffelt 

I pushed an update that injects `0x38` requests after `0x37` requests. Give the latest a try and let me know if it fixes the door status issue for you.

taylorcoffelt, 2024-01-12:
@mariusmuja Yep, it's reporting the status now, logs below.

I'll see about getting the motion captures later today.

<details><summary>Details</summary>
<p>

```
[10:39:03][W][ratgdo_secplus1:213]: [546138] Discard incomplete packet, [38 ...]
[10:39:03][D][ratgdo_secplus1:177]: [546255] Ignoring byte [40], baud: 1200
[10:39:03][D][ratgdo_secplus1:177]: [546275] Ignoring byte [00], baud: 1200
[10:39:03][D][ratgdo_secplus1:177]: [546523] Ignoring byte [53], baud: 1200
[10:39:03][D][ratgdo_secplus1:177]: [546526] Ignoring byte [01], baud: 1200
[10:39:03][D][ratgdo_secplus1:224]: [546790] Received packet: [3A 58]
[10:39:03][D][ratgdo:185]: Light state=OFF
[10:39:03][D][ratgdo:191]: Lock state=UNLOCKED
[10:39:04][D][ratgdo_secplus1:224]: [547056] Received packet: [39 00]
[10:39:04][D][ratgdo_secplus1:188]: [547305] Received command: [37]
[10:39:04][W][ratgdo_secplus1:213]: [547449] Discard incomplete packet, [38 ...]
[10:39:04][D][ratgdo_secplus1:177]: [547566] Ignoring byte [40], baud: 1200
[10:39:04][D][ratgdo_secplus1:177]: [547569] Ignoring byte [00], baud: 1200
[10:39:04][D][ratgdo_secplus1:177]: [547816] Ignoring byte [53], baud: 1200
[10:39:04][D][ratgdo_secplus1:177]: [547836] Ignoring byte [01], baud: 1200
[10:39:05][D][ratgdo_secplus1:224]: [548085] Received packet: [3A 58]
[10:39:05][D][ratgdo:185]: Light state=OFF
[10:39:05][D][ratgdo:191]: Lock state=UNLOCKED
[10:39:05][D][ratgdo_secplus1:224]: [548352] Received packet: [39 00]
[10:39:05][D][ratgdo_secplus1:188]: [548601] Received command: [37]
[10:39:05][W][ratgdo_secplus1:213]: [548745] Discard incomplete packet, [38 ...]
[10:39:05][D][ratgdo_secplus1:177]: [548863] Ignoring byte [40], baud: 1200
[10:39:05][D][ratgdo_secplus1:177]: [548882] Ignoring byte [00], baud: 1200
[10:39:06][D][ratgdo_secplus1:177]: [549130] Ignoring byte [53], baud: 1200
[10:39:06][D][ratgdo_secplus1:177]: [549133] Ignoring byte [01], baud: 1200
[10:39:06][D][ratgdo_secplus1:224]: [549398] Received packet: [3A 58]
[10:39:06][D][ratgdo:185]: Light state=OFF
[10:39:06][D][ratgdo:191]: Lock state=UNLOCKED
[10:39:06][D][ratgdo_secplus1:224]: [549665] Received packet: [39 00]
[10:39:06][D][ratgdo_secplus1:188]: [549905] Received command: [37]
[10:39:07][W][ratgdo_secplus1:213]: [550049] Discard incomplete packet, [38 ...]
[10:39:07][D][ratgdo_secplus1:177]: [550166] Ignoring byte [40], baud: 1200
[10:39:07][D][ratgdo_secplus1:177]: [550186] Ignoring byte [00], baud: 1200
[10:39:07][D][ratgdo_secplus1:177]: [550435] Ignoring byte [53], baud: 1200
[10:39:07][D][ratgdo_secplus1:177]: [550437] Ignoring byte [01], baud: 1200
[10:39:07][D][ratgdo_secplus1:224]: [550702] Received packet: [3A 58]
[10:39:07][D][ratgdo:185]: Light state=OFF
[10:39:07][D][ratgdo:191]: Lock state=UNLOCKED
[10:39:08][D][ratgdo_secplus1:224]: [550969] Received packet: [39 00]
[10:39:08][D][ratgdo_secplus1:188]: [551219] Received command: [37]
[10:39:08][W][ratgdo_secplus1:213]: [551363] Discard incomplete packet, [38 ...]
[10:39:08][D][ratgdo_secplus1:177]: [551480] Ignoring byte [40], baud: 1200
[10:39:08][D][ratgdo_secplus1:177]: [551483] Ignoring byte [00], baud: 1200
[10:39:08][D][ratgdo_secplus1:177]: [551731] Ignoring byte [53], baud: 1200
[10:39:08][D][ratgdo_secplus1:177]: [551750] Ignoring byte [01], baud: 1200
[10:39:09][D][ratgdo_secplus1:224]: [551999] Received packet: [3A 58]
[10:39:09][D][ratgdo:185]: Light state=OFF
[10:39:09][D][ratgdo:191]: Lock state=UNLOCKED
[10:39:09][D][ratgdo_secplus1:224]: [552267] Received packet: [39 00]
[10:39:09][D][ratgdo_secplus1:188]: [552515] Received command: [37]
[10:39:09][W][ratgdo_secplus1:213]: [552659] Discard incomplete packet, [38 ...]
[10:39:09][D][ratgdo_secplus1:177]: [552777] Ignoring byte [40], baud: 1200
[10:39:09][D][ratgdo_secplus1:177]: [552779] Ignoring byte [00], baud: 1200
[10:39:10][D][ratgdo_secplus1:177]: [553043] Ignoring byte [53], baud: 1200
[10:39:10][D][ratgdo_secplus1:177]: [553046] Ignoring byte [01], baud: 1200
[10:39:10][D][ratgdo_secplus1:224]: [553311] Received packet: [3A 58]
[10:39:10][D][ratgdo:185]: Light state=OFF
[10:39:10][D][ratgdo:191]: Lock state=UNLOCKED
[10:39:10][D][ratgdo_secplus1:224]: [553563] Received packet: [39 00]
[10:39:10][D][ratgdo_secplus1:188]: [553811] Received command: [37]
[10:39:10][W][ratgdo_secplus1:213]: [553955] Discard incomplete packet, [38 ...]
[10:39:11][D][ratgdo_secplus1:177]: [554073] Ignoring byte [40], baud: 1200
[... 547 lines cut ...]
[10:40:24][D][ratgdo:282]: [626234] Position update: -inf
[10:40:24][D][ratgdo_secplus1:177]: [626242] Ignoring byte [53], baud: 1200
[10:40:24][D][ratgdo_secplus1:177]: [626253] Ignoring byte [01], baud: 1200
[10:40:24][D][ratgdo_secplus1:224]: [626515] Received packet: [3A 5C]
[10:40:24][D][ratgdo:185]: Light state=ON
[10:40:24][D][ratgdo:191]: Lock state=UNLOCKED
[10:40:24][D][ratgdo:282]: [626736] Position update: -inf
[10:40:24][D][ratgdo_secplus1:224]: [626779] Received packet: [39 00]
[10:40:24][D][ratgdo_secplus1:188]: [627025] Received command: [37]
[10:40:24][W][ratgdo_secplus1:213]: [627168] Discard incomplete packet, [38 ...]
[10:40:24][D][ratgdo:282]: [627245] Position update: -inf
[10:40:24][D][ratgdo_secplus1:177]: [627287] Ignoring byte [40], baud: 1200
[10:40:24][D][ratgdo_secplus1:177]: [627296] Ignoring byte [00], baud: 1200
[10:40:24][D][ratgdo_secplus1:177]: [627544] Ignoring byte [53], baud: 1200
[10:40:24][D][ratgdo_secplus1:177]: [627563] Ignoring byte [01], baud: 1200
[10:40:24][D][ratgdo:282]: [627747] Position update: -inf
[10:40:24][D][ratgdo_secplus1:224]: [627817] Received packet: [3A 5C]
[10:40:24][D][ratgdo:185]: Light state=ON
[10:40:24][D][ratgdo:191]: Lock state=UNLOCKED
[10:40:25][D][ratgdo_secplus1:224]: [628081] Received packet: [39 00]
[10:40:25][D][ratgdo:282]: [628251] Position update: -inf
```

</p>
</details>

taylorcoffelt, 2024-01-12:
>I added obstruction from status support, you need to set the  input_obst_pin: "0" if no obstruction sensor is connected for obstruction from status to be used. Let me know if it works as expected.

@mariusmuja looks like `input_obst_pin: "0"` conflicts with the default pin setting of `ratgdov25i_dry_contact_light`

<details><summary>Output</summary>
<p>

```
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/ratgdo.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@sec_plus_v1
INFO Updating https://github.com/ratgdo/esphome-ratgdo@sec_plus_v1
INFO Detected timezone 'America/Denver'
Failed config

ratgdo: [source <unicode string>:13]
  
  Pin 0 is used in multiple places.
  - id: ratgdov25i
    input_gdo_pin: 
      number: 4
      mode: 
        input: True
        output: False
        open_drain: False
        pullup: False
        pulldown: False
        analog: False
      inverted: False
    output_gdo_pin: 
      number: 5
      mode: 
        output: True
        input: False
        open_drain: False
        pullup: False
        pulldown: False
        analog: False
      inverted: False
    input_obst_pin: 
      number: 0
      mode: 
        input: True
        output: False
        open_drain: False
        pullup: False
        pulldown: False
        analog: False
      inverted: False
    protocol: secplusv1
    on_sync_failed: 
      - then: 
          - homeassistant.service: 
              service: persistent_notification.create
              data: 
                title: ratgdov2.5i sync failed
                message: Failed to communicate with garage opener on startup.
                notification_id: esphome_ratgdo_ratgdov25i_sync_failed
              data_template: {}
              variables: {}
binary_sensor.gpio: [source <unicode string>:130]
  
  Pin 0 is used in multiple places.
  platform: gpio
  id: ratgdov25i_dry_contact_light
  pin: 
    number: 0
    inverted: True
    mode: 
      input: True
      pullup: True
      output: False
      open_drain: False
      pulldown: False
      analog: False
  name: Dry contact light
  entity_category: diagnostic
  filters: 
    - delayed_on_off: 500ms
  on_press: 
    - then: 
        - light.toggle: 
            id: ratgdov25i_light
  disabled_by_default: False

```

</p>
</details>

mariusmuja, 2024-01-12:
> Yep, it's reporting the status now, logs below.

Nice! It seems like your GDO is replying to `0x38` requests, but when the door is closed it only throttles the replies. When the door is opening/open/closing it's replying to every request. Would be interesting to see a log where you stopped the door in the middle to see what happens then.

mariusmuja, 2024-01-12:
> looks like `input_obst_pin: "0"` conflicts with the default pin setting of `ratgdov25i_dry_contact_light`

I'll push a fix soon.

kalbasit, 2024-01-12:
Excuse my noobiness, I'm very new to this. Looking at my garage opener, it has a purple learn button but it has this very simple open/close button. So it sec1? Is that button "dry contact"? Do I need to change any of the yaml to test?
[image]
[image]

[79 more replies]

closed 2024-01-20 by bdraco