ratgdo/esphome-ratgdo #65: Brief obstructions are not being sensed

J-Pipe, 2023-10-18:
Not sure if this is a ratgdo issue or esphome, but if I quickly walk through the sensor, it does not get triggered. Then my chamberlain motor light turns on but that also does not register in Home Assistant. If I stand in front on the sensor for about a second, everything works normally. Not sure if there is a debounce filter that makes it not register the brief obstructions?

zd3sf, 2025-01-27:
Same problem here. I am trying to use the obstruction sensor in my alarm, but it only registers obstruction after 3-4 seconds. I am not sure where the debounce filter is....it is not in the YAML.

J-Pipe, 2025-01-27:
FYI: I never found a solution

zd3sf, 2025-01-27:
Code responsible. I'll monitor logs if the controller sees the obstruction but delays its reporting

```
/*************************** OBSTRUCTION DETECTION ***************************/

    void RATGDOComponent::obstruction_loop()
    {
        long current_millis = millis();
        static unsigned long last_millis = 0;
        static unsigned long last_asleep = 0;

        // the obstruction sensor has 3 states: clear (HIGH with LOW pulse every 7ms), obstructed (HIGH), asleep (LOW)
        // the transitions between awake and asleep are tricky because the voltage drops slowly when falling asleep
        // and is high without pulses when waking up

        // If at least 3 low pulses are counted within 50ms, the door is awake, not obstructed and we don't have to check anything else

        const long CHECK_PERIOD = 50;
        const long PULSES_LOWER_LIMIT = 3;

        if (current_millis - last_millis > CHECK_PERIOD) {
            // ESP_LOGD(TAG, "%ld: Obstruction count: %d, expected: %d, since asleep: %ld",
            //     current_millis, this->isr_store_.obstruction_low_count, PULSES_EXPECTED,
            //     current_millis - last_asleep
            // );

            // check to see if we got more then PULSES_LOWER_LIMIT pulses
            if (this->isr_store_.obstruction_low_count > PULSES_LOWER_LIMIT) {
                this->obstruction_state = ObstructionState::CLEAR;
                this->obstruction_sensor_detected_ = true;
            } else if (this->isr_store_.obstruction_low_count == 0) {
                // if there have been no pulses the line is steady high or low
                if (!this->input_obst_pin_->digital_read()) {
                    // asleep
                    last_asleep = current_millis;
                } else {
                    // if the line is high and was last asleep more than 700ms ago, then there is an obstruction present
                    if (current_millis - last_asleep > 700) {
                        this->obstruction_state = ObstructionState::OBSTRUCTED;
                    }
                }
            }
            last_millis = current_millis;
            this->isr_store_.obstruction_low_count = 0;
        }
    }
```

PaulWieland, 2025-01-27:
The ESPHome obstruction detection code is requiring [700ms](https://github.com/ratgdo/esphome-ratgdo/blob/02799643c9e922b9a540803cd11b8b2dabee0c0a/components/ratgdo/ratgdo.cpp#L376) of samples before reporting an obstruction. [In my original implementation](https://github.com/ratgdo/mqtt-ratgdo/blob/5231b3b97e3e5072485824c4439e6d4c73ec3ddd/src/ratgdo.cpp#L430), I was only checking for 70ms.

We can try lowering this value to increase the response time, but it runs the risk of creating false positives.

zd3sf, 2025-01-27:
Hi Paul, I don't think we should mess with detection code, the problem is that in the front end, it takes 3-4 seconds of continuous obstruction to register. Any less than that, it wouldnt even show up. About 9 months ago, it used to be instantaneous, not sure what happened. I tried change write to flash intervals but no effect there

zd3sf, 2025-01-28:
I just tested again last night the issue was gone. I'm not sure how. When the delay was observed, I had just powered on the GDO and the ratGDO, and I kept testing for few hours after, but 12 hours later, fast response like normal.