ratgdo/esphome-ratgdo #67: What is the motion sensor for?

J-Pipe, 2023-10-19:
Don't see any documentation for the motion sensor. The ratgdo project doesn't look like it has a motion sensor. Not sure where this came from. Not seeing the value ever change.

YBonline, 2023-10-19:
> Don't see any documentation for the motion sensor. The ratgdo project doesn't look like it has a motion sensor. Not sure where this came from. Not seeing the value ever change.

This is the status of the motion sensor that is located on some of the wall buttons. If your wall button doesn't have a motion sensor, it will not ever change.

closed 2023-10-19 by J-Pipe

mkaatman, 2024-01-01:
Is the motion sensor only in the esphome version or does mqtt 2.1 have it as well?

restauffer, 2024-01-01:
I an using MQTT with a SEC +1 gdo and I do not see the motion detector exposed as a topic in MQTT. It seems the ESP version has more features but  I can't use it with a SEC +1 GDO.

PaulWieland, 2024-01-01:
Motion detection isn’t possible with sec + 1.0.