ratgdo/esphome-ratgdo #70: Update to v2.26a

gullygossner, 2023-10-21:
Is there any way to update the ratdgo firmware to v2.26a within the esphome firmware? 

My opener isn’t working correctly with v2.1 and v2.26a could solve the issues.

fatbob01, 2023-10-23:
> Is there any way to update the ratdgo firmware to v2.26a within the esphome firmware?
> 
> My opener isn’t working correctly with v2.1 and v2.26a could solve the issues.

Seems to be available here: https://paulwieland.github.io/ratgdo/flash.html

gullygossner, 2023-10-23:
The linked firmware is what I ended up using, I was hoping to use the
esphome port. I’m probably missing something but I don’t see the ability to
use v2.26a with that port.

PaulWieland, 2023-10-25:
ESPHome will be updated as soon as we get all the kinks worked out of of security + 1.0. It's 90% there but there is still one weird behavior I am chasing where the door stops responding when used with a certain wall panel. I just received that wall panel to test with so hopefully I can figure it out soon.

rlowens, 2023-11-15:
> Also, it may be worth noting somewhere in the documentation that this version doesn't support 1.0 yet

https://paulwieland.github.io/ratgdo/01_features.html

miles267, 2023-12-09:
So glad I found this thread.  Have an older (2014) Liftmaster/Chamberlain GDO with PURPLE learn button.  Doesn't work with my new ratgdo 2.5i units.  Have been jumping thru hoops.  Wiring.  Rewiring.  Flashing ESPHome firmware.  Reflashing firmware.  No success.  Seems it's due to this current limitation here.  Am excited that security 1.0 support is coming.

PaulWieland, 2023-12-09:
[image]

miles267, 2023-12-09:
Thank you @PaulWieland. Sorry for my oversight.  I've since integrated my Security 1.0 Liftmaster garage door successfully with MQTT so the firmware is working great.  This confirms my wiring was correct all along.  Are there any ESPHome beta firmware available for testing yet that support Security 1.0 for the older GDOs?

4rsfed, 2023-12-14:
Any ETA on support for 1.0?

Do you need beta testers?

bdraco, 2023-12-16:
duplicate of https://github.com/ratgdo/esphome-ratgdo/issues/62

closed 2023-12-16 by bdraco