ratgdo/esphome-ratgdo #73: ESP32C3

donavanbecker, 2023-10-24:
Anyone able to help me with knowing which pins to use and what to plug into for this to work with this board?

mariusmuja, 2023-10-24:
Note, the boards from the link you sent don't seem to use ESP32C3, but ESP32S.

Do you have a [ratgdo](https://paulwieland.github.io/ratgdo/#order) board from Paul? If so, you can see which pins to use [here](https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v2board_esp32_d1_mini.yaml) or [here](https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25board_esp32_d1_mini.yaml), depending if you have board version (2 or 2.5).

If you don't have a ratgdo board, you can use the schematic [here](https://github.com/mariusmuja/esphome-ratgdo) to wire it up.

donavanbecker, 2023-10-24:
Thanks! I did buy one of his boards but also have some of these boards laying around so figured I would try them.

donavanbecker, 2023-10-24:
So my board has usb-c so i can use that to power it so i should be able to leave that part of it out on the schema?

mariusmuja, 2023-10-24:
Yes, you can use usb-c to power the board, but make sure you connect the board GND to the garage opener GND (white ports).

donavanbecker, 2023-10-24:
Thanks @mariusmuja, might have more questions. Going to leave this open until i have this hooked up.

donavanbecker, 2023-10-25:
@mariusmuja, I am guessing I need to get relays also?

[Relay](https://www.amazon.com/CHENBO-Channel-Module-Shield-Arduino/dp/B07874KSLY/ref=sr_1_3?crid=3BYOM8BGR0IFW&keywords=KY-019+5v+Relay&qid=1698203540&sprefix=%2Caps%2C270&sr=8-3)

And this would be the relay?
[image]

mariusmuja, 2023-10-25:
No, that's a MOSFET with Vgs > 2V, a 2n7000 should work, but lots of others too. You'll need two of them.

donavanbecker, 2023-10-25:
[2n7000](https://www.amazon.com/s?k=2n7000+N-channel+MOSFETs&crid=6YXZ6UYSBVZK&sprefix=2n7000+n-channel+mosfets%2Caps%2C309&ref=nb_sb_noss_2)?

donavanbecker, 2023-10-25:
and do I need this also? [10Kohm (kilo-ohm) resistors](https://www.amazon.com/Resistor-Tolerance-Resistors-Limiting-Certificated/dp/B08QRJZ82J/ref=sr_1_1_sspa?crid=22X1BTEHUITGM&keywords=10Kohm+%28kilo-ohm%29+resistors&qid=1698205342&sprefix=10kohm+kilo-ohm+resistors%2Caps%2C232&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1)

mariusmuja, 2023-10-25:
> [2n7000](https://www.amazon.com/2N7000-Mosfet-Transistor-N-Channel-Effect/dp/B0BJQ92VNT/ref=sr_1_6?crid=6YXZ6UYSBVZK&keywords=2n7000+N-channel+MOSFETs&qid=1698205477&sprefix=2n7000+n-channel+mosfets%2Caps%2C309&sr=8-6)?

Yes, those should work.

mariusmuja, 2023-10-25:
> and do I need this also? [10Kohm (kilo-ohm) [...]

There are no 10K resistors on the schematic.

closed 2023-10-25 by donavanbecker