ratgdo/esphome-ratgdo #78: ESP32 on the new ratgdo 2.5 boards?

markmghali, 2023-10-25:
I ordered a Ratgdo 2.0 board but I guess I am going to get the 2.5 board now since mine has not shipped yet. Is the 2.5 board compatible with the esp32 d1 mini as it was with the 2.0 board?

bdraco, 2023-10-25:
Yes, but we don't have the images on the website to be flashed for the sole reason I don't have an image to put up there of the v2.5 board with the esp32 in it.  If you send one I can add it to the website.

You'll have to do the YAML manually for now.

markmghali, 2023-10-25:
Ok cool! I don't have my Ratgdo yet just wanted to verify it works still. 

Thank you

4rsfed, 2023-11-07:
[image]

Just arrived today. How can I help get this on the web flasher?

bdraco, 2023-11-07:
I'll get it put up.  Can you take a better picture so the ESP32 wording is readable?

rlowens, 2023-11-07:
I think you also need to solder the pin headers on. Might want to do that before taking the picture so later users aren't confused.

bdraco, 2023-11-07:
> I think you also need to solder the pin headers on. Might want to do that before taking the picture so later users aren't confused.

That would be nice, but happy to take it without it and update it later as well as long as the ESP32 is readable

markmghali, 2023-11-07:
Did you have to cut the outside rows of pins to make it fit?

4rsfed, 2023-11-07:
[image]

4rsfed, 2023-11-07:
> Did you have to cut the outside rows of pins to make it fit?
> […](#)

No

bdraco, 2023-11-07:
That looks like an ESP8266?

4rsfed, 2023-11-07:
Correct, looks like I didn't catch that this ticket is marked ESP32

PR #101 (2023-11-07): Add v2.5 board esp32
I didn't get the v2.5 boards yet, but I did an absolutely stunning 🤦 job with gimp to make an image so we could get it on the flasher.

closes #78
closed 2023-11-07 by bdraco