ratgdo/esphome-ratgdo #80: Request: "Door Status" on sooner while opening rather than open?

alexruffell, 2023-10-26:
I use the door status output to turn on parking lasers that help with parking the cars in the right spot. I often enter the garage before the door is even fully up so the laser turns on a bit too late. It would be nice if the door status changed sooner as soon as the door starts opening. If you think about it, whether the door is partially or fully open, it is open and someone can enter. If there are use cases that would require that status to be high only when the door is fully open, then it would be nice for this to be configurable.

mariusmuja, 2023-10-26:
Can you give a little more detail on what door status you're using? Is it the state of the door cover entity in home assistant? Something else?

alexruffell, 2023-10-26:
On the ratdgo board there are two status outputs that are high when the door is open and when an obstruction is detected. In the base.yaml file, this is where they are defined:

```
  - platform: gpio
    id: "${id_prefix}_status_door"
    internal: true
    pin:
      number: ${status_door_pin}  # D0 output door status, HIGH for open, LOW for closed
      mode:
        output: true
    name: "Status door"
    entity_category: diagnostic
  - platform: gpio
    id: "${id_prefix}_status_obstruction"
    internal: true
    pin:
      number: ${status_obstruction_pin}  # D8 output for obstruction status, HIGH for obstructed, LOW for clear
      mode:
        output: true
    name: "Status obstruction"
    entity_category: diagnostic
```

And here is where the status is changed based on the door being open/closed:

```
cover:
  - platform: ratgdo
    id: ${id_prefix}_garage_door
    device_class: garage
    name: " "
    ratgdo_id: ${id_prefix}
    on_closed:
      - switch.turn_off: ${id_prefix}_status_door
    on_open:
      - switch.turn_on: ${id_prefix}_status_door
```

If only I could use "on_opening" instead of "on_open" I would be set... but I don't believe that option exists at all in ESPHome even though those are existing states.

mariusmuja, 2023-10-26:
Ahh, right! I'm using a trimmed down version of `base.yam`l and I forgot about those entities. 

Yes, I agree `on_opening` and `on_closing` triggers could be useful, we can add those.

alexruffell, 2023-10-28:
@mariusmuja Thank you for adding it, and so quickly too!

audiato, 2023-10-31:
Just wanted to check if this has been merged yet? Mine doesn't show an "opening" status at all at the moment - seems to go straight from closed to open.

PR #81 (2023-11-05): Adding cover on_opening and on_closing triggers
This adds `on_opening` and `on_closing` triggers for the cover component, fixes #80.
closed 2023-11-05 by bdraco