ratgdo/esphome-ratgdo #82: How do you disable web_server?

shred86, 2023-10-28:
Apologize for probably a basic question but I'm new to ESPHome and after a couple hours of trying to figure this out, I'm hoping to get some help.

I purchased a ratgdo v2.0 and flashed it via the esp web tool, set it up in HA and everything works great. However, I would like to disable the web_server for ratgdo. I assumed I would need to be able to customize the config for the ratgdo so I installed ESPHome on my computer (Mac mini) using brew so I have access to the ESPHome dashboard. I can see my ratgdo device was automatically discovered so I clicked adopt and now it's adopted to the ESPHome dashboard.

However, looking at the config file it created, it looks like it's just pulling in the default config from this repo:

````yaml
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v2board_esp8266_d1_mini_lite.yaml@main
````

My question is, how would I go about disabling `web_server:` which is enabled by default in the config it's pulling from?

Appreciate any pointers!

shred86, 2023-10-29:
So I couldn't find a way to ignore the `web_sever` entry in the v2board_esp8266_d1_mini_lite.yaml file, so what I did was copy the contents of that file into my .yaml file for ratgdo from the ESPHome dashboard, made a few modifications and installed that.

Long explanation for anyone that runs into this:

When you adopt the ratgdo device from the ESPHome dashboard, it will create a .yaml file that looks something like this:

````yaml
substitutions:
  name: ratgdov2-c84eae
  friendly_name: ratgdov2 c84eae
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v2board_esp8266_d1_mini_lite.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: # redacted

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
````

As you can see, it's importing the v2board_esp8266_d1_mini_lite.yaml file (which is the one that has `web_server:` enabled) and I believe overriding any settings you have set in this .yaml file created for your ratgdo in the ESPHome dashboard (the  .yaml above). 

I just copied the contents of the v2board_esp8266_d1_mini_lite.yaml into the .yaml file created in the ESPHome dashboard, and made the following changes:
- Removed `web_server:`
- Add the API key that was generated when adopting ratgdo from the ESPHome dashboard
- Added my wifi information
- I had to add `use_address:` with my ratgdo static IP address to get this to work - for some reason it wouldn't resolve using the ratgdo hostname anymore.

````yaml
---
substitutions:
  id_prefix: ratgdov2
  friendly_name: ratgdov2
  uart_tx_pin: D4
  uart_rx_pin: D2
  input_obst_pin: D7
  status_door_pin: D0
  status_obstruction_pin: D8
  dry_contact_open_pin: D5
  dry_contact_close_pin: D6
  dry_contact_light_pin: D3

esphome:
  name: ${id_prefix}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true

esp8266:
  board: d1_mini_lite
  restore_from_flash: true

packages:
  remote_package:
    url: https://github.com/ratgdo/esphome-ratgdo
    files: [base.yaml]
    refresh: 1s

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

api:
  encryption:
    key: # redacted

ota:

improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: # redacted

logger:
````

You will have to reconfigure the device from Home Assistant to use the new API key. Otherwise, everything should still work like before from HA.

corbanmailloux, 2024-01-22:
As @crkochan noted, this was a feature request on ESPHome.
It's now been released: https://github.com/esphome/esphome/pull/5821

So you can simply add this to your config:
```yaml
web_server: !remove
```

I've tested this with my own devices because I was hitting an issue where the OTA process would fail due to the size of the binary. Removing the webserver reduces the binary size enough to allow the OTA to work.

crkochan, 2025-07-22:
With recent changes (#454), this is now broken.

```
Failed config

ota.web_server: [source /data/packages/74f605c3/base.yaml:60]
  
  Component ota.web_server requires component web_server_base.
  platform: web_server
  id: ratgdov2_ratgdo_ota_ws
```

I have not yet sussed out a solution as I don't want to remove the whole of the `ota:` component.

PaulWieland, 2025-07-22:
Thanks for that, I will add a comment to the yaml file for those who want to disable the WS