ratgdo/esphome-ratgdo #85: [Request] Battery status

YBonline, 2023-11-01:
The MyQ app and the wall panel on my garage door show the status of the battery (whether its charged, discharged, or needs replacement). It would be nice of RatGDO made this information available as well.

PaulWieland, 2023-11-02:
This is a duplicate of https://github.com/PaulWieland/ratgdo/issues/131

mariusmuja, 2023-11-02:
> The MyQ app and the wall panel on my garage door show the status of the battery (whether its charged, discharged, or needs replacement). It would be nice of RatGDO made this information available as well.

This could be accomplished, but we'll need to figure out the meaning of the data fields in the battery status messages (what is charged/discharged,...)

thayerfox, 2024-01-03:
Just chiming in to add my desire for this feature request, thanks.

ptmuldoon, 2024-02-09:
I just got my ratgdo yesterday.  Install was a breeze.  But I also would like to be able to see the battery status.   In the MyQ app, it now shows the battery as disconnected, which is incorrect as I did unplug the opener and could still open and close the door via the battery.

monstruwaken, 2024-02-28:
I would also like this so i can fully get rid of myq

rct, 2025-02-14:
> This could be accomplished, but we'll need to figure out the meaning of the data fields in the battery status messages (what is charged/discharged,...)

Do you need people to collect data to help reverse engineer this?  

One of my two GDOs is beeping every 30 seconds. I think it is signaling time to replace the battery, though the MyQ app still shows both batteries and fully charged and doesn't show anything under diagnostics yet.

timtIAD, 2025-06-02:
I noticed this was marked "closed".

I am still not seeing this as a capability. Am I missing something?

NukeThemTillTheyGlow, 2026-08-18:
Add my 2 cents, I recently had a 3 day power outage while out of town.  Came home expecting 2 of my 3 garage doors that have battery backup (all 3 using RATGDO's), but only one worked (after starting up network/wifi).  Reason, dead battery.  then the next day after I think 2 cycles, the one that worked died.    Not the end of the world, but I'd love to set up an automation in HA OS to track the status and alert me ahead of time.

inventor7777, 2026-10-13:
I would also appreciate this being added,

I can see that it seems to decode the information fine, but no HA sensors.

```
ratgdo-garage-door 2026-10-12T21:28:03-05:00 ratgdo_secplus2: [D][ratgdo_secplus2:358]: Received packet: [55:01:00:16:3A:43:37:8A:6D:F5:FB:92:28:15:5C:19:C0:F6:EB]
ratgdo-garage-door 2026-10-12T21:28:03-05:00 ratgdo: [D][ratgdo:521]: Battery state=FULL
```