ratgdo/esphome-ratgdo #87: Voice code not working with exposed HA Voice Assistant cover

amaisano, 2023-11-04:
When I used the MyQ HA integration to expose the cover entity to Alexa, I would be asked every time by my Echo devices to say the 4 digit voice code to open the garage. The first time I asked this Alexa asked me to set the code.

Through Ratgdo Esphome, Alexa ALSO asked me to set a voice code the first time I exposed the cover, **but it never asks for the code via Esphome, so anyone can voice open the garage without a voice code.**

Can we get the voice code working with this integration?

dermm, 2023-11-21:
Hi,

I have made exactly the same observation.
It is probably because it contains partial opening.
I mean the option "supported_features" is set to 15 to allow set position and Stop.

https://github.com/home-assistant/core/blob/dev/homeassistant/components/cover/__init__.py

# These SUPPORT_* constants are deprecated as of Home Assistant 2022.5.
# Please use the CoverEntityFeature enum instead.
SUPPORT_OPEN = 1
SUPPORT_CLOSE = 2
SUPPORT_SET_POSITION = 4
SUPPORT_STOP = 8
SUPPORT_OPEN_TILT = 16
SUPPORT_CLOSE_TILT = 32
SUPPORT_STOP_TILT = 64
SUPPORT_SET_TILT_POSITION = 128

amaisano, 2023-11-23:
So is this something we can adjust?

I also noticed the close command no longer beeps the garage and flashes the lights as a warning. It just instantly closes.