ratgdo/esphome-ratgdo #100: Successful flashing but not showing status or wifi option

kramttocs, 2023-11-07:
Received two 2.5 boards today.
The first one flashed fine but after that it doesn't show the device info or the connect to wifi option. It just shows this again:

[image]

The second one does and connected to wifi successfully.
I can get into the logs/console but nothing shows unless I 'Reset Device' and then I get some stuff with one of the last messages being 'Boot loop detected. Proceeding to safe mode'.

I've tried flashing again (selecting Erase like I did the first time) but same behavior.

[image]

goldenjeevan, 2024-05-09:
I have the same issue, but see no resolution to this...

kramttocs, 2024-05-09:
I had to get a replacement unit