ratgdo/esphome-ratgdo #99: Decoding sec+1 door status

PaulWieland, 2023-11-07:
I've made the following data captures and mapped the various byte values to physical door states. Some of these door state values have been provided by users.

There must be a way of interpreting the individual bit values.

Here's what I have so far:

| "normal" state codes      | hex  | binary   |
|---------------------------|------|----------|
| Open                      | 0x52 | 1010010  |
| Closed                    | 0x55 | 1010101  |
| Opening                   | 0x01 | 1        |
| Closing                   | 0x04 | 100      |

| user reported state codes |      |          |                                                                                   |
|---------------------------|------|----------|-----------------------------------------------------------------------------------|
| Open?                     | 0xEA | 11101010 |                                                                                   |
| Opening?                  | 0x51 | 1010001  | Physical obstruction blocks door causing it to stop and reverse                   |
| Opening?                  | 0x52 | 1010010  |                                                                                   |
| Opening?                  | 0x80 | 10000000 | Obstruction beam is broken during close which causes the door to stop and reverse |
| Opening?                  | 0x81 | 10000001 | Obstruction beam is broken during close which causes the door to stop and reverse |
| Opening?                  | 0x41 | 1000001  |                                                                                   |
| Closing?                  | 0x44 | 1000100  |

mariusmuja, 2023-11-07:
A hypothesis: only last 3 (or 4) bits are important for door state (the other bits representing something else):

| State | binary | 
|---|---|
| open | 010 |
| closed | 101 |
| opening | 001 |
| closing | 100 |

This matches most examples, but not all... maybe some of the reported codes are incorrect?... or this hypothesis is incorrect...

PaulWieland, 2023-11-07:
I think you're onto something. The only one that doesn't fit is 0x80 (000). I wonder if 000 means stopped.

PaulWieland, 2023-11-07:
Translated back we get:

|bin|hex|state|
|---|---|----|
|001|0x1|opening
|010|0x2|open
|100|0x4|closing
|101|0x5|closed

I wonder what happened to 3 :)

mariusmuja, 2023-11-07:
Maybe it's 3 that means stopped, not 000 ?

PaulWieland, 2023-11-07:
Tested and that mapping seems to work, including the momentary stopped state when blocking the door. 
https://github.com/ratgdo/mqtt-ratgdo/blob/574d716119eeb452410c07967caf348aaccf4124/src/static_code.cpp#L26

mariusmuja, 2023-11-07:
How about when stopping the door partially opened? (not by blocking the door)

PaulWieland, 2023-11-07:
Yes good point. Thats 110 / 0x6.

apaperclip, 2023-11-10:
Hey all, here is some additional info from my hw 2.5, sw 2.5, sec+ 1.0 setup.  some of it jives with the above, some doesn't so I wanted to share

I gathered the current state when an error occurred for the past couple days

| door state| unknown code | binary from log|
|--------|--------|--------|
| open| EA| 11101010|
| open| 5E| 1011110|
| closed | 95| 10010101|
| closed | 45| 1000101|
| closed | 95| 10010101| 
| closed | 75| 1110101|
| closed | D5| 11010101|
| closed | 5D| 1011101| 
| closed | 58| 1010101|
| open * but see note below | 3A| 111010|
| closed | 57| 1010111| 
| closed | 54| 1010100|
| closed | 58| 1010101|
| closed | 95| 10010101| 
| closed | 0| 0| 
| open | 12| 10010| 
| open * see note below| D2| 11010010| 

 3A logs - happened right after a close was issued
```
[2023-11-10 01:55:56] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 01:55:57] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 01:55:58] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 01:55:59] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 01:56:00] 3A5C 3900 QUEUE MSG ARRIVED [ratgdo/garage/command/door] 
[2023-11-10 01:56:01] close
[2023-11-10 01:56:01] MQTT: close the door
[2023-11-10 01:56:01] 3A5C 3031 383A  | door state: 3A 111010
[2023-11-10 01:56:02] Door state unknown
[2023-11-10 01:56:02] QUEUE MSG SENT [ratgdo/garage/status/door] 
[2023-11-10 01:56:02] unknown
[2023-11-10 01:56:02] 393A 3804  | door state: 4 100
[2023-11-10 01:56:03] Door state closing
[2023-11-10 01:56:03] QUEUE MSG SENT [ratgdo/garage/status/door] 
[2023-11-10 01:56:03] closing
[2023-11-10 01:56:03] 3A5C 3900 3A5C 3804  | door state: 4 100
[2023-11-10 01:56:04] 3A5C 3900 3A5C 3804  | door state: 4 100
[2023-11-10 01:56:05] 3A5C 3900 3A5C 3804  | door state: 4 100
[2023-11-10 01:56:06] 3A5C 3900 3A5C 3804  | door state: 4 100
[2023-11-10 01:56:07] 3A5C 3900 3A5C 3804  | door state: 4 100
[2023-11-10 01:56:08] 3A5C 3900 3A5C 3804  | door state: 4 100
[2023-11-10 01:56:09] 3A5C 3900 3A5C 3804  | door state: 4 100
```

D2 logs - happened during obstruction 
```
[2023-11-10 13:30:49] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:30:50] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:30:51] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:30:52] Obstruction status obstructed
[2023-11-10 13:30:52] QUEUE MSG SENT [ratgdo/garage/status/obstruction] 
[2023-11-10 13:30:52] obstructed
[2023-11-10 13:30:52] 3A5C Obstruction status clear
[2023-11-10 13:30:52] QUEUE MSG SENT [ratgdo/garage/status/obstruction] 
[2023-11-10 13:30:52] clear
[2023-11-10 13:30:52] 3904 3A5C 38D2  | door state: D2 11010010
[2023-11-10 13:30:52] Door state unknown
[2023-11-10 13:30:52] QUEUE MSG SENT [ratgdo/garage/status/door] 
[2023-11-10 13:30:52] unknown
[2023-11-10 13:30:53] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:30:53] Door state open
[2023-11-10 13:30:53] QUEUE MSG SENT [ratgdo/garage/status/door] 
[2023-11-10 13:30:53] open
[2023-11-10 13:30:54] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:30:55] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:30:56] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:30:57] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:30:58] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:30:59] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:31:00] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:31:01] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:31:02] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:31:03] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:31:04] 3A5C 3900 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:31:05] 3A5C Obstruction status obstructed
[2023-11-10 13:31:05] QUEUE MSG SENT [ratgdo/garage/status/obstruction] 
[2023-11-10 13:31:05] obstructed
[2023-11-10 13:31:05] 3900 3A5C Obstruction status clear
[2023-11-10 13:31:05] QUEUE MSG SENT [ratgdo/garage/status/obstruction] 
[2023-11-10 13:31:05] clear
[2023-11-10 13:31:05] 3852  | door state: 52 1010010
[2023-11-10 13:31:06] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:31:07] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:31:08] 3A5C 3904 3A5C 3852  | door state: 52 1010010
[2023-11-10 13:31:09] 3A5C 3904 3A5C 3852  | door state: 52 1010010
```

I'm happy to gather any additional info you need as well as continue to report back every so often with additional occurrences.

apaperclip, 2023-11-13:
I've continued to track these but l noticed something different that may be related.  Overnight (door closed nobody in the garage) the light status changed on/off  in the same second, and then ~30m later it did it again.  Perhaps whatever is causing these unknown values landed on the light on status value?

oddly - I've had the same unknown value occur when the door was opened and closed

STGMavrick, 2023-11-15:
> I've continued to track these but l noticed something different that may be related. Overnight (door closed nobody in the garage) the light status changed on/off in the same second, and then ~30m [...]

Out of curiosity, what's the door control model number for your opener? Since you're testing sec +1.0 I'm curious if you have the same control panel I had issues with. 888LM? There were issues with a certain mfg date range causing lights to turn on/off randomly, doors to open, and/or eventually the beep of death. If not, disregard.

apaperclip, 2023-11-15:
> Out of curiosity, what's the door control model number for your opener? Since you're testing sec +1.0 I'm curious if you have the same control panel I had issues with. 888LM? There were issues with [...]

I do have the 888lm that  adds sec+ 2.0 to 1.0 wireline to the gdo.   Now that you mention I think I recall getting a new wall panel from lifttmaster years ago, but I dont recall why.   

The light on/off behavior could still be there. I've noticed that when being in the garage and motion triggers the light on., sometimes it'll flash before staying on. I assumed it was some oddity with the motion sensor or the light (which is not wired, but rf and had to be associated just like a remote).  Perhaps these status changes are just capturing that behavior that I wouldn't otherwise notice because I've never been able to log the state and I'm not in the garage to see it.

STGMavrick, 2023-11-15:
> I do have the 888lm that adds sec+ 2.0 to 1.0 wireline to the gdo. Now that you mention I think I recall getting a new wall panel from lifttmaster years ago, but I dont recall why.
>
> The light on/off [...]

I had all sorts of issues - even bought new wall controls for both my 8500 jackmounts. I eventually found Liftmaster had a recall on them and sent me new units. Haven't had issues since. Before that, I thought I had ghosts.

Definitely why I mentioned it. I'm an automation engineer by trade - get all sorts of customers that had broken things and didn't realize it until they were able to start monitoring it. "Worked before you got here". Doesn't necessarily mean there's not something in the ratgdo code for 1.0 support, but worth ruling out existing controllers.

apaperclip, 2023-11-15:
Thanks! I will check and see if mine is in the recall window.  I did a quick google just now and it looks like there was another recall as recently as feb of this year.

apaperclip, 2023-11-18:
I do not have a sticker on the back of my 888lm or any part #'s, but I did find the old email where it was replaced in 2017. It had fast blinking yellow/red leds.   In my research I learned that the 889lm replaced the 888lm and found a form to request a replacement.   Even though the site that generates the email with the coupon code for free replacement still works, the online store no longer honors the code. Perhaps this replacement campaign was a while ago.

STGMavrick, 2023-11-18:
The replacement campaign was a couple years ago from my memory. But since it was an unknown campaign unless your panel(s) died completely there'd be no way of knowing about it. So I thought it worth a mention in the chance it may skew +1.0 decoding.

STGMavrick, 2023-12-19:
> I do not have a sticker on the back of my 888lm or any part #'s, but I did find the old email where it was replaced in 2017. It had fast blinking yellow/red leds. In my research I learned that the [...]

Hey - side note, how were you able to get data coming across. I have the V2.5 running latest yaml on a LM 8500. I get no data. Wiring all good, door works as normal with wall control.

PaulWieland, 2023-12-19:
You need a wall panel which starts and maintains the data stream from the GDO. The mqtt firmware has an [emulator](https://github.com/ratgdo/mqtt-ratgdo/blob/beabf47c5808f2533ea25f5be94112c13bf858a4/src/ratgdo.cpp#L148) which does the same thing.

STGMavrick, 2023-12-19:
> You need a wall panel which starts and maintains the data stream from the GDO. The mqtt firmware has an [emulator](https://github.com/ratgdo/mqtt-ratgdo/blob/beabf47c5808f2533ea25f5be94112c13bf858a4/src/ratgdo.cpp#L148) which does the same thing.

Ahhh. I'm assuming the 888LM does not. I read the 889LM is the updated version that gives the motor status.