ratgdo/esphome-ratgdo #105: Can a basic authentication be added to the webserver? Or option to disable it alltogether

skynet01, 2023-11-09:
Since Ratgdo controls the garage and all ..
.. any chance you can add a user/pass authentication option? It's already built-in into the ESP-Home webserver: https://esphome.io/components/web_server.html just needs to be enabled.

having an option to disable the web server altogether would be great as well :)

rlowens, 2023-11-09:
You can just adopt the ratgdo into your local ESPHome Addon and edit the device .yaml to do that yourself.

skynet01, 2023-11-09:
... yes but then if there are any updates I won't get them automatically. 
 Thought it would be a good feature to have to request it here since anyone connected to your network will be able to open your garage door.

rlowens, 2023-11-09:
> ... yes but then if there are any updates I won't get them automatically.

No, only changes to the included packages are updated. Your local device .yaml is never updated, so you can add the changes there.

```
web_server:
  auth:
    username: !secret web_username
    password: !secret web_password

```

> Thought it would be a good feature to have to request it here since anyone connected to your network will be able to open your garage door.

Fair point.

skynet01, 2023-11-09:
nice, I didn't know I could declare my own parameters if you already declared "web_server" in your yaml. Thanks, for the tip

mdjhnson, 2024-04-16:
I'm not running Home Assistant so I've got a container of ESPHome running. I'm able to add my ratgdo and edit the yaml. However, I keep breaking it somehow. It fails to connect to the network and boot. 

Here's the yaml I'm trying to install. Can anyone help me figure out what I'm doing wrong?

```
substitutions:
  name: ratgdov25i-30f6e3
  friendly_name: ratgdov2.5i 30f6e3
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: WGf4TkAe6ANgN2wiCvM5yAPlt8UX164jCr1WbxGjOYw=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
  auth:
    username: complexusernamehere
    password: complexpasswordhere
```

rlowens, 2024-04-16:
> It fails to connect to the network and boot.

What are the exact errors? Connect it to USB and read the logs.

mdjhnson, 2024-04-16:
[esp-web-tools-logs (1).txt](https://github.com/ratgdo/esphome-ratgdo/files/14996311/esp-web-tools-logs.1.txt)

rlowens, 2024-04-16:
That's a lot of crashing. Mostly Out Of Memory errors. Do you get any crashes without the web_server: auth: section? Maybe just too much for the ESP to handle?

mdjhnson, 2024-04-17:
Nope. No crashing without. If I erase the device and reimagine, it works well.

tmason818, 2024-11-28:
> No, only changes to the included packages are updated. Your local device .yaml is never updated, so you can add the changes there.
>
> ```
> web_server:
>   auth:
>     username: !secret web_username [...]

Forgive me for such a simple question but: can anyone mention _where_ this .yaml is found?? On the RATGDO? In HAOS? Does the method of adding the lines above only apply to the ESPHome _Add-on_ and not the ESPHome _Integration_? Trying to get clear answers on a topic with so many overlapping and interchangeable terms is confusing me.😅.