ratgdo/esphome-ratgdo #106: Include ID's for customization

Linwood-F, 2023-11-10:
Very, very, very low priority request.... 

Most entities include an ID, and while esphome is a bit flakey in parsing substituted !extend values, this sort of setting seems to work and I have maybe a dozen of them, for entities I do not seem to need in Home Assistant (and are still visible in the web server): 

`
button: 
  - id: !extend ${id_prefix}_toggle_door
    internal: true
`

Two entities do not have them: restart and safe_mode_boot.  Maybe that's it, but if practical could all entities have id's so one can extend them?

bdraco, 2023-11-10:
Since you already know what to do, please feel free to open a PR to add missing ids

Linwood-F, 2023-11-10:
> Since you already know what to do, please feel free to open a PR to add missing ids

Hmmm... I know what code to do, I am not at all sure I know how to do a PR without a development fork of the project. But I'll take a look and see.

PR #196 (2024-01-30): feat: ensure all components in base.yaml have an id
fixes #106

Example usage of these ids:

```yaml
packages:
  remote_package:
    url: https://github.com/ratgdo/esphome-ratgdo
    files: [base.yaml]
    refresh: 1s
button: 
  - id: !remove ${id_prefix}_safe_mode
  - id: !extend ${id_prefix}_restart
    internal: true
```
closed 2024-01-30 by bdraco