ratgdo/esphome-ratgdo #109: 2.5i Board compatibility

shanelord01, 2023-11-15:
Hi,

I just received two 2.5i boards. They use the ESP-12F which has integrated WiFi, ESP8266 and PCB antenna.

I have successfully loaded the "ESPHome config for v2.5 board with ESP8266 D1 Mini lite" on these. Just wanted to ensure I'm OK to do so (seems to be running OK).

[image]

Thanks,
Shane.

PaulWieland, 2023-11-15:
Yes, its fine.

BEFH, 2023-11-19:
Does 2.5 support setting exact garage door position? It doesn't seem like this revision allows soldering to the mainboard.

PaulWieland, 2023-11-19:
No soldering is needed. Version 1.0 of ratgdo required soldering, but there are very few of those in existence and none of them are supported by ESP Home at all.

closed 2023-11-19 by PaulWieland