ratgdo/esphome-ratgdo #110: Deadband required for dry contact inputs

PaulWieland, 2023-11-15:
An ESP Home user reported that their door was being unintentionally activated [See here](https://github.com/PaulWieland/ratgdo/issues/212). They happened to notice that the dry contact input was the culprit.

I just did some testing and realized that there doesn't seem to be any deadband setup for these inputs which means noise can cause unwanted input triggers.

I would suggest we require the dry contact inputs to be closed for 500ms (that's what the mqtt firmware uses) before accepting that input as valid.

bdraco, 2023-11-16:
Probably https://esphome.io/components/binary_sensor/#binary-sensor-filters

bdraco, 2023-11-16:
I think https://github.com/ratgdo/esphome-ratgdo/pull/112 should do it, but I don't have these sensors actually hooked up so I'll have to do some debug log watching to validate it.

PaulWieland, 2023-11-16:
I can test, but not until next week. All you have to do is connect a wire to GND and then quickly short it to one of the trigger pins - it should be ignored unless you hold it for the set duration.

bdraco, 2023-11-16:
oh doh. I did it backwards (I did it on the outputs not the inputs)

Guess I should do this after coffee... fixing :)

bdraco, 2023-11-16:
```
[08:56:05][D][binary_sensor:036]: 'Dry contact open': Sending state ON
[08:56:05][D][cover:076]: 'Door' - Setting
[08:56:05][D][cover:084]:   Position: 100%
```

yeah it works fine when I short the wire to open instead of door 🤦

bdraco, 2023-11-16:
```
[08:59:26][D][binary_sensor:036]: 'Dry contact close': Sending state ON
[08:59:26][D][cover:076]: 'Door' - Setting
[08:59:26][D][cover:084]:   Position: 0%
```

close works as well

bdraco, 2023-11-16:
```
[09:00:26][D][binary_sensor:036]: 'Dry contact light': Sending state ON
[09:00:26][D][light:036]: 'Light' Setting:
[09:00:26][D][light:047]:   State: ON
[09:00:26][D][ratgdo:467]: Collision detected, waiting to send packet
``` 

so gtg

PR #112 (2023-11-16): Add a 500ms delay filter to the dry connect binary sensors
fixes #110
closed 2023-11-16 by bdraco