ratgdo/esphome-ratgdo #114: RATGDO does not appear to pull an IP address after initial flash.

BobRathke, 2023-11-16:
Just received mine today, THANK YOU for your product!

I have flashed with ESP and the device will not authenticate to my WiFi network. I can Choose my WIFi network name during install, but it will not authenticate.

Is it possible to assign a static IP? This has helped in the past with some of the micro controllers on ym network.

-Bob

limwakra, 2023-11-27:
Try to change your ssid to one without spaces. "Wifi_Fo_Fum" or something like that.

drdelaney, 2023-11-29:
I have issues with other ESP home devices on my network because i hide my SSID. To work around that, in the wifi section I add in:
`wifi:
  fast_connect: true`

Darkblade48, 2023-12-02:
> I have issues with other ESP home devices on my network because i hide my SSID. To work around that, in the wifi section I add in: `fast_connect: true`

I'm having the same problem as OP. I'm sure my Wifi SSID is correct (it shows up in the drop down) and I have retyped my password multiple times, slowly, to ensure that I am not making typos, but despite this, my logs (when I connect to the fallback webserver that the ratgdo makes at 192.168.4.1) shows similar messages as the original poster.

My SSID does not have spaces. 

I am unsure where to add 'fast_connect: true' to the web-based uploader (or is there a way to make this line change and force a manual flash?)

FieryFools, 2023-12-26:
I had the same issue when setting up my ratgdo.  IIRC, there is an option to manually enter your SSID.  I had to go that route.  I think in my case is I have a mesh wi-fi setup and the mesh system uses the same SSID for the user and wireless backhaul (kinda sloppy solution from Wi-Fi vendor) and my devices could not connect because the mesh thought the devices were trying to connect to the backhaul and not user side.  Not sure why entering SSID manually works but it did.  Also, ratgdo is not the only device to have this happen to.

unwarrantd, 2024-01-15:
I have the 2.52i board and experienced this issue today. This was after a fresh flash that included device wipe.  I received timeout errors for every connection attempt regardless of WiFi broadcast configuration.  I watched logs in reboot, and it looks like each attempt was successful. The webUI error was false. Hidden iot network gives an error, but it is working just fine. If you’re unsure on yours, simple open the terminal page from WebUI and select “reset”.  The logs will output your IP and the “visit device” button now loads the correct web page.  Not sure if this was the same issue as others, but it may help others that didn’t notice it.