ratgdo/esphome-ratgdo #120: Light turns on by itself a random times? [2.5i board, ESPHome]

shanelord01, 2023-11-21:
Module: 2.5i
Firmware: 2023.11.2 (Nov 19 2023, 21:43:41)

I have twi 2.5i boards [correctly wired](https://github.com/PaulWieland/ratgdo/issues/132) to my Merlin 2.0+ door openers.

[image]
[image]

They are then integrated into Home Assistant. At random times the lights turn on bny themselves and this is what the log shows:

```
10:16:25 | [D] | [ratgdo:210] | Status: door=CLOSED light=OFF lock=UNLOCKED
-- | -- | -- | --
10:16:43 | [D] | [light:036] | 'Light' Setting:
10:16:44 | [D] | [number:012] | 'Rolling code counter': Sending state 131.000000
10:16:44 | [D] | [esp8266.preferences:238] | Saving preferences to flash...
10:17:01 | [D] | [ratgdo:210] | Status: door=CLOSED light=ON lock=UNLOCKED
```

Home Assistant looks like this:
[image]

If it wasn't for my auto-off automation, these lights stay on constantly.

Any ideas?

Thanks,
Shane.

judeibe, 2023-11-27:
I  having the same symptoms on a Craftsman security 1.0.

finder39, 2023-12-09:
This is happening for me too on a Chamberlain Liftmaster security 1.0

judeibe, 2023-12-10:
> This is happening for me too on a Chamberlain Liftmaster security 1.0

I solved this case. Power cycle the unit by unplugging. It will be fine the next restart. Do not turn on the light by pressing the button. If you have a motion sensor on the switch turn it off as well.

finder39, 2023-12-10:
> > This is happening for me too on a Chamberlain Liftmaster security 1.0
> 
> I solved this case. Power cycle the unit by unplugging. It will be fine the next restart. Do not turn on the light by pressing the button. If you have a motion sensor on the switch turn it off as well.

This seems to work! Thanks!

Also strange that the light button on the switch doesn't directly turn the light on or off since installed the Ratgdo, but I never used it anyway

NukeThemTillTheyGlow, 2024-01-04:
> > > This is happening for me too on a Chamberlain Liftmaster security 1.0
> > 
> > 
> > I solved this case. Power cycle the unit by unplugging. It will be fine the next restart. Do not turn on the light by pressing the button. If you have a motion sensor on the switch turn it off as well.
> 
> This seems to have worked! Thanks!
> 
> Also strange that the light button on the switch doesn't directly turn the light on or off since installed the Ratgdo, but I never used it anyway

So you are using a 2.5 board with ESPHome firmware with a Security 1.0 GDO and it works?   I have a 1.0 GDO, its a red button, and having issues with the MQTT firmware, but smoothing sailing with my two Security 2.0 GDO's and Esphome firmware.