ratgdo/esphome-ratgdo #121: Use on a ESP32 Dev Board

DJAMIRSAM, 2023-11-23:
I fell like im being dumb because nobody else asked this but if I wanted to use this with an ESP32 Dev Board what pins would i need to connect? and which package do i need to install?

DJAMIRSAM, 2023-11-23:
to add this is a bare board not one purchased from the site

SomeGuy581, 2023-11-23:
You will need to add a few resistors and MOSFETs. See https://github.com/Kaldek/rat-ratgdo

And others have asked but Paul isn't willing to release his schematics and deletes the question. I expect this will be deleted when he sees it and I may be banned from this repo.

mariusmuja, 2023-11-23:
> I fell like im being dumb because nobody else asked this but if I wanted to use this with an ESP32 Dev Board what pins would i need to connect? and which package do i need to install?

Do you mean to use an ratgdo shield with an esp32 dev board? Look at the yaml file for an esp32 board (v2board_esp32_d1_mini.yaml for example) and the same version ratgdo board as you have (2.0 or 2.5) and translate the pin names as in the image [here](https://templates.blakadder.com/wemos_D1_Mini_ESP32.html), for example `D2` -> `GPIO21`.

DJAMIRSAM, 2023-11-23:
> Do you mean to use an ratgdo shield with an esp32 dev board? Look at the yaml file for an esp32 board (v2board_esp32_d1_mini.yaml for example) and the same version ratgdo board as you have (2.0 or [...]

How would I translate the pin names? And I have a esp32 dev board which still uses the D-pins but the ones that are being used in the code don't line up with my board, which ones do I need to change (and what to) to get it to work with a esp32 dev (not mini) board? And do I have to change other things in the code as-well?

rlowens, 2023-11-23:
You didn't answer the questions.

Do you mean to use a ratgdo shield with an esp32 dev board?

Also, what version of ratgdo board and exactly what esp32 dev board (link or paste a picture please).

For example, if you were using a ratgdo v2.5 with a D1 mini-style ESP32 board like this:
[image]
[image]

You would just use this esphome configuration: https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25board_esp32_d1_mini.yaml

DJAMIRSAM, 2023-11-23:
> You didn't answer the questions.
> 
> Do you mean to use a ratgdo shield with an esp32 dev board?
> 
Ah my bad I didn’t understand the question, 

This is the board I have 
https://lastminuteengineers.com/esp32-pinout-reference/

SomeGuy581, 2023-11-24:
So you do not have a ratgdo, correct? To connect that dev board without a ratgdo you will need several resisters and MOSTFETs. See my link above.

donavanbecker, 2024-04-17:
Would this one work?

https://a.co/d/eZOa2N6

closed 2024-04-17 by PaulWieland

shivaagarwal1, 2024-09-03:
> > You didn't answer the questions.
> > Do you mean to use a ratgdo shield with an esp32 dev board?
> 
> Ah my bad I didn’t understand the question,
> 
> This is the board I have https://lastminuteengineers.com/esp32-pinout-reference/

Did you figure it out? I have the same board and wondering if I can create RATGDO at home.

DJAMIRSAM, 2024-09-03:
Just ended up buying it , helps our the community and realistically it’s worth it

shivaagarwal1, 2024-09-03:
> Just ended up buying it , helps our the community and realistically it’s worth it

Thank you. Shipping to Canada is expensive, but I guess it is what it is.