ratgdo/esphome-ratgdo #122: Can read state but not control door

rob121, 2023-11-23:
Issue: I can read the state of the door, obstruction etc. but cannot control the door

Wiring appears ok, my thought is that it's some kind of rolling code issue?  It's a security 2.0 liftmaster door.    It's a brand new install is there not some kind of configuration for pairing to the door or something?

mariusmuja, 2023-11-23:
No pairing is needed. It could be a rolling code issue, but it's unlikely since the Client ID is selected randomly and it's not likely that another device with the same ID was connected to the opener. To be sure, you can go to the device settings page in HA (or the web page of the device) and increase the "Client ID" to the next value, so it would be seen by the opener as new device.

If it still doesn't work it's likely a wiring/hardware issue between the TX pin and the opener data line.

rob121, 2023-11-23:
Changing the client id didn't do anything, same issue - sense everything,
door will not open via the esphome device.

What does "If it still doesn't work it's likely a wiring/hardware issue
between the TX pin and the opener data line." mean?   What do I do to
resolve that?

mariusmuja, 2023-11-23:
> What does "If it still doesn't work it's likely a wiring/hardware issue between the TX pin and the opener data line." mean? What do I do to resolve that?

Since you are able to receive you must have the wiring correctly, because it's the same wire used for both transmit and receive. What version of the ratgdo board are you using? Make sure you are using the firmware matching that version. Is it possible that you are using a FW for board v2.0 on a v2.5 board (or the other way around)? Since the TX pin has been changed between the two board versions, it could result in the issue you're having.

rob121, 2023-11-23:
I will check that out, it’s a 2.5 board.

rob121, 2023-11-23:
Ok I went through the same process - but I had bought a 2 pack of wemos d1,
so I flashed the other one and it works when I dropped it in place. So
wiring was right - I don't know what the problem was but the new one works!
Thanks so much for your help!!!

beeldubz, 2023-11-27:
I'm having this issue as well. I had it working for a night but the next time I tried it it had stopped working. I did add it to the esphome add on in home assistant. Would this issue be possible if I put the d1 mini on backwards? Or would it just not work? Otherwise I'm not sure what caused the issue for me.

beeldubz, 2023-11-28:
In homeassistant I get an error saying "Failed to communicate with garage opener on startup; Check the Garage door Rolling code counter number entity history and set the entity to one number larger than the largest value in history. [ESPHome devices](http://homeassistant.local:8123/config/devices/dashboard?domain=esphome)" I'm not sure if it's related but I do update the rolling code and still am unable to control the door but I'm still able to see accurate readings of the door.

the-smart-home-maker, 2023-12-26:
I have the same problem. I have the 2.5i RATGDO hardware wired to a Chamberlain ML 700 EV device. I have installed ESPHome and I get the state of the obstruction sensor correctly but that's all state information I get. No opening status, can not control anything. The rolling code is increasing steadily - is that normal? I am pretty sure that the wiring is correct. I can still control everything via the wall button as well as via MyQ, but not via RATGDO. What could be the issue here?

the-smart-home-maker, 2023-12-26:
> I have the same problem. I have the 2.5i RATGDO hardware wired to a Chamberlain ML 700 EV device. I have installed ESPHome and I get the state of the obstruction sensor correctly but that's all state information I get. No opening status, no motor information etc. and I also can not control anything. The rolling code is increasing steadily - is that normal? I am pretty sure that the wiring is correct. I can still control everything via the wall button as well as via MyQ, but not via RATGDO. What could be the issue here?

Ok I got the solution for my case. In my case, as I have the ML700EV, the solution is that I have to connect the red wire not to the red terminal on the opener which is numbered with 1 but to the green one left to it which is numbered 0!

[image]

lmatter, 2024-01-01:
I was having a similar problem.  It was working, then I decided to "Adopt" the device in my Esphome add-on, and it stopped being able to control the door (still read the state just fine).  I also got the rolling code error.   
What fixed it for me was reinstalling the firmware, deleting the device from Home assistant, and letting it get rediscovered.  
Then I hit a different problem, it complained that the config didn't have API in it.  I rebooted both ratgdo and HA and then I was able to configure it after that.

Nimda11, 2024-01-02:
I am also having this issue. Circa 2018 LM 2.0 (Yellow Button) Jack Mount. Obstruction works, door control doesn't. RATGDO shows the door is open actually. RATGDO 2.5i, pretty sure I've flashed it correctly. Swapped out wires, wall button works fine.

Tried Reflashing, after removal from ESPHome and HA.

Similar logs to above

22:47:02 | [W] | [component:214] | Component preferences took a long time for an operation (0.06 s).
22:47:02 | [W] | [component:215] | Components should block for at most 20-30ms.

PaulWieland, 2024-01-02:
> Circa 2018 LM 2.0 (Yellow Button) Jack Mount.

What’s the [model number](https://github.com/PaulWieland/ratgdo/wiki)?

LSR19, 2024-01-10:
I am having the same problem as others are reporting.  I recently installed a Ratgdo v2.5 board using MQTT to control my LiftMaster LM8550W door opener.  When I installed the board a week or so ago, everything worked perfectly.  Today, though, my Home Assistant accurately reports the state of the garage door and light, but cannot control either.  I have no idea why.  The wiring is correct, because otherwise it would not have worked when first installed and Home Assistant now would not be showing the state of the door or light.  Does anyone have a solution for this, or at least what to try to get this to work again?  I've read all the posts in this thread, but didn't see a solution to the problem.

restauffer, 2024-01-10:
> I am having the same problem as others are reporting. I recently installed a Ratgdo v2.5 board using MQTT to control my LiftMaster LM8550W door opener. When I installed the board a week or so ago, everything worked perfectly. Today, though, my Home Assistant accurately reports the state of the garage door and light, but cannot control either. I have no idea why. The wiring is correct, because otherwise it would not have worked when first installed and Home Assistant now would not be showing the state of the door or light. Does anyone have a solution for this, or at least what to try to get this to work again? I've read all the posts in this thread, but didn't see a solution to the problem.
> 
> EDIT - I just did an OTA update to the most recent firmware (from 2.56 to 2.57) but that didn't correct the problem.

A few comments:

1) If you installed MQTT version you should post this in the MQTT Github repository, (https://github.com/ratgdo/mqtt-ratgdo/issues) not the one for ESPHome.

2) Include a log file showing traffic while you are trying commands.

I am not the person who can help you once you do those things but doing them will more quickly get you help from someone who has the expertise.

Good luck getting it working. You may be helping others as well as yourself.

LSR19, 2024-01-10:
Hi - thanks for your post.  I'm somewhat new to all of this, and wasn't aware of the other repository.  I'll post there.

However, for this repository, this is apropos:  I just re-flashed my v2.5i board to use ESPHome instead of MQTT, and it's working perfectly again. The ESPHome version seems much better. at least because there are many more sensors and switches/buttons available. If it goes down again, I'll post here.

Nimda11, 2024-01-15:
> Lift Master 8500

I'm just circling back to this issue. In my case I am being affected by the issue reported here [Issue 62](https://github.com/ratgdo/esphome-ratgdo/issues/62).

Am running a jack mount LM 8500 circa 2018 with a LM888 controller/button (came with the gdo), which to my understanding supports S+2 for wireless, but the wired protocol is still S+1. I verified this by flashing the MQTT firmware (2.57) and once flashed setting the protocol to S+1 in the ratgdo firmware. Everything works.

I'd rather run ESP, and will await a new build that I presume will be created as a result of the work described in the linked issue above. Hope this helps someone else, I think the relevant part of my build is the LM888 which only speaks S+1 over the wire.

bobbyschuitemaker, 2024-01-29:
> Ok I got the solution for my case. In my case, as I have the ML700EV, the solution is that I have to connect the red wire not to the red terminal on the opener which is numbered with 1 but to the [...]

Hero! that was the solution! thanks

edfield, 2024-04-02:
Having same problem

opener liftmaster 3585P
control panel 889LM
ratgdo 2.5i
ESPhome Sec 1+

Control panel is basically dead, yellow light is on but nothing else works.. Door won;t open using HA, ratgdo webpage, or the control panel. Toggling the light from HA shows on, then goes off a few seconds later. HA shows door is open, but door is closed.

These errors keep repeating in the ratgdo webpage. 

16:09:21	[W]	[ratgdo_secplus1:262]	[2437280] Discard incomplete packet: [39 ...]
16:09:21	[W]	[ratgdo_secplus1:262]	[2437413] Discard incomplete packet: [3A ...]

HAS log 
[home-assistant_2024-04-02T21-25-14.274Z.log](https://github.com/ratgdo/esphome-ratgdo/files/14843942/home-assistant_2024-04-02T21-25-14.274Z.log)

restauffer, 2024-04-03:
@edfield  I think others will be better able to help you if you send the log from esphome rather than the HA log. 

That said, I have a Sec1+ GDO and the 889LM panel and it has worked fine for months until an incident this weekend. It started with an approx 60 second power failure to the whole house. When power came back on I noticed the garage door had opened on its own, There were also some other strange occurrences such as the door opening when I turned on the light. When I looked at the esphome log it had quite a few of the same error you included in your email ([W] [ratgdo_secplus1:262] [2437280] Discard incomplete packet: [39 ...]).   I rebooted the ratgdo board (power cycled) while the GDO was already powered and all began working properly again. 

I have a theory on why his happened but won't get into it here. The upshot is, with the GDO powered, unplug the power from the ratgdo.  If my theory is correct, the wall panel will start to work again. Now, apply power to the ratgdo again. Give it a bit to reboot and check is all works properly.  If my routine does not solve the problem, then post the log from your esphome instance for the ratgdo and wait for people smarter than me to reply.

edfield, 2024-04-03:
I had read about that issue after I posted this. There is also a newer firmware that was released yesterday. I will try some more next week as I have to go out of town this weekend. Will post back if I still can't get it working.

kleinmantara, 2024-05-08:
> Ok I got the solution for my case. In my case, as I have the ML700EV, the solution is that I have to connect the red wire not to the red terminal on the opener which is numbered with 1 but to the green one left to it which is numbered 0!

Hi, I also have an ML700EV and have measured terminal 0 (signal) and 2 (GND) with an oscilloscope. But I don't see any data, only a stable voltage. I also disconnected the button for the gate (open/closed) on terminals 1 and 2, same result no data on terminal 0.
Do you have to do anything else?
[image]

the-smart-home-maker, 2024-05-12:
> > Ok I got the solution for my case. In my case, as I have the ML700EV, the solution is that I have to connect the red wire not to the red terminal on the opener which is numbered with 1 but to the green one left to it which is numbered 0!
> 
> 
> 
> @the-smart-home-maker Hi, I also have an ML700EV and have measured terminal 0 (signal) and 2 (GND) with an oscilloscope. But I don't see any data, only a stable voltage. I also disconnected the button for the gate (open/closed) on terminals 1 and 2, same result no data on terminal 0.
> 
> Do you have to do anything else?
> 
> [image]
> 
> 

@kleinmantara I have created a YouTube tutorial which also shows my wiring. Maybe this helps you?

https://youtu.be/JgKqJLtlx2A?si=0jWowgUIeuQFC4ED