ratgdo/esphome-ratgdo #123: noob question: I have a MyQ hub for Amazon deliveries and a 2015 Craftsman garage door opened with a yellow learn button

dreed47, 2023-11-24:
I have a MyQ hub for Amazon deliveries and a 2015 Craftsman garage door opened with a yellow learn button.  Like everyone else using HA my MyQ integration stopped working and I'd like to get something working again.  I purchased and received a ratgdo v2.5 a while ago but just now getting around to looking at it.  I'm just unsure how to proceed.  Since Chamberlin makes the craftsman opening im hoping I can get the ratdgo to work with it?  Other than a D1 mini (which I have) what else would I need?  Would I ignore the MyQ hub all together and wire the ratdgo up to the Craftsman?  Which wiring diagram should I be following?

rlowens, 2023-11-24:
Per https://support.chamberlaingroup.com/s/article/Determine-the-Color-of-the-Learn-Button-on-Your-Garage-Door-Opener you have a Security+ 2.0 garage door opener. Should work fine with the normal setup instructions at https://paulwieland.github.io/ratgdo/ and the "Version 2.5 Security + 1/2 Wiring Diagram"

You need the ratgdo, D1 mini (with pin headers soldered-on pointing downward), and 3 strands of wire.

You don't need to change anything about the MyQ hub - you can still use it for Amazon deliveries if you want.

rlowens, 2023-11-24:
You do need to decide if you want the MQTT or ESPHome firmware.

dreed47, 2023-11-24:
thanks...im using HA and have both ESPHome and Mosquitto broker ... is there a reason I would choose one over the other?

rlowens, 2023-11-24:
If your GDO was Security+ 1.0 or Dry Contact, you would need the MQTT firmware since the ESPHome firmware only supports Security+ 2.0 for now, but you are Security+ 2.0 so either works.

If you want to customize the firmware, it is easy in ESPHome (could wire in an AM312 passive IR motion sensor, temperature sensor, etc. to the spare GPIOs; could [set the web server to require a password](https://github.com/ratgdo/esphome-ratgdo/issues/105), etc.)

dreed47, 2023-11-24:
thanks for your help ...ill give this a shot

dreed47, 2023-11-24:
worked great...thanks!

closed 2023-11-24 by dreed47