ratgdo/esphome-ratgdo #124: Light Status after Motion Event

akeslo, 2023-11-24:
Hi,

After a motion event with the garage door opener that triggers the light to turn on (Example: motion or door open/close), the light will turn off after the defined configuration on the opener however the Ratgdo doesn't appear to receive an update indicating the light went off.  Therefore, after the light goes on with motion the light stays on within HomeAssitant.

Thanks!

mariusmuja, 2023-11-25:
The state of the light is part of a status message sent by the opener. The opener normally sends a status message when the light goes off (at least most openers do), I guess maybe your opener doesn't? My opener also sends a status message periodically, about every 5min or so. 

To better understand what's happening, can you post the log from the esp here? (you could copy paste it here from the esp webserver page). Also indicate the times when the light turns on and off. Try to record more then 5 min of log to also see if any periodic status messages are sent.

akeslo, 2023-11-25:
Thanks @mariusmuja I wasn't waiting long enough.  It looks like the 5m update comes in and the status gets reflected correctly off of this. The light turning off doesn't trigger an update on it's own.

Thanks for the help!

closed 2023-11-25 by akeslo