ratgdo/esphome-ratgdo #125: Potential Rolling Code Issue with Wireless Remotes

BakingChaos, 2023-11-27:
Having an issue with rolling code remotes (Homelink & Liftmaster) only sporadically working on 8216e9 with a 2.5 board. Before I began I wiped the memory because I couldn’t find some of my remotes and it was about time to purge them. Pairing went well and the GDO functioned as expected once. From then on it takes anywhere from 2-9 presses of the remote to toggle the door open/close. Interestingly 2.5i was flashed via the web portal linked here. I suspect the modules are similar enough it just loads 2.5/2.5i as identical? 

Wiring was done per the 2.5 wiring diagram and I’m pulling power from the 3.3v pin on the GDO control board. Wireless remotes aside everything works perfectly including wall switch, obstruction sensors, positional readings, etc. 

[image]

Before I load the MQTT variant to definitively point to the firmware is there a debug I can initiate to watch for wireless remote calls?

rlowens, 2023-11-28:
Try powering the ratgdo from a USB power adapter. You may be reducing the GDO's wireless reception power by drawing from that 3.3V line.

BakingChaos, 2023-11-29:
Unfortunately you are correct. Initial testing of desoldering the 3.3v lead I added yields immediate positive results.

closed 2023-11-29 by BakingChaos