ratgdo/esphome-ratgdo #136: Question : How to update firmware?

shanelord01, 2023-12-09:
I may have missed this in the documentation and if so, I apologise.

Just wanting to know how to update the firmware once installed? The Web server has an OTA capability, but I can't find any files to use. Or do I need to plug into the USB port to upgrade - if so, what are the steps to do so?

Any instructions on updating would be appreciated.

Thanks.

bdraco, 2023-12-09:
You can use the dashboard addon https://esphome.io/guides/getting_started_hassio.html

shanelord01, 2023-12-09:
> You can use the dashboard addon https://esphome.io/guides/getting_started_hassio.html

Thankyou - after adopting (not something I've worked with before) both of my units are showing up in the ESPHome dashboard and have the latest firmware.

tomkessler, 2023-12-10:
I'm having the same issue with an insufficient space error.  Is it possible these boards don't have enough space to handle OTA updates?  I'm on the v2.5i board.

btraversjr21, 2023-12-12:
I have the same issue. I have also had this issue with certain light bulbs. I read if you comment out the encryption portion, that does seem to work. I have not done that here because I think encryption is a good idea for this. I can't even change the hostname due to this issue. I was able to change the name when adopting, but I want to change the hostname.
`
api:
  encryption:
    key: XXXXXXXXXXXXXXXXXXXX
`

jolexa, 2023-12-14:
I'm experiencing the same issue with the v2.5i revision board. I'm glad I found this issue because I'm new to ESPHome but I'm now at the point where I cannot change anything like the previous comment (ie, hostname).
Should `insufficient space to OTA on v2.5i` be pulled out of this thread and made into a new issue? I believe the workaround is to use a computer to update (manually) as there is an option to erase all first? That is now prohibitive after it is installed in my garage 😄

shanelord01, 2023-12-21:
Ok after jumping through a few posts, I've worked out the following two-step process.

1. In ESPHome Add-on, select "Edit" under the ratgdo you want to update.
2. Copy the yaml text here to a txt file somewhere - you will need it again later.
3. Double check you have saved the yaml text - just to be safe :)
4. Replace this yaml with the following temp yaml, replacing name & friendlyname with those from your device

```
substitutions:
  name: ratgdov25i-xxxxxx
  friendly_name: ratgdov2.5i xxxxxx
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp8266:
  board: d1_mini_lite
```
5. Click "Install" in the top right corner. Select wireless. Wait for it to compile and load the OTA. It should state "OTA Successful"
6. Click "Stop" in the bottom right.
7. You should now be able to see the temp yaml you put in. Copy the yaml you saved in the txt file earlier back - replacing the temp yaml.
8. Click "Save" in the top right.
9. Click "Install" in the top right.
10. Select wireless. Wait for it to compile and load the OTA. This will take allot longer. 
11. When complete it should state "OTA Successful".
12.  Click "Stop" in the bottom right corner - your ratgdo 2.5i should now be updated to the latest version.

Hope this helps others.

jsermer, 2023-12-21:
@shanelord01 thanks for the update instructions.  i was able to follow them with a minor addition of `ota:` to my temp yaml, then the upgrade went smooth.

shanelord01, 2023-12-21:
> @shanelord01 thanks for the update instructions. i was able to follow them with a minor addition of `ota:` to my temp yaml, then the upgrade went smooth.

Oops! Must have missed that in my copy and paste. Thanks I’ve fixed that now.

robby-dermody, 2023-12-22:
Thanks for this, it seems we have to do this even with a stock ESP ratgdo config if we want to update the firmware via ESPHome in HASS. Kind of a bummer. 

The ratgdo appears to use an ESP8266-based D1 mini lite board build. Moving up to the D1 mini build seems to be a negligible cost (a few bucks?), e.g.: https://www.aliexpress.us/item/2251832342786284.html?gatewayAdapt=glo2usa4itemAdapt

@PaulWieland : given more and more people seem to be using their ratgdo's with the ESPHome-based firmwares, any chance we could move to a D1 mini board for future HW versions to avoid this two step process when flashing ESPhome upgrades, or config changes to an ESPhome-based build? More than 1MB of onboard flash is needed in both of these cases.

davidcom2, 2023-12-22:
> Ok after jumping through a few posts, I've worked out the following process that loads a minimal firmware, then the full update.
> 
> 1. In ESPHome Add-on, select "Edit" under the ratgdo you want to update.
> 2. Copy the yaml text here to a txt file somewhere - you will need it again later.
> 3. Double check you have saved the yaml text - just to be safe :)
> 4. Replace the yaml in ESPHome you can see with the following temp yaml, replacing name & friendly_name with those from your device
> 
> ```
> substitutions:
>   name: ratgdov25i-xxxxxx
>   friendly_name: ratgdov2.5i xxxxxx
> esphome:
>   name: ${name}
>   name_add_mac_suffix: false
>   friendly_name: ${friendly_name}
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> esp8266:
>   board: d1_mini_lite
> ota:
> ```
> 
> 5. Click "Install" in the top right corner. Select wireless. Wait for it to compile and load the OTA. It should state "OTA Successful"
> 6. Click "Stop" in the bottom right.
> 7. You should now be able to see the temp yaml you put in. Copy the yaml you saved in the txt file earlier back - replacing the temp yaml.
> 8. Click "Save" in the top right.
> 9. Click "Install" in the top right.
> 10. Select wireless. Wait for it to compile and load the OTA. This will take allot longer.
> 11. When complete it should state "OTA Successful".
> 12. Click "Stop" in the bottom right corner - your ratgdo 2.5i should now be updated to the latest version.
> 
> Hope this helps others.

Hey thanks. This worked for me. Couple things: 

Will this procedure have to be done again every time the ratgdo has an update in ESPHome? Hopefully not!

One note for others: I access my home assistant through DuckDNS and NGINX and when I click "Instal", the screen is blank. In order to get through the update process, I had to connect into the local IP of my home assistant install from there, I was able to carry out shanelord01's instructions.

ausil, 2023-12-24:
Just got my ratgdo today. This is a major issue that needs to be resolved. esphome as it is updates extremely often, having to update twice every time is going to wear out the flash even quicker. Either a unit with a larger storage needs to be used or the image needs to be made smaller.

ausil, 2023-12-24:
Reading #155 the fix is to use d1_mini not d1_mini_lite as the board at least as long as you have a ESP-12F which has 4MiB or flash and not 1MiB that the d1_mini_lite has

shanelord01, 2023-12-24:
> Reading #155 the fix is to use d1_mini not d1_mini_lite as the board at least as long as you have a ESP-12F which has 4MiB or flash and not 1MiB that the d1_mini_lite has

Yes. I can't test it until the next update but I've added:

esp8266:
  board: d1_mini

to the bottom of my "real" yaml - after loading the temp yaml first then loading my "real" yaml with this code included. 

I'm hoping this means I'll be able to just hit "update" next time.

shanelord01, 2023-12-26:
> I'm hoping this means I'll be able to just hit "update" next time.

And the answer is - yes. Just tapped update when Home Assistant prompted me. All good now.

jolexa, 2023-12-27:
This one can be closed now, fixed by #156 (and commentary on #155)

bdraco, 2023-12-27:
fixed by #156

closed 2023-12-27 by bdraco

bfish2, 2024-01-17:
UGH - I did this process again today (for the umpteenth time) but forgot to add my Ratgdo ID and friendly name ID to the text herein.  When I finished, my Ratgdo won't come back online via Wifi.  

What do I have to do to correct my error?  Thank you!

rlowens, 2024-01-17:
> What do I have to do to correct my error?

Take a laptop to the ratgdo or the ratgdo to a PC and flash via USB.

roopesh, 2024-01-18:
> * In ESPHome Add-on, select "Edit" under the ratgdo you want to update.
> * Copy the yaml text here to a txt file somewhere - you will need it again later.
> * Double check you have saved the yaml text - just to be safe :)
> * Replace the yaml in ESPHome you can see with the following temp yaml, replacing name & friendly_name with those from your device

I found it easier to add `ota:` and comment out
`packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v2board_esp8266_d1_mini_lite.yaml@main`
that way I didn't have to copy and paste out my configuration.  Not a big deal, but an easier process (for me).

shanelord01, 2024-01-18:
> UGH - I did this process again today (for the umpteenth time) but forgot to add my Ratgdo ID and friendly name ID to the text herein. When I finished, my Ratgdo won't come back online via Wifi.
> 
> What do I have to do to correct my error? Thank you!

If you do the steps I have above https://github.com/ratgdo/esphome-ratgdo/issues/136#issuecomment-1865903037 as they are now, you should NEVER have to do them again - just hit update in Home Assistant when prompted - the board memory size issue has been taken care of.

ushroff, 2024-01-21:
I don't think I have a board memory issue as it seems to work fine until the network step. After the Esphome update to 2023.12.7  I can no longer resolve the IP address. It is a fixed address and was working fine prior to the update.

cer2225, 2024-01-28:
Thank you for this.  It works perfectly but I was wondering if we could just change the package line and remove the _lite which would point it to the v25board_esp8266_d1_mini.yaml?

FROM:
ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main
TO
ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini.yaml@main)

Would doing this work as opposed to adding:

esp8266:
   board: d1_mini

Just curious more than anything.

-CR

bdraco, 2024-01-28:
That probably should work just fine

cer2225, 2024-01-29:
Thanks bdraco

billraff, 2024-02-19:
Yesterday after ignoring updates for quite a while I decided to follow the steps in the link above (https://github.com/ratgdo/esphome-ratgdo/pull/156) to update my device. I did this wirelessly and all seemed to go well. I didn't see any errors during the steps I followed. Once the update finished ESPHome showed the device as offline. My wireless router did see the device at the ip I had reserved for it. I restarted HA, the VM HA is running in and unplugged and re-plugged the ratgdo device all to no avail. 
So to get back online I brought the device inside and flashed it using the computer. I replaced the device in the garage and all seems to be working with the exception of one thing. I can 'visit' and 'edit' the device but if I try and reach the logs I receive the following error:

INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/ratgdov25-496086.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Detected timezone 'America/Chicago'
INFO Starting log output from 192.168.87.25 using esphome API
INFO Successfully connected to ratgdov25i-496086 @ 192.168.87.25 in 0.008s
WARNING ratgdov25i-496086 @ 192.168.87.25: Connection error occurred: ratgdov25i-496086 @ 192.168.87.25: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2023.12.9).
WARNING Can't connect to ESPHome API for ratgdov25i-496086 @ 192.168.87.25: Error while finishing connection: Finishing connection cancelled due to fatal exception: ratgdov25i-496086 @ 192.168.87.25: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2023.12.9). (HandshakeAPIError)

Anyone seen this? Anyone know why this is happening and what I can do to fix the connection to the logs?

GitHubGoody, 2024-03-09:
> Thank you for this. It works perfectly but I was wondering if we could just change the package line and remove the _lite which would point it to the v25board_esp8266_d1_mini.yaml?
>
> FROM: [...]

Since this helped me out, I figure I would close the loop on a [related troubleshooting path I recently took](https://community.home-assistant.io/t/ratgdo-local-mqtt-control-for-your-chamberlain-liftmaster-security-2-0-garage-door-opener/437380/554) that could also help someone else in the future.

normalee1993, 2024-06-09:
So after I install the temp yaml file, I can no longer have the device connect to wifi, in the logs it is unable to resolve the IP address of it. any suggestions?

Below is output of attempting to install the final YAML in step 9:

`esp8266_copy_factory_bin([".pioenvs/ratgdov25i-18f9ed/firmware.bin"], [".pioenvs/ratgdov25i-18f9ed/firmware.elf"])
========================= [SUCCESS] Took 42.93 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of ratgdov25i-18f9ed.local
ERROR Error resolving IP address of ratgdov25i-18f9ed.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname`

restauffer, 2024-06-10:
Take a look at the link. Put in your manual static IP information under WiFi. It will use that and not try mDNS.

https://esphome.io/components/wifi.html