ratgdo/esphome-ratgdo #137: Dry Contact Troubleshooting

roblandry, 2023-12-09:
Just installed everything and wired according to the instructions "Example wiring for a Genie opener". 

[image]

Currently i get the open and close sensors in esphome and home assistant, but the door state is wrong (never changes) and open/close control does not work.

[image]

Esphome config:

```yaml
substitutions:
  id_prefix: ratgdov25i
  friendly_name: ratgdov2.5i
  uart_tx_pin: D1
  uart_rx_pin: D2
  input_obst_pin: D7
  status_door_pin: D0
  status_obstruction_pin: D8
  dry_contact_open_pin: D5
  dry_contact_close_pin: D6
  dry_contact_light_pin: D3

web_server:

esphome:
  name: ${id_prefix}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
  project:
    name: ratgdo.esphome
    version: "2.5i"

esp8266:
  board: d1_mini_lite
  restore_from_flash: true

dashboard_import:
  package_import_url: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main

packages:
  remote_package:
    url: https://github.com/ratgdo/esphome-ratgdo
    files: [base.yaml]
    refresh: 1s

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

api:
  id: api_server
  encryption:
    key: REDACTED

ota:

improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

logger:
```

rlowens, 2023-12-10:
Dry Contact GDO control is only in the MQTT firmware so far. https://paulwieland.github.io/ratgdo/01_features.html

kramttocs, 2023-12-31:
@rlowens Is there a current work item for adding it to esphome that those interested could monitor to be alerted when the dry contact functionality is added?