ratgdo/esphome-ratgdo #138: Possible to implement “Lock GDO?”

pcm1ke, 2023-12-11:
Not sure if this is the proper place to put this, but I see this particular firmware has a “lock remotes” option and I was wondering if it would be possible to expose another switch to “lock gdo” effectively locking the cover in whatever state it’s in, whether that be open or closed? This way we can expose two lock entities to HomeKit, one being to lock out RF remotes and one effectively locking out _ALL_ operation of the cover. Thank you for great firmware.

rlowens, 2023-12-11:
I think that might be possible. It would take sending the "lock wireless remotes" command, then pulling the red 12V communication wire to ground and keeping it there (instead of just pulsing it as with normal communication), which would disable the wall control panel (which also gets power from the communication wire, so it would quickly lose power completely, losing clock time if it has that, and disabling the motion sensor if it has one).

I don't know if this would be bad on the ratgdo tx MOSFET or on the GDO to keep this pulled to ground long-term.

You could also just set up an automation in Home Assistant that triggers on the door starting to open and closes it. I've seen someone mention doing that.

I question what this lock would be useful for though: if you are inside the garage do you REALLY want the control panel button to not work unless you get out your phone/etc to enable it?

pcm1ke, 2023-12-11:
Some sort of exposed switch that would essentially lock the gdo and stop the cover from functioning would be nice if you simply want to “lock out” all automations easily. For example, some people use geofences and may want to temporarily halt the open/closing behavior when they cross the geofence. Just one example, I’m sure you may be able to think of other automations that people may have that they want to temporarily stop from working easily with a “master lock.”

calisro, 2023-12-12:
Id use this to lock out my kids from hitting the garage button.  So yes I can see uses.  Although it would suck to lose the motion detection.

jshatch, 2023-12-14:
Could you use the obstruction sensor for this?  Make the GDO think it is triggered and not put it down?