ratgdo/esphome-ratgdo #140: Press the push bar to activate control.

towerhand, 2023-12-14:
I was going to write this on the main PaulWieland/ratgdo GitHub page, but reporting issues seem to be disabled. Please let me know if there is a better place to ask for help. Anyway, I'm using the ESPhome firmware, and everything works perfectly, except the wall controller is disabled and has a "Press the push bar to activate control. See user manual" error.

Pressing the push bar multiple times doesn't do anything, and I already tried removing the power for an hour, rewiring everything, and fully removing ratgdo. Still, the wall controller won't start working again. 

Has anyone encountered this same issue and knows how to fix it?

bdraco, 2023-12-14:
I've had the same issue on one of my family's GDOs without ratgdo. It ended up being a bad wall control

towerhand, 2023-12-14:
Thank you @bdraco. That's what I was afraid of since they seem pretty expensive for what they are.

closed 2023-12-14 by towerhand

tfcarlin, 2023-12-14:
I also had this issue. Pressing the wall control button for > "x" seconds worked for me. I don't recall "x" but certainly > 5 seconds. It was flaky for a few days and has since started working correctly again. Very strange.

towerhand, 2023-12-14:
> I also had this issue. Pressing the wall control button for > "x" seconds worked for me. I don't recall "x" but certainly > 5 seconds. It was flaky for a few days and has since started working correctly again. Very strange.

Thanks for sharing. I tried that and it didn't help.

I contacted Chamberleain and they will be sending me a new wall controller.