ratgdo/esphome-ratgdo #141: ratgdo runs out of available flash when adding a few lines to config

robby-dermody, 2023-12-14:
Hi guys, on the heels of the insights from issue #130, I was trying to use the esphome build to work with a Liftmaster gate operator (which utilizes dry contacts). I tried the MQTT build but it was not working at all and the diagnostics were very limited. Anyhow, I added the ESPHome component to my HA instance, adopted my ratgdo device, and modified its config to be as follows:

```
#this part is all stock config...
substitutions:
  name: ratgdov25i-fad812
  friendly_name: frontgate-ratgdo
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: REDACTED

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

##############
#added this below
esp8266:
  early_pin_init: false

switch:
  - platform: gpio
    pin:
      number: D1
    id: frontgate_ratgdo_sbc
    name: "Frontgate Door"
    icon: "mdi:door"
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: frontgate_ratgdo_sbc

cover:
   - platform: template
     name: "Gate Operator"
     lambda: |-
       if (id(ratgdov25i_dry_contact_open).state) {
          return COVER_OPEN;
       } else {
          return COVER_CLOSED;
       }
     open_action:
       - switch.turn_on: frontgate_ratgdo_sbc
     close_action:
       - switch.turn_on: frontgate_ratgdo_sbc
     stop_action:
       - switch.turn_on: frontgate_ratgdo_sbc
     optimistic: true
```

I realize this config could have issues, but have yet to test because when I try to install the config to the device, the image exceeds the available space:

```
Processing ratgdov25i-fad879 (board: d1_mini_lite; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- Secplus @ 1.0.0+sha.f98c322
|-- EspSoftwareSerial @ 8.1.0+sha.57f1aa7
|-- Improv @ 1.2.3
RAM:   [=====     ]  45.9% (used 37580 bytes from 81920 bytes)
Flash: [======    ]  59.9% (used 612885 bytes from 1023984 bytes)
========================= [SUCCESS] Took 1.23 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of ratgdov25i-fad879.local
INFO  -> 10.0.3.25
INFO Uploading /data/build/ratgdov25i-fad879/.pioenvs/ratgdov25i-fad879/firmware.bin (617040 bytes)
INFO Compressed to 415833 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
```

It's off by like 4-5kb. I'm wondering if things are really this tight (maybe I have to exclude some things from the included stock config to get it to fit), or maybe my device just needs to be erased and reinitialized. 

Thanks for any help!

mariusmuja, 2023-12-14:
> It's off by like 4-5kb. I'm wondering if things are really this tight (maybe I have to exclude some things from the included stock config to get it to fit), or maybe my device just needs to be erased and reinitialized.

Yes, on esp8266 things are tight for OTA.

For OTA uploads the old and the new firmwares together should not exceed the flash size. You can do what you suggested, exclude some of the stock config that you don't use to reduce the new firmware size, or have a minimal firmware just with OTA enabled and flash that one followed by the firmware you want to flash (so OTA flashing is always a 2-step OTA).

rlowens, 2023-12-14:
> maybe my device just needs to be erased and reinitialized.

Or just do as the error suggests: 
> Please try flashing a minimal firmware (remove everything except ota)

You ota flash the minimal firmware to clear space, then you have room for the larger original firmware to flash ota. Makes updates a 2-step process, but should work fine. Save the minimal version as a separate yaml so you can just flash it when needed.

I think this should be it:

```
substitutions:
  name: ratgdov25i-fad812
  friendly_name: frontgate-ratgdo
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: REDACTED

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp8266:
  board: d1_mini_lite

ota:

```

robby-dermody, 2023-12-14:
Thanks, and forgive my ignorance here (I just got into ESPHome and working with ratgdo yesterday). I think where I got confused is that I assumed that the snippets from #130 were in addition to the default config for the device, since they did not include `substitutions`, `esphome` and so on. That's why I'm a bit confused over why that device was running it, and not mine, assuming they both have 1MB of onboard flash.

Also, by stripping out the `v25iboard.yaml` reference, it appears I would lose the web interface, logging, etc, which was the benefit of going with ESPHome over the MQTT firmware -- that I could see what was happening in the device itself (dry contact statuses, for instances) and debug further.

Are there a few bells and whistles I could try disabling instead that would free up that 5-10kb and allow me to keep the web interface and logging capabilities?

rlowens, 2023-12-14:
> I think where I got confused is that I assumed that the snippets from #130 were in addition to the default config for the device, since they did not include `substitutions`, `esphome` and so on. [...]

Ah, I see the confusion. Yes, they were only showing the custom code, not the full config with the usual stuff at the top. You would still want web, ota, logging, etc. set as you want. Just put those parts in your yaml. You can view what the current file expands to with all the includes and defaults etc. by clicking the ... and then Validate in the ESPHome UI.

rv4mike, 2023-12-21:
How does one determine what the current minimal firmare is that will allow a OTA update to be flashed to then follow with a full OTA firmware update now having room on the ESP8266 board to do a full OTA firmware update

kramttocs, 2023-12-21:
Getting this error when ESPHome is showing an update to 2023.12.1 (from 11.6) and I haven't added anything.

kramttocs, 2023-12-21:
https://github.com/ratgdo/esphome-ratgdo/issues/136#issuecomment-1865903037

robby-dermody, 2023-12-22:
I'm going to close this in favor of #136 and the solution there

closed 2023-12-22 by robby-dermody