ratgdo/esphome-ratgdo #142: Understanding the difference in door sensor, motor, sensor, etc... and just using the available ports on gdo when soldering to 3.3v

bedge117, 2023-12-15:
This isn't so much of an issue as it is a request for clarification.  I currently have 2 ratgdo's 2.5 connected to my 2 gdos.

I have them connected via a usb adapter for power.  I am going to solder them to a 3.3v source on the controller to eliminate the need for an extra power brick at each outlet.

That being said, the instructions on installation are vague as to what the benefit of soldering to rotary encoder instead of using the external wiring. 

So this request is to explain the 2 status pins "Door Status" and "Obst Status."  Because It seems like this is simply a 3.3v output to use for my own purpose (status lights, etc...)
Also, I would also like to know if I choose the solder method, can I not just use the 3.3v from Gdoor and wire everything else via the solderless method?

Is the RPM1 and RPM2 more accurate?  Or if using the 3.3 v it can't accurately retrieve the information from the red wire?  Or, is the solder method obsolete for v2.5?  

I noticed that my boards do not have rpm1 or rpm2, so maybe that method refers to the v2 boards only?

Thanks!

rlowens, 2023-12-15:
> I am going to solder them to a 3.3v source on the controller to eliminate the need for an extra power brick at each outlet.

Note that some people that did that got much-reduced power on the wireless remote range/reliability and had to switch to not using that internal 3.3v source. Different GDO models probably have different amounts of spare power on internal 3.3v source. If your model has a 12v battery-backup connection (even if you don't have the battery itself installed), using a 12v->3.3v power supply has been much more reliable.

> That being said, the instructions on installation are vague as to what the benefit of soldering to rotary encoder instead of using the external wiring.

Those instructions are only for the old ratgdo v1. Any version after that doesn't use the rotary encoder connections at all.

> So this request is to explain the 2 status pins "Door Status" and "Obst Status." Because It seems like this is simply a 3.3v output to use for my own purpose (status lights, etc...)

Correct. You can wire them to status lights, etc.

> Also, I would also like to know if I choose the solder method, can I not just use the 3.3v from Gdoor and wire everything else via the solderless method?

Yes. But be aware of what "solder method" you are looking at, it may not be correct for your ratgdo v2.5.

> is the solder method obsolete for v2.5?

Yes.

bedge117, 2023-12-15:
> > I am going to solder them to a 3.3v source on the controller to eliminate the need for an extra power brick at each outlet.
> 
> Note that some people that did that got much-reduced power on the wireless remote range/reliability and had to switch to not using that internal 3.3v source. Different GDO models probably have different amounts of spare power on internal 3.3v source. If your model has a 12v battery-backup connection (even if you don't have the battery itself installed), using a 12v->3.3v power supply has been much more reliable.
> 
> > That being said, the instructions on installation are vague as to what the benefit of soldering to rotary encoder instead of using the external wiring.
> 
> Those instructions are only for the old ratgdo v1. Any version after that doesn't use the rotary encoder connections at all.
> 
> > So this request is to explain the 2 status pins "Door Status" and "Obst Status." Because It seems like this is simply a 3.3v output to use for my own purpose (status lights, etc...)
> 
> Correct. You can wire them to status lights, etc.
> 
> > Also, I would also like to know if I choose the solder method, can I not just use the 3.3v from Gdoor and wire everything else via the solderless method?
> 
> Yes. But be aware of what "solder method" you are looking at, it may not be correct for your ratgdo v2.5.
> 
> > is the solder method obsolete for v2.5?
> 
> If you want.

Thank you for the speedy reply!

I will use a buck converter from the 12 volt then.  I have a few spare lying around.  Not sure why I didn't think about that in the first place. Thanks for the tip!

I think you cleared everything up for me.  :)

closed 2023-12-15 by bedge117