ratgdo/esphome-ratgdo #150: Is there a way to set a user/pass for the web interface?

mattgio22, 2023-12-18:
I have a ratgdo 2.5 and flashed it with the web installer - it all works great. My only issue is that anyone on my network can easily navigate to that page and change things. Is there a way to set a user/pass for logging into the web interface? Thanks!

mariusmuja, 2023-12-18:
Yes, you'll have to add the `auth` option to the `web_server` component in the yaml configuration: https://esphome.io/components/web_server.html

mattgio22, 2023-12-18:
Thanks so much for that! I've read that page a couple of times now, and have a follow up question. Please forgive me being a noob, but where exactly would I edit the yaml config? Since I flashed via the web installer, I never actually had to use any yaml file during the initial setup. Thanks again.

bdraco, 2023-12-18:
Adopt it in the ESPHome dashboard and than you can edit the YAML

mattgio22, 2023-12-18:
OK - thanks. I got it now. Somehow I managed to get it all working without actually using the ESPHome dashboard. This looks like a much more robust way going forward. Thank you both!

clueo8, 2023-12-22:
> OK - thanks. I got it now. Somehow I managed to get it all working without actually using the ESPHome dashboard. This looks like a much more robust way going forward. Thank you both!

Looking to do the same; can you explain what you did for this?  Thanks!

closed 2024-04-01 by PaulWieland

Bradley-Parker, 2024-04-13:
For added clarity as an HA user coming at this as their first ESPHome Device, 

The part I was missing is to install and use the ESPHome Dashboard add-on inside of HA, and not to look for the yaml editor in the ESPHome Integration set up (Integration != Add-on). 

I was scratching my head trying to set it in the webui at https://web.esphome.io/, before realizing I needed to set up the add on at https://esphome.io/guides/getting_started_hassio.html, and now I can edit my yaml