ratgdo/esphome-ratgdo #153: Help wanted:  Decipher command/response for getting number of paired devices

mulcmu, 2023-12-21:
Gdo responds to a Command 0x307 sent from ratgdo with Command 0x308.  The 0x308 response appears to contain the number of paired devices in byte2.  When 0x307 is issued with non-zero nibbles, it appears a subset of the paired devices is returned:
* 0 total number of paired devices is returned.  
* 1 total wireless push button remotes
* 2 total wireless keypads
* 3 total wireline devices 
* 4 total myQ (speculation)

I don't have any myQ accessories to pair so help is needed to see if and where they are included in the 0x308 response.  Below yaml can be added to the `button:` block to collect the 0x308 data with logging set to VERBOSE to collet the data.  The command for clearing memory has a similar structure, so this will support adding ability to clear paired devices by ratdgo.

```
  - platform: template
    id: ${id_prefix}_query_devices
    name: "Query Paired Devices"
    entity_category: diagnostic
    on_press: 
      then:
        - lambda: !lambda |-
            ESP_LOGW("ratgdo", "############### Start Get Paired Devices #############");
            id($id_prefix).send_command( (ratgdo::Command) 0x307, 0);
        - delay: 300ms
        - lambda: !lambda |-
            id($id_prefix).send_command( (ratgdo::Command) 0x307, 1);
        - delay: 300ms
        - lambda: !lambda |-
            id($id_prefix).send_command( (ratgdo::Command) 0x307, 2);
        - delay: 300ms
        - lambda: !lambda |-
            id($id_prefix).send_command( (ratgdo::Command) 0x307 ,3); 
        - delay: 300ms
        - lambda: !lambda |-
            id($id_prefix).send_command( (ratgdo::Command) 0x307 ,4); 
        - delay: 300ms
        - lambda: !lambda |-
            ESP_LOGW("ratgdo", "############### END Get Paired Devices #############");
```

After flashing new code with above changes, connect to logs over wireless and share 0x308 the response output here along with details of the connected devices.

```
[09:58:17][W][ratgdo:334]: ############### Start Get Paired Devices #############
[09:58:17][V][ratgdo:642]: Send command: GET_PAIRED_DEVICES, data: 00000000
[09:58:17][V][ratgdo:340]: [52392] Encode for transmit rolling=000773c fixed=03002f5539 data=00000007
[09:58:17][V][ratgdo:685]: Sending packet
[09:58:17][V][ratgdo:373]: Packet: [55 01 00 11 12 5B 28 1E 49 64 96 89 0F 2E 78 45 44 05 26]
[09:58:17][D][number:012]: 'Rolling code counter': Sending state 30525.000000
[09:58:17][V][ratgdo:461]: Baud: 9615
[09:58:17][V][ratgdo:373]: Packet: [55 01 00 1A 2F D6 DC A9 B6 93 DD 62 05 04 32 9D 12 86 69]
[09:58:17][V][ratgdo:109]: [52536] received rolling=001e78d fixed=6308ca9016 data=06000008
[09:58:17][V][ratgdo:117]: cmd=308 (PAIRED_DEVICES) byte2=06 byte1=00 nibble=0
[09:58:17][V][ratgdo:642]: Send command: GET_PAIRED_DEVICES, data: 00000001
[09:58:17][V][ratgdo:340]: [52740] Encode for transmit rolling=000773d fixed=03002f5539 data=00000107
[09:58:17][V][ratgdo:685]: Sending packet
[09:58:17][V][ratgdo:373]: Packet: [55 01 00 10 16 D9 61 3A DB 2D B6 6A 22 13 AF 2A A0 C3 DE]
[09:58:17][D][number:012]: 'Rolling code counter': Sending state 30526.000000
[09:58:17][V][ratgdo:461]: Baud: 9615
[09:58:17][V][ratgdo:373]: Packet: [55 01 00 85 3C 17 D0 3D F6 DB E6 11 22 0A 3C 5A 09 46 94]
[09:58:17][V][ratgdo:109]: [52897] received rolling=001e78e fixed=6308ca9016 data=03000108
[09:58:17][V][ratgdo:117]: cmd=308 (PAIRED_DEVICES) byte2=03 byte1=00 nibble=1
[09:58:18][V][ratgdo:642]: Send command: GET_PAIRED_DEVICES, data: 00000002
[09:58:18][V][ratgdo:340]: [53097] Encode for transmit rolling=000773e fixed=03002f5539 data=00000207
[09:58:18][V][ratgdo:685]: Sending packet
[09:58:18][V][ratgdo:373]: Packet: [55 01 00 66 1F 7D F5 8B 7F B7 FB 19 13 48 F2 87 09 48 5D]
[09:58:18][D][number:012]: 'Rolling code counter': Sending state 30527.000000
[09:58:18][V][ratgdo:461]: Baud: 9615
[09:58:18][V][ratgdo:373]: Packet: [55 01 00 84 3E 1E F4 AF BF FF 7E 02 39 E6 83 03 A4 90 79]
[09:58:18][V][ratgdo:109]: [53255] received rolling=001e78f fixed=6308ca9016 data=01000208
[09:58:18][V][ratgdo:117]: cmd=308 (PAIRED_DEVICES) byte2=01 byte1=00 nibble=2
[09:58:18][V][ratgdo:642]: Send command: GET_PAIRED_DEVICES, data: 00000003
[09:58:18][V][ratgdo:340]: [53447] Encode for transmit rolling=000773f fixed=03002f5539 data=00000307
[09:58:18][V][ratgdo:685]: Sending packet
[09:58:18][V][ratgdo:373]: Packet: [55 01 00 65 2D B6 98 3D A4 DA 69 0A 08 2E 28 F8 A4 9F 10]
[09:58:18][D][number:012]: 'Rolling code counter': Sending state 30528.000000
[09:58:18][V][ratgdo:461]: Baud: 9615
[09:58:18][V][ratgdo:373]: Packet: [55 01 00 A2 0C E0 1F 89 04 12 61 A0 16 79 4C 66 79 E9 92]
[09:58:18][V][ratgdo:109]: [53604] received rolling=001e790 fixed=6308ca9016 data=02000308
[09:58:18][V][ratgdo:117]: cmd=308 (PAIRED_DEVICES) byte2=02 byte1=00 nibble=3
[09:58:18][V][ratgdo:642]: Send command: GET_PAIRED_DEVICES, data: 00000004
[09:58:18][V][ratgdo:340]: [53805] Encode for transmit rolling=0007740 fixed=03002f5539 data=00000407
[09:58:18][V][ratgdo:685]: Sending packet
[09:58:18][V][ratgdo:373]: Packet: [55 01 00 A4 37 FB F6 EF DF 7F BF 1A 3E FE 21 EE B6 93 84]
[09:58:18][D][number:012]: 'Rolling code counter': Sending state 30529.000000
[09:58:18][D][esp32.preferences:114]: Saving 1 preferences to flash...
[09:58:18][V][esp32.preferences:126]: sync: key: 1885324629, len: 4
[09:58:18][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[09:58:19][V][ratgdo:461]: Baud: 9615
[09:58:19][V][ratgdo:373]: Packet: [55 01 00 A1 16 89 A9 52 69 A4 93 91 1E 69 8C 46 ED F0 D6]
[09:58:19][V][ratgdo:109]: [53950] received rolling=001e791 fixed=6308ca9016 data=00000408
[09:58:19][V][ratgdo:117]: cmd=308 (PAIRED_DEVICES) byte2=00 byte1=00 nibble=4
[09:58:19][W][ratgdo:350]: ############### END Get Paired Devices #############
```

mariusmuja, 2023-12-21:
I can confirm that nibble 4 represents the number of MyQ devices paired. I did a factory reset and it showed 0, then added a MyQ bridge and it shows 1.

mariusmuja, 2023-12-21:
We should add this to the `Sync` step and present it in the diagnostics section.

mulcmu, 2023-12-21:
Superb! Thanks for confirming.  

I will roll this into #149 with functionality to clear memory as well.  I was thinking to implement both the get() and clear() operations as a service back to HA.  Accidental activation of the clear() would be undesirable so putting it in a service call seems like a good spot to make usage deliberate.  Not an operation I would expect users to put in an automation.  A new ratgdo event to trigger when additional device is learned would be useful.

mulcmu, 2024-01-20:
Merged in #149.

closed 2024-01-20 by mulcmu

mwaterbu, 2024-05-16:
Is there a simple way to call query_paired_devices(), or see a breakdown in the log of the counts/types of paired devices without needing to add all the sensors ([which were removed due to high memory usage](https://github.com/ratgdo/esphome-ratgdo/pull/149/commits/68086fef1ee72e80a06c13c2a1bb88d2ed047728#r1457954651))?

The [Web UI Documentation](https://ratgdo.github.io/esphome-ratgdo/webui_documentation.html) shows a button to call this query, but I don't see it in the actual UI.