ratgdo/esphome-ratgdo #154: Dry Contact Configuration

Maverick766, 2023-12-22:
Is there any update on when the ESP version will support dry contact configuration?

dmkjr, 2023-12-25:
I have been playing with this trying to get my old garage door opener to work with dry contacts.  And then I see this comment.  Is there a known issue?  I have an Overhead Door Legacy unit that only has three inputs.  "Push Button". "Common".  "Obstruction Sensors".

Haven't found two contacts that can be jumped together with the 2.5i version of the board I just received.  I have it wired where the push button is still functioning through the ratdgo.

rlowens, 2023-12-26:
> Is there any update on when the ESP version will support dry contact configuration?

You can write your own ESPHome .yaml for dry contact control. See https://github.com/ratgdo/esphome-ratgdo/issues/130#issuecomment-1848101230

dmkjr, 2023-12-26:
> > Is there any update on when the ESP version will support dry contact configuration?
> 
> You can write your own ESPHome .yaml for dry contact control. See [#130 (comment)](https://github.com/ratgdo/esphome-ratgdo/issues/130#issuecomment-1848101230)

Thank you. Will try now that. So if my garage door operates basically like a doorbell, what pins would act as the “common” and the “open/close” switch leg?  

I only have “push button” (basically open/close), common and “safety wire” as my three inputs on the unit. Shorting out common and push button allow the door to operate.

rlowens, 2023-12-26:
> So if my garage door operates basically like a doorbell, what pins would act as the “common” and the “open/close” switch leg?

Common connects to Ground on the ratgdo and "open/close" connects to Red/Ctrl on the ratgdo (pin D1 on ratgdo v2.5+, D4 on ratgdo v2.0).

dmkjr, 2023-12-28:
> > So if my garage door operates basically like a doorbell, what pins would act as the “common” and the “open/close” switch leg?
> 
> Common connects to Ground on the ratgdo and "open/close" connects to Red/Ctrl on the ratgdo (pin D1 on ratgdo v2.5+, D4 on ratgdo v2.0).

Thank you.  That worked great!  Now I need to put some sensors on to determine if it's opened or closed.  May try my hand at creating a device within Home Assistant to use the sensor to open/close but use my alarm to determine if the door is opened or closed.

dmkjr, 2023-12-28:
> > So if my garage door operates basically like a doorbell, what pins would act as the “common” and the “open/close” switch leg?
> 
> Common connects to Ground on the ratgdo and "open/close" connects to Red/Ctrl on the ratgdo (pin D1 on ratgdo v2.5+, D4 on ratgdo v2.0).

Quick question - I wont get to this today anyway - but the button on the wall has a light button that functions.  It cuts the light on, but I'm assuming it is doing this by a wireless communication?  I only see two wires from thjen CAT5 that the contractos has run hooked up from the push button.  Common and Open/Close.  There is not a contact that shows light on the GDO.

Fail-Safe, 2023-12-28:
@dmkjr I'm very curious as to the same question about the light button on the wall. I also have an [Overhead Door Legacy [850]](https://www.overheaddoor.com/garage-door-opener-legacy-850). Will be keeping an eye on this thread 👍

dmkjr, 2023-12-28:
> @dmkjr I'm very curious as to the same question about the light button on the wall. I also have an [Overhead Door Legacy [850]](https://www.overheaddoor.com/garage-door-opener-legacy-850). Will be keeping an eye on this thread 👍

whew, I thought I was missing something.  Wish you had the answer, but glad I'm not the only one.

matchmee, 2024-01-06:
So does this mean that Dry Contact only (like my Genie GDO) will not get ESPHome functionality anytime soon without me messing with the yaml file? I know just enough about this stuff to really mess things up. :) Would someone be willing to check my assumptions of the steps? 
1. Flash 2.5 ESPHome on ratgdo board
2. Add integration for ESPHome
3. Edit yaml file with correct information
4. It works? 

I have the current MQTT version working with magnetic sensors giving me accurate status updates (opening, open, closing, close). I just would love to have the functionality of % opening as well. And also maybe even figure out how to get the light and lock to work since they are on my wall control button.

maksym3d, 2024-07-13:
I'm also looking to enable light control on my older Lift Master / Craftsman opener. The remote panel toggles door by simply shorting the red and white wires, whereas light is triggered by shorting these wires via a 1uF capacitor (and vacation is triggered via 22uF capacitor).

I do not see an obvious way to add the capacitor to the current ratgdo v2.5x. I think it should be possible by connecting one more wire from one of ratgdo's pin in series with the capacitors to the Red terminal of the opener and then modifying the firmware to short this wire to the white wire, similar to how it is currently done to toggle the door.

Alternatively, it might be possible to emulate the capacitor by modulating or timing the 'press' on the current set of wires - I'm trying to recruit a collective wisdom to understand how:

https://electronics.stackexchange.com/questions/719335/capacitor-in-series-with-a-button

It would be also nice to poll the status of light with some kind of a photoresistor, similar to how we are polling doors with the magnetic limit switches. I think that's what the 'light' wire on the header is, but don't see it mentioned anywhere in the wiring how-to.

maksym3d, 2024-07-28:
I have hooked up an oscilloscope to the panel control wire to see what that 1uF capacitor is doing. Apparently there is a 98Hz carrier frequency and the capacitor chops about the last 2ms of the 10ms signal:

[image]

I have tried to emulate this by sending 0xFE characters at 1000 baud serial, which produces a very close signal, but the opener does not seem to recognize it. A PWM signal at 98Hz with a 20% duty cycle would probably work a lot better, but I have not figured how to do it in the code yet. I don't see a function readily available in ESPHome, but they do have a support for PWM through YAML configuration entries.

For now, I have connected D5 pin (dry open wire on the 6-wire header) through a 1uF capacitor to the red Control wire and just used that for the 'Light Toggle' in my custom firmware. The 'Light' trigger wire can be used to sense the light either through a power relay spliced into the light bulb line of the opener, or maybe through a photo resistor.

PaulWieland, 2024-07-28:
Dry contact control was added some time ago. The binary and corresponding wiring diagram is available on the ratgdo installer page. 

https://ratgdo.github.io/esphome-ratgdo/

Light control support cannot be implemented in conjunction with dry contact control because it’s an analog signal and the control signal varies between gdo models.

closed 2024-07-28 by PaulWieland