ratgdo/esphome-ratgdo #155: ratGDO ESP firmware update to 2023.12.1 fails with error: not enough space to store OTA file

davidcom2, 2023-12-22:
This is my first time ever posting an issue on github. Apologies in advance if I don't report this issue correctly, or if this is the wrong place to report it. 

I'm trying to update my esphome ratgdo in Home Assistant. I'm getting the following message: 

`ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)`

Can anyone give advice on how to resolve this? I'm afraid I don't know how to do what is being suggested ("Please try flashing a minimal firmware (remove everything except ota)")

Thanks!

towerhand, 2023-12-22:
https://github.com/ratgdo/esphome-ratgdo/issues/136#issuecomment-1865903037

edrikk, 2023-12-22:
Not sure how/if it applies to the esp version, but thought to share just in case.

https://github.com/ratgdo/homekit-ratgdo/pull/50

davidcom2, 2023-12-22:
> [#136 (comment)](https://github.com/ratgdo/esphome-ratgdo/issues/136#issuecomment-1865903037)

Thanks. I followed those instructions and was able to complete the update. I don't know if that update process will have to be done every time the ratgdo has an update in ESPHome. Hopefully not because it's somewhat involved just to update. 

Thanks again!

robby-dermody, 2023-12-22:
So given the comment at https://github.com/ratgdo/homekit-ratgdo/pull/50 that the ratgdo has 4MB of flash (thanks @edrikk ), is that actually the case? And if so, why does the ESPhome config list the board as a `d1_mini_lite`, which has 1MB flash capacity from what I can tell? Maybe this is as simple as changing the `board:` config to get the access to that extra memory?

kpfleming, 2023-12-23:
The 2.5i board I received today has an ESP-12F which does have 4MB of flash, so I've configured ESPHome for `d1_mini` instead of `d1_mini_lite` and it's working fine. I suspect given the backorder situation there may be a variety of 2.5i boards out there with different ESP modules on them and the 'standard' config is designed to work on all of them.

bdraco, 2023-12-23:
AFAICT all the 2.5is are ESP-12F so we can change https://github.com/ratgdo/esphome-ratgdo/blob/587ff4c7c5c40d438469448319a665823c02840f/static/v25iboard.yaml#L25 to `d1_mini` but need @PaulWieland to confirm before making that change

PaulWieland, 2023-12-24:
@bdraco Confirmed.

bdraco, 2023-12-24:
New images will be up in 10-15 minutes

bdraco, 2023-12-24:
https://github.com/ratgdo/esphome-ratgdo/actions/runs/7311723573

closed 2023-12-24 by bdraco

shred86, 2024-01-02:
I'm having this same issue on a ratgdo 2.0 board (esp8266 d1 mini lite). Is there a work around for this version? All of the comments on the various issues are in reference to the v2.5i board.

bdraco, 2024-01-02:
Replace the esp with the non lite version.

shred86, 2024-01-02:
Sorry, dumb question but just to clarify - are you saying buy a new physical esp device because there's no work around for the v2.0 esp8266 board? Or to change the config file to use the d1_mini instead of d1_mini_lite?

shred86, 2024-01-02:
I'm not sure. I ordered it from Paul Wieland's website when the v2.0 boards were the latest. On the ESPHome ratgdo web install page, it looks identical to what's pictured as the ESP8266 (left most). Of course I can't find the email receipt since I was hoping maybe that identifies it. Otherwise, I'll just have to pull it down from the garage door opener and check I suppose.

shred86, 2024-01-02:
That being said, I'm pretty sure it's a D1 Mini Lite since I've been using v2board_esp8266_d1_mini_lite.yaml. If that's the case, then the only workaround I can think of that seems to have worked is just completely stripping down the configuration to minimize the file size. I was able to get the latest version of ESPHome firmware using the yaml example below. I removed several things that I didn't think were necessary, to include the web server. I suppose the only downside of this is if there's any changes upstream, then I'll have to manually make the updates (unless they're changes in the base.yaml file which is still being pulled).

````yaml
substitutions:
  id_prefix: ratgdov2
  name: ratgdov2-xxxxxx
  friendly_name: ratgdov2 xxxxxx
  uart_tx_pin: D4
  uart_rx_pin: D2
  input_obst_pin: D7
  status_door_pin: D0
  status_obstruction_pin: D8
  dry_contact_open_pin: D5
  dry_contact_close_pin: D6
  dry_contact_light_pin: D3

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: ratgdo.esphome
    version: "2.0"

esp8266:
  board: d1_mini_lite
  restore_from_flash: true

packages:
  remote_package:
    url: https://github.com/ratgdo/esphome-ratgdo
    files: [base.yaml]
    refresh: 1s

time:
  - platform: homeassistant
    id: homeassistant_time

api:
  id: api_server
  encryption:
    key: redacted

ota:

improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:
````

rlowens, 2024-01-02:
Note that you can still do the 2-step OTA update: install a minimal .yaml with just the device name and OTA, then install the full .yaml.

rlowens, 2024-01-02:
> That being said, I'm pretty sure it's a D1 Mini Lite since I've been using v2board_esp8266_d1_mini_lite.yaml.

Doesn't mean that's what the device actually has, just that it worked with that setting.

To test the real flash size, you can download [esptool](https://github.com/espressif/esptool/releases/latest) and run:

`esptool.exe flash_id`

Example results on my NodeMCU dev board (I don't have any d1 mini style boards):

> esptool.exe flash_id
> esptool.py v4.7.0
> Found 2 serial ports
> Serial port COM7
> Connecting....
> Detecting chip type... Unsupported detection protocol, switching and trying again...
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Features: WiFi
> Crystal is 26MHz
> MAC: dc:4f:22:12:15:81
> Uploading stub...
> Running stub...
> Stub running...
> Manufacturer: 20
> Device: 4016
> Detected flash size: 4MB
> Hard resetting via RTS pin...
> 

Note: the Windows Defender detection of virus AndroidOS/ZkarletFlash is a false-positive, but instead of going thru the hassle of allowing it in Windows Defender and then re-trying the download, you can just get the v4.6.2 older release that seems to scan fine for me. https://github.com/espressif/esptool/releases/tag/v4.6.2

I reported this problem at https://github.com/espressif/esptool/issues/944

bdraco, 2024-01-02:
If it turns out to not be a lite, we can add the non-lite version to the site

shred86, 2024-01-03:
Here's the output of the debug with that sensor added:

````
[02:58:36][D][debug:076]: ESPHome version 2023.12.5
[02:58:36][D][debug:080]: Free Heap Size: 18616 bytes
[02:58:36][D][debug:108]: Flash Chip: Size=1024kB Speed=40MHz Mode=DOUT
[02:58:36][D][debug:323]: Chip ID: 0x00C84EAE
[02:58:36][D][debug:324]: SDK Version: 2.2.2-dev(38a443e)
[02:58:36][D][debug:325]: Core Version: 3.0.2
[02:58:36][D][debug:326]: Boot Version=7 Mode=1
[02:58:36][D][debug:327]: CPU Frequency: 80
[02:58:36][D][debug:328]: Flash Chip ID=0x0016405E
[02:58:36][D][debug:329]: Reset Reason: Software/System restart
[02:58:36][D][debug:330]: Reset Info: Software/System restart
````

bdraco, 2024-01-03:
I added a d1 mini board to the site https://ratgdo.github.io/esphome-ratgdo/

rlowens, 2024-01-03:
> Edit: The "Flash Real Size" is showing 4194304.0.

Great! So it really isn't a "lite"

amaisano, 2024-01-25:
Sorry, what's the solution to this? I already tried the new firmware. The next ESPHome update this week causes the same issue again...

bdraco, 2024-01-25:
flash once from the website and than it will be updatable from the dashboard

PPatla, 2024-01-25:
If you have encryption enabled you can just rem it out, do the update then put it back.

I started with the 2.5i firmware, then REMd it out to go to the beta, then changed over the to main and all my ESPHOME updates work no problem

substitutions:
  name: ratgdov25i
  friendly_name: ratgdo single
packages:
#  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main
# ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@sec_plus_v1
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 3RqH5ILTG5fWhaetXSiLg74P0u28MhIDsf37N8fVg+o=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

shaunburdick, 2024-01-25:
I was able to do it by flashing a base image first via OTA, then my normal.

If it helps, here is the process:

1. Comment out your normal config, add just ESPHome and OTA:
```yaml
substitutions:
  name: ratgdov25i
  friendly_name: Ratgdo Garage Door

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
esp8266:
  board: d1_mini
ota:
```
2. Perform the OTA Build and Install
3. Put your normal config back
4. Perform the OTA Build and Install

I kept the small version as a comment in case I need it again:
```yaml
substitutions:
  name: ratgdov25i
  friendly_name: Ratgdo Garage Door

# FOR USE
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: !secret api_encryption_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# # FOR UPDATE
# esphome:
#   name: ${name}
#   name_add_mac_suffix: false
#   friendly_name: ${friendly_name}
# wifi:
#   ssid: !secret wifi_ssid
#   password: !secret wifi_password
# esp8266:
#   board: d1_mini
# ota:
```

niemyjski, 2024-01-30:
I'm running into this with https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25board_esp8266_d1_mini_lite.yaml

niemyjski, 2024-02-03:
I bought 4 2.5 devices and I can't flash any of them.... Do you have the exact spec of d1 mini that you bought? One of mine even says d1 mini on it but it won't flash.

rlowens, 2024-02-03:
> I'm running into this with https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25board_esp8266_d1_mini_lite.yaml

Why are you using that file? The "lite" version doesn't have enough storage to allow OTA, as described in this thread. 

> I bought 4 2.5 devices and I can't flash any of them.... Do you have the exact spec of d1 mini that you bought? One of mine even says d1 mini on it but it won't flash.

The "d1 mini" isn't the "d1 mini lite" and should be switched to the non-"lite" version of the file, so that future OTA will work.

What is the exact error you are getting now and what happens when you use the steps in https://github.com/ratgdo/esphome-ratgdo/issues/155#issuecomment-1911016828 ?

niemyjski, 2024-02-03:
I flashed with ESPHOME 12.9 multiple times with the exact config that is out there, and it worked. It doesn't work since .10 minor release. I'm not going to try that because they are installed, and I don't want to get up on a big ladder and unwire it...  One device says D1 Mini but it is not a d1 mini. Also the png's on the site should be removed, they are of the worst quality I have ever seen.

rlowens, 2024-02-03:
> I'm not going to try that because they are installed, and I don't want to get up on a big ladder and unwire it.

The steps here are all done OTA, no need to take it down and rewire anything. Unless you are saying you aren't willing to correct the error and want to just stick with the current firmware? (In which case, why are you posting here?)

edrikk, 2024-02-04:
I decided to give the esphome fork a try today - this is a fresh install coming from the mqtt fork.
I also am seeing the same behavior as niemyjski:

- Initial install (coming from mqtt version) works without issue via the web page "ota" - (browse, pick file, update)
- Once on the Esphome version, trying to load the exact same esphome firmware as above fails with "Update Failed: ERROR[4]: Not Enough Space" error if performed via the broswer page (browse, pick file, update).
- Doing the update inside Esphome platform (i.e. install -> wirelessly once adopted works without issue.

rlowens, 2024-02-04:
> This is on a 2.52 device, using v25board_esp8266_d1_mini_secplusv1.yaml to start the process.

Looks like that file has the wrong board setting. I've created a PR to fix it https://github.com/ratgdo/esphome-ratgdo/pull/219

edrikk, 2024-02-04:
Confirmed working now.  Thank you. :)

PR #156 (2023-12-24): fix: v2.5i boards should use d1_mini
fixes #155