ratgdo/esphome-ratgdo #157: opener 8500 esphome 2.5i board not working

thplex48178, 2023-12-24:
Hi, this may be a duplicate, but can't find info if the esphome firmware supports the 8500.  According to the purchase paper it was supposed to be 2.0,  but I tried to install 2 of my 3 and neither seems to work, so I was going to do number 3 with mqtt and saw there was an option for a beta.  Trying to avoid mqtt due to other integrations already in place and I did not see an easy way to test the other mqtt fw with just a web interface as is the case with esphome.

Trying to limit the troubleshooting process. 

thanks

PaulWieland, 2023-12-24:
The 8500 uses the sec plus 1.0 serial protocol.

closed 2023-12-24 by PaulWieland

benze, 2025-03-10:
@PaulWieland I have an 8500 and am also a bit confused; the docs for the 8500 say that it is Security+ 2.0.   Is their documentation misleading in that they only use Sec+ 2 for wireless protocol whereas they are running a different protocol version for hard-wired serial comms?

Thanks for the clarification!